espressif/esp-idf #1117: How to clean the compiled bootloader

Gfast2, 2017-10-13:
Hello esp-idf,

```
--- idf_monitor on /dev/ttyUSB0 921600 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
���������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

```
does this means that my bootloader is not fired up at the very beginning?

_Ps:
I'm not sure if this only caused by the update of my toolchain and esp-idf framework_

Gfast2, 2017-10-13:
I believe I got some:
```
W (51) rtc_clk: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 26 MHz.
```
But if I do change spi speed to 26MHz. This error won't appear again. but The effect is the same. 🤔

negativekelvin, 2017-10-14:
You don't change spi speed to 26mhz,  you change crystal frequency

Gfast2, 2017-10-15:
Hi @negativekelvin , Thanks for your help. ⚡️ 

I did spend some time to compare the working `sdkconfig` from my git history with the one on the `HEAD` right now. I've figured out **`'make monitor' buad rate`** is no more the default one -> 115200. 

---

Even though I did the **`make clean`** and recompile the project and do the upload (with make monitor setting equal to 921600). When I start my _serial monitor_. I still get garbage infos. Then I use another serial monitor software, and figured out I can get the acutual logs with 115200 baud (which ***should be*** 921600 according to my understand), I'm pretty sure, the only reason for this is caused by the unmatching baud rate between this setting and the actual setting from **`make monitor`**.  which let me believe, what I've changed will not be accepted totally. What was wrong? 

(Since this issue won't block my working process any further. I will close this issue if its more then 24 hour not been discussed) 📶

Gfast2, 2017-10-15:
[***`Here`***](https://gist.github.com/Gfast2/2df7979d4a09132196c0df0b18c59a5e) is `sdkconfig`.
I use '912600' as keyword searched the whole file. And figured out `make monitor` should use 921600 as its baud rate for communication,

Gfast2, 2017-10-15:
I did checkout how will it looks like if I change this setting in _ESP-IDF_'s stock "hello_world" example. Yes, I won't properly neither.
Now I believe I'd say. This part of _ESP-IDF_ perhaps should be more ergonomical. 🤣

negativekelvin, 2017-10-16:
You would have to change this too https://gist.github.com/Gfast2/2df7979d4a09132196c0df0b18c59a5e#file-sdkconfig-L161

But I think stage 1 bootloader will always be 115200

Gfast2, 2017-10-16:
Hi @negativekelvin ,
Thanks for you declaration. Because the developments goes on. I won't check your suggestion sooner. But later!

And I do hope this setting can be reflected on the GUI Interface, too or let the _Serial Monitor_ more configuration possiblity after it is fired up. (Like the _Serial Monitor_ in Arduino IDE dose)

But thanks again for your time and very very helpful advices!! 👯‍♂️

closed 2017-10-16 by Gfast2