espressif/esp-idf #1121: LMP timeout error after "create connection" HCI command with classic HID device

execuc, 2017-10-14:
Hello,

My aim is to connect classic HID bluetooth devices to ESP32. A simple test program send HCI commands in order to be connected to a Sony PS4 controller. After an inquiry it fails at the result of "create connection" with a connection complete event showing a LMP timeout error (0x22). This test program works with other HID devices like the Xiaomi gamepad.
Same PS4 controller works with various USB bluetooth dongles sending same HCI commands (from the usb host shield project).

I have also tested the btstack library port to ESP32 but it also failed at the connection part with the same LMP timeout. Do you have any idea, please ?

mywang-espressif, 2017-10-24:
Hi, can you provide some log files in terms of the HCI commands|events|data,  or air packets from packet sniffer, or project that produces this issue, so that I can analyse the issue, and build the test environment and reproduce the issue on my own?

execuc, 2017-10-25:
Hi,

I have a simple test program derived from example/bt/app_bt (attachment : bt_custom.txt). It is a mini state machine which initialize bluetooth, launch inquiry and start a connection. I am sorry because this test program is no clean.

There a two logs join :
 * debug_ps4_v2_conn.txt : NOK test with a ps4 controller. At the end, connection complete event send a 0x22 status (04 03 0b 22 80 00 50 01 a0 5f 1b 40 01 00 ).
 * debug_xiaomi_conn.txt : OK test with a xiaomi gamepad. Connection complete event is OK with 0x00 status (04 03 0b 00 80 00 7b ee cb 93 fd ac 01 00). But esp32 resets after 0x1B event (Max slot event ?) but it is not the problem here and it works with btstack.

Thank you very much

[debug_ps4_v2_conn.txt](https://github.com/espressif/esp-idf/files/1413502/debug_ps4_v2_conn.txt)
[debug_xiaomi_conn.txt](https://github.com/espressif/esp-idf/files/1413503/debug_xiaomi_conn.txt)
[app_bt_custom.txt](https://github.com/espressif/esp-idf/files/1413504/app_bt_custom.txt)

mywang-espressif, 2017-11-02:
Hi @execuc , thanks for your example code and HCI dump files. I have read them, the procedure is simple actually and I cannot decide what is wrong according the level of HCI loggings. there must be something wrong in the controller/radio of the devices.
Do you have a bluetooth packet sniffer to capture some air packets? Or I may need to find a Sony PS4 controller.

execuc, 2017-11-05:
I thank you for watching the logs. I'm sorry but I do not have a bluetooth sniffer. If you could have a PS4 controller, it is good, otherwise it does not matter.
PS4 controller is maybe a little specific and if the other classic bluetooth HID peripherals works well with ESP32 then it's not a big issue... but I only test two devices (PS4 and Xiaomi gampad). Did you test such other devices ?

mywang-espressif, 2018-03-21:
Hi @execuc , I am sorry for that I do not have such HID devices in hand. The error code(0x22)  indicates LMP_response_timeout, and I think the issue lies in the LMP transaction. Since the air packets are not available, I cannot decide the root cause. 
The test example is quite a simple one. You said that  you have tested with BTStack library, I think BTStack library provides more basic settings for controller during initialization, is it possible that we get more information from BTStack loggings?

FayeY, 2018-04-04:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

closed 2018-04-04 by FayeY

siddhya, 2018-08-01:
Any progress on this? I would also like to interface the PS4 remote (which enumerates as BT classic HID).
Is there any reference code for it?