espressif/esp-idf #1122: How to communicate with the blufi example

jvandewync, 2017-10-15:
Hello,

First of all, the blufi example works just fine with the EspBlufi Android app. However, when I try to create a swift application (iOS) for communicating a custom SSID and password, I'm not sure what to send. I have tried to send some strings such as "ssid:my_ssid", "passwd:my_passwd", "confirm:", but it is not working. Is it possible to have a more complete documentation on how to communicate with the blufi example?

Weijian-Espressif, 2017-10-23:
hi @jvandewync , you can refer Esp-blufi ios code, you can use the lib directly,here is the link. https://github.com/EspressifApp/EspBlufiForiOS

closed 2017-10-23 by FayeY

n0p, 2017-10-23:
But what if we want to implement blufi in anything that's not iOS or Android native (for example linux or even ionic)
A protocol reference would be nice

jvandewync, 2017-10-23:
I have already investigated the iOS code, but thanks for the link @Weijian-Espressif. Similarly, I was wondering if Espressif could provide a protocol documentation.

Weijian-Espressif, 2017-10-24:
The document is not open,  please tell us if there are other needs

ravitej-BLR457, 2017-11-21:
Hi , While trying this Blufi ios .. getting this error.  Request you to help.

Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$_PacketCommand", referenced from:
      objc-class-ref in BLEViewController.o
  "_OBJC_CLASS_$_DH_AES", referenced from:
      objc-class-ref in BLEViewController.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Weijian-Espressif, 2017-11-21:
@ravitej-BLR457  I test my project, everything is OK. it seem like that your library path  is wrong, please check.

ravitej-BLR457, 2017-11-21:
Hi , Thanks for the reply . The sample is working. 
I'm trying to use for one of our app , can you please point to some doc or code how to import this library and use for customisation.  Does this support Swift code?

jvandewync, 2017-11-22:
It is working fine for me in Swift.

ravitej-BLR457, 2017-11-22:
Hi,

Request you to share some sample using the same lib in Swift. Will be very
helpful.

If any doc to integrate and test also helps.

Thanks in advance.

Regards
Ravitej Tatipamula
Ph: 9342883292

ravitej-BLR457, 2017-11-22:
Hi , Do we have any document or sample swift code suing the library specific to  ios . It will be a great help.