espressif/esp-idf #1123: bluetooth client example

denyip, 2017-10-16:
Let's say if I have 6 char in one service, with below code in the example, I should be able to create a capacity to store 6 set of char elem result, and then use esp_ble gattc_get_char_by uuid to search the uuid one by one. Is it the correct way for me to read and store the result for the second char like below code "(char_elem_result+1)" ? my esp crash and restart.....

`char_elem_result = (esp_gattc_char_elem_t *)malloc(sizeof(char_elem_result) * count);
.
.
status = esp_ble_gattc_get_char_by_uuid( gattc_if,
							p_data->search_cmpl.conn_id,
							gl_profile_tab[PROFILE_A_APP_ID].service_start_handle,
							gl_profile_tab[PROFILE_A_APP_ID].service_end_handle,
							first_char_uuid,
							(char_elem_result),
							&count);
.
.
status = esp_ble_gattc_get_char_by_uuid( gattc_if,
							p_data->search_cmpl.conn_id,
							gl_profile_tab[PROFILE_A_APP_ID].service_start_handle,
							gl_profile_tab[PROFILE_A_APP_ID].service_end_handle,
							second_char_uuid,
							(char_elem_result+1),
							&count);
`

denyip, 2017-10-17:
I use the latest version IDF, with the module from brand A, when I use the function esp_ble_gattc_get_all char, it will trigger the heap corruption. 

However, when I use the module from another brand B, this problem doesn't appear, anyone can help?

CORRUPT HEAP: multi_heap.c:173 detected at 0x3ffcf03c
abort() was called at PC 0x400882fa on core 0

Yulong-espressif, 2017-10-22:
@denyip I found that if your code is wrong, it should be written as follows:
char_elem_result = (esp_gattc_char_elem_t *) malloc (sizeof (esp_gattc_char_elem_t) * count);
Any other question, please let me know, thanks.

denyip, 2017-10-23:
then it's the wrong code in the IDF example, in the example, it  wrote sizeof(char_elem_result)....

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_client/main/gattc_demo.c

Yulong-espressif, 2017-10-23:
@denyip Thank you very much, it is really a bug in the IDF bt excample. We will change it in the last few days and thank you again.

Yulong-espressif, 2017-11-04:
@denyip Can you update the master now, the issue is fixed in the new version of the master. Thanks,

closed 2017-11-14 by FayeY