espressif/esp-idf #1124: ESP-IDF  ESP-32 hellow_world Monitor error issues Help me.

DaftpunkK, 2017-10-16:
my spec:  sparkfun ESP32 Thing
                windows 10 64bit
                mingw32 installed
                make monitor baud rate : 115200 bps

I have tried some simple examples(hello_world..etc) using esp-idf.

The make flash part works fine.

However, when I run make monitor, the strange characters appear in all examples.

There is no separate error indication.

Is there anyone who knows about the problem?

ShendoXT, 2017-10-16:
Crystal frequency used to be set to auto-detect but that has been [changed recently](https://www.esp32.com/viewtopic.php?f=13&t=2800).
Your board probably has a 26 Mhz crystal.

Run make menuconfig and select a frequency manually.

DaftpunkK, 2017-10-18:
thank you ShendoXT

closed 2017-10-19 by FayeY