espressif/esp-idf #1126: socket connect failed errno=113

aughey, 2017-10-16:
I'm having issues reliability connecting to a remote host.  I'm now running the example application, http_request unmodified.  This application connects to example.com and retrieves a web page., waits 10 seconds, and repeats  It always works the first time, sometimes the second time, but will always fail shortly thereafter with the "socket connect failed errno=113" message.

```
I (125474) example: 2...
I (126474) example: 1...
I (127474) example: 0...
I (128474) example: Starting again!
I (128474) example: Connected to AP
I (128474) example: DNS lookup succeeded. IP=93.184.216.34
I (128474) example: ... allocated socket
E (145984) example: ... socket connect failed errno=113
I (149984) example: Connected to AP
I (149984) example: DNS lookup succeeded. IP=93.184.216.34
I (149984) example: ... allocated socket
E (167704) example: ... socket connect failed errno=113
I (171704) example: Connected to AP
I (171704) example: DNS lookup succeeded. IP=93.184.216.34
I (171704) example: ... allocated socket
E (189454) example: ... socket connect failed errno=113
```

I originally had issues posting to web pages using Espruino.  Frustrated, I tried micropython with the same results.  First post always works and fails thereafter.  I'm sure sockets are closed properly.  And I am using the latest esp-idf where I've built from source, and I've download the pre-built bin files for Espruino and micropython with the same underlying issue.  Sometimes after a few failures, it succeeds again.  I'm sure I'm testing on a reliable network.

nkolban, 2017-10-17:
Can you pastebin the whole app so that we can see the error in context?

negativekelvin, 2017-10-17:
@nkolban  

> I'm now running the example application, http_request unmodified.

nkolban, 2017-10-17:
For those that follow along ... the URL to this application is:

https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_request/main

nkolban, 2017-10-17:
Here's a matching forum post:

https://www.esp32.com/viewtopic.php?f=13&t=2166

It may offer some assistance.  One chap said he was running an old SDK and another mentioned it was a firewall problem.

aughey, 2017-10-17:
I've seen similar posts to the one you mention @nkolban .  I haven't found a good solution.  In that post, one person said updating the SDK fixed it, and another still has trouble.  I've tried using Espruino, MicroPython, compiling http_request directly, and using the Arduino sdk.  Same issue with all of them.  The first connect always works, the second and sometimes 3rd and 4th connects work, but within less than 10 connects, it fails.  People suggest sockets are not being closed, and I'm absolutely sure everything is cleaned up.

I'm testing this today on a second ESP32 board.  Same firmware, different board.  My engineering gut says it should be software, but I have no other ideas at this point.

nkolban, 2017-10-17:
Do you have an alternate path to the Internet?  Perhaps set your cell phone up as an access point and try connecting through that?

Alternatively, again as a test, setup a Web Server on a PC in the same WiFi LAN as your ESP32.  I'm loosely thinking that one of the posts mentioned firewall ... it may be that your router/home access point is doing some "strange filtering".  Again, wild guess on my part but any tests we perform that result in different outcomes changes our data points.

FayeY, 2017-10-19:
Hi @aughey , did the above suggestion help you resolve the issue?

aughey, 2017-10-20:
So yes and no.  I did test the ESP32 on a cell-phone wifi hotspot and it performed correctly.  And the same test on a different WiFi network failed.  So, I think this specific issue can be closed, but I'm going to want to open up another once I collect more information.

The ESP32 should be more resilient on whatever network it is using.  This is an IoT device that is brought to its knees on a very common consumer home wifi network.  Clearly something is different, likely with how the final RST or FIN TCP messages are being sent.  Since I can easily reproduce the problem, I will setup a wifi sniffer in monitor mode to capture the differences on each network and report back in a different issue.

nkolban, 2017-10-20:
Can you post the name/type/release of your home WiFi access point?  It may be that there is a common pattern we can see with a specific model.  The vast majority of ESP32 users don't have any issue connecting to the Internet through their home access points.  That was said not to belittle your observation ... a problem is a problem and should be fixed.  However, it is not outside the realms of possibility that you just got unlucky and a particular model of access point itself has some subtle bug that may be able to be identified.  If we know the access point type, we might be able to use that as a key phrase on current and future puzzles to see if there is a pattern.  At a minimum, we can then start polling the community to see if anyone else is using that device and if they are working, what software revisions or settings are  set on the access point that allows them to work.

aughey, 2017-10-24:
It's a TP-LINK 450M.  

Model No. TL-WR940N.  
Firmware Version: | 3.16.9 Build 160617 Rel.58968n
Hardware Version: | WR940N v4 00000000

There is a firmware upgrade available, but I want to capture what's going wrong before I try to upgrade.  My goal isn't to get it to work, my goal is to make sure it can't be broken.  If an IoT device can be brought down that easily, it wouldn't be viable as a commercial device.

liuzfesp, 2017-11-26:
@zhangyanjiaoesp please check it.

zhangyanjiaoesp, 2017-12-11:
@aughey Can you post the ESP IDF version (or commit ID)you used?

Pines, 2017-12-25:
I do have the exactly same problem.

ESP32 through TP-LINK 885N 450M. 
Have problem when calling connect(s,name,namelen), errno is 113 "No Route to Host"

Retry couple of ten times connect never success, (I have testing mobile phone connect to same WiFi router, TCP with same IP everything OK.) once this happen, it normally can't connect, even reset CPU problem will remain (have reset router once, as i recall no lucky).

This surely be worthy to have a look. my ESP-IDF last update is around 10.1 this year.
Does newly version solve this, and what changes in code are being related.

Pines, 2017-12-27:
I update latest ESP-IDF master and git submoudle update.
confirm today, this occur at TP-Link 885N and a Netgear 150M(a little old one) WiFi AP

@zhangyanjiaoesp @aughey @nkolban guys, any new progress.

aughey, 2017-12-27:
Back when I first posted this, I was able to capture some network traffic using Wireshark. 

I've posted the screen captures and narrative to this repository.  https://github.com/aughey/esp32_tcp_issue

zhangyanjiaoesp, 2017-12-28:
@aughey  Can you post your idf commit ID?

zhangyanjiaoesp, 2017-12-28:
@Pines you mean the latest idf still have the problem? All the tests will fail? Or just with the router you provided?

Pines, 2017-12-28:
@aughey 
Today I Make TP-Link internet connection to [Yellow] [WAN] port.
Test three ESP32 chips, couple times reset on each, TCP connect 113 error didn't occur.

This means, must make TP-Llink 885N as  a router,

when internet connect to [BLUE] [LAN] port, make TP-Link as AP. there will be TCP finding route problem.
Just my guessing, TP-Link designing purpose is run as a WiFi Router, so they didn't making much effort dealing with condition a Internet connect to LAN port.

Not sure my experience will suit others, but hoping this provide some inspiration. anything new please let me know. thanks to aughey , this thread is very helpful.

aughey, 2017-12-29:
The issue is with the latest idf with these particular routers.  I believe idf is taking some liberties with the protocol that the routers are treating either more conservatively or possibly incorrectly.  I can take a closer look at it again and capture some additional data.  

What I captured seemed to be inconsistent with the source code I looked at.  If I had a wired connection, the capturing would be easier.

John

heyinling, 2017-12-29:
@aughey could you plz also help to provide your configuration of the router, especially the LAN (DHCP server) config. we'll try to reproduce this issue. We'll try to reproduce it.

@Pines when you connect to LAN port, plz disable DHCP server on router and then check if it works.

aughey, 2017-12-29:
The router is: TP-Link, 450M Wireless N Router, Model No. TL-WR940N, Firmware Version 3.16.9 Build 160617 Rel.58968n

Nothing special with the DHCP config beyond the default setup.

I did another capture this evening with nothing revealing.  See next comment.

aughey, 2017-12-29:
Did two tshark captures.  One with the ESP32 and one with a Raspberry PI.  The captures are almost identical which probably doesn't help much.

This data was captured on a local network.  All devices wifi.  Two raspberry pi's and an esp32 running micropython.  

See below...

# ESP32 running MicroPython

<pre>
pi@raspberrypi:~ $ sudo tshark -P -i wlan0 -f 'host 192.168.0.120'
tshark: Lua: Error during loading:
 [string "/usr/share/wireshark/init.lua"]:46: dofile has been disabled due to running Wireshark as superuser. See http://wiki.wireshark.org/CaptureSetup/CapturePrivileges for help in running Wireshark as an unprivileged user.
Running as user "root" and group "root". This could be dangerous.
Capturing on 'wlan0'
  1   0.000000 192.168.0.120 -> 192.168.0.111 TCP 58 60810→12345 [SYN] Seq=0 Win=5744 Len=0 MSS=1436
  2   0.000166 192.168.0.111 -> 192.168.0.120 TCP 58 12345→60810 [SYN, ACK] Seq=0 Ack=1 Win=29200 Len=0 MSS=1460
  3   0.045905 192.168.0.120 -> 192.168.0.111 TCP 54 60810→12345 [ACK] Seq=1 Ack=1 Win=5744 Len=0
  4   0.052613 192.168.0.120 -> 192.168.0.111 TCP 66 60810→12345 [PSH, ACK] Seq=1 Ack=1 Win=5744 Len=12
  5   0.052761 192.168.0.111 -> 192.168.0.120 TCP 54 12345→60810 [ACK] Seq=1 Ack=13 Win=29200 Len=0
  6   0.053359 192.168.0.120 -> 192.168.0.111 TCP 54 60810→12345 [FIN, ACK] Seq=13 Ack=1 Win=5744 Len=0
  7   0.053631 192.168.0.111 -> 192.168.0.120 TCP 54 12345→60810 [FIN, ACK] Seq=1 Ack=14 Win=29200 Len=0
  8   0.129037 192.168.0.120 -> 192.168.0.111 TCP 54 60810→12345 [ACK] Seq=14 Ack=2 Win=5743 Len=0
</pre>

# Raspberry PI Linux

<pre>
pi@raspberrypi:~ $ sudo tshark -P -i wlan0 -f 'host 192.168.0.119'
Capturing on 'wlan0'
  1   0.000000 192.168.0.119 -> 192.168.0.111 TCP 74 54272→12345 [SYN] Seq=0 Win=29200 Len=0 MSS=1460 SACK_PERM=1 TSval=80958 TSecr=0 WS=64
  2   0.000197 192.168.0.111 -> 192.168.0.119 TCP 74 12345→54272 [SYN, ACK] Seq=0 Ack=1 Win=28960 Len=0 MSS=1460 SACK_PERM=1 TSval=69941078 TSecr=80958 WS=64
  3   0.011233 192.168.0.119 -> 192.168.0.111 TCP 66 54272→12345 [ACK] Seq=1 Ack=1 Win=29248 Len=0 TSval=80960 TSecr=69941078
  4   0.011909 192.168.0.119 -> 192.168.0.111 TCP 78 54272→12345 [PSH, ACK] Seq=1 Ack=1 Win=29248 Len=12 TSval=80960 TSecr=69941078
  5   0.012050 192.168.0.111 -> 192.168.0.119 TCP 66 12345→54272 [ACK] Seq=1 Ack=13 Win=28992 Len=0 TSval=69941079 TSecr=80960
  6   0.012640 192.168.0.119 -> 192.168.0.111 TCP 66 54272→12345 [FIN, ACK] Seq=13 Ack=1 Win=29248 Len=0 TSval=80960 TSecr=69941078
  7   0.012912 192.168.0.111 -> 192.168.0.119 TCP 66 12345→54272 [FIN, ACK] Seq=1 Ack=14 Win=28992 Len=0 TSval=69941079 TSecr=80960
  8   0.046844 192.168.0.119 -> 192.168.0.111 TCP 66 54272→12345 [ACK] Seq=14 Ack=2 Win=29248 Len=0 TSval=80963 TSecr=69941079
</pre>

I retract what I thought was happening with bad tcp setup sequences.  There still could be something there.

Pines, 2017-12-29:
@heyinling 
When "Internet connection"(a cable come from switch, switch will connect to a real optical fiber Internet modem ) connect to TP-Link 885N LAN port. 
Because we got a DHCP server for whole office already, so as i remember i disable the DHCP  server on TP-Link.

that means the TCP connect 113 error 
1. Internet to LAN
2. DHCP server on TP-Link is disable, but there is another DHCP server from LAN port, so ESP32 will been allocate a dynamic IP.
3. I remember I did this testing over a NetGear 150M WiFi router, internet to LAN. DHCP server disable, 113 error
4. UDP service like NTP, DNS always is good, NTP might trying like at most 5 times, finally it will success, even when TCP is down.

zafrirron, 2018-01-27:
I'm having the same issue,
Regardless of firewalls and routers...
running local web web server on internal network, accessing the server url with no issue from other clients (wired and wireless) but failed to use the example http_request app to get the same GET result.
get error 133 on socket connect.....
latest esp-idf used.

zhangyanjiaoesp, 2018-01-29:
@aughey  
I have bought the TL-WR940N，450M router for the test, but the Hardware Version is v6,so I can't produce your problem.
@Pines 
I use router TL-WR885N 450M to test, Firmware Version : 2.0.4, Hardware Version: v4.2, the test case is http_request example, commit is:3bc1a83. I have tested four scenarios:
(1) TL-WR885N internet connection to[WAN] port，DHCP server on. 
      http_request example works fine.
(2) TL-WR885N internet connection to[LAN] port，DHCP server on.
      http_request example occurs 113 error.
      

> I (39700) example: Connected to AP                                              
I (39700) example: DNS lookup succeeded. IP=93.184.216.34                       
I (39700) example: ... allocated socket                                         
E (57440) example: ... socket connect failed errno=113 

(3) TL-WR885N internet connection to[WAN] port，DHCP server off.
      http_request example occurs 202 error. 

> Connected to AP
E (18040) example: DNS lookup failed err=202 res=0x0
I (19040) example: Connected to AP
E (26020) example: DNS lookup failed err=202 res=0x0
I (27020) example: Connected to AP
E (34010) example: DNS lookup failed err=202 res=0x0

(4) TL-WR885N internet connection to[LAN] port，DHCP server off.
      http_request example works fine.
In scenarios(2) and (3), http_request example returns error, but it's normal behavior.

FayeY, 2018-03-29:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2018-03-29 by FayeY

Pines, 2018-05-30:
Hi, guys:

After read this topic, I think might be more ppl encounter this issue.
https://www.esp32.com/viewtopic.php?f=2&t=2665&start=30

I would like share my personal update on this topic, hoping help ppl still steak with this.

I capture the TCP/IP packets, the sequence is like following.
1) Esp32 send a tcp packet with SYN flag, indicate start TCP 3-way handshake with remote.
2) Normally in success scenario, server response SYN+ACK.

When fail, connect function didn't , errorno is 113. 
I capture no SYN+ACK from server, but a ICMP packet from gateway, this ICMP packet type 3,code 0. 
which means destination network unreachable.

1) Then i printed dynamic IP esp32 module allocate from dhcp server.
2) shutdown esp32 module
3) assign this ip as static ip to a desktop computer, then use this computer try ping or connect to remote ip as esp32 module did, exact same error happened ( destination network unreachable ).

this is the result ICMP from gateway.

OK, long to short. Its NAT( network address translation ) memory a session from local ip to remote ip.
and, normally NAT have a limited max session.

When start esp32 and start TCP connect to remote server, NAT add new session, sometime esp32 didn't shutdown correctly ( means didn't close the socket connection, send FIN through NAT), NAT session remain.

So, the reason for socket error 113 in my case, NAT session reach the max.
Now, I try make sure every socket I opened have a official close, yes, that is it.

hope this help, and anything new please let me noticed.

Ritesh236, 2018-10-05:
@Pines ,

We are facing same socket 113 error when we are downloading firmware from server in which ESP32 is working as Client.

In this scenario sometimes we are getting 113 error while connecting with created socket and our task has been deleted due to that condition

We are using ESP32 IDF 2.1 Stable Release for our development purpose. We have also enabled LWIP debug and error logs and found that SYN error we are getting when this type of issue is being occurred.

So, Do you have any idea or any clue regarding this issue?

Amorph, 2020-02-17:
Have same issue on the latest SDK

wijnsema, 2020-08-31:
I faced this error 113 error also. In the end it turned out to be a problem with my **Windows server**. It uses "**Network Policy and Access Services**" for routing with the help of the Windows DHCP service. It has been running for years without much problems but the ESP32 Devkit with IDF 4.0 couldn't make a reliable connection, getting the 113 errors once in a while.

my-dudhwala, 2023-09-25:
Getting same error while trying to connect to internet via IPv6...

Ritesh236, 2023-09-26:
> Getting same error while trying to connect to internet via IPv6...

Hello,

Which ESP32 IDF you are using at your end? Would you please provide more steps over here like what are you doing basically into your code development?

Regards,
Ritesh Prajapati