espressif/esp-idf #1130: close esp_log

FFtust, 2017-10-18:
HI,
   I tried to close all the logs by configing the esp_log.h, but I found the logs about wifi that I could not close,
this is the out logs:
    I (12185) wifi: wifi firmware version: f092575
    I (12187) wifi: config NVS flash: enabled
    I (12187) wifi: config nano formating: disabled
    I (12193) wifi: Init dynamic tx buffer num: 32
    I (12193) wifi: Init dynamic rx buffer num: 64
    I (12197) wifi: wifi driver task: 3ffe346c, prio:23, stack:4096
    I (12201) wifi: Init static rx buffer num: 10
    I (12205) wifi: Init dynamic rx buffer num: 0
    I (12209) wifi: Init rx ampdu len mblock:7
    I (12213) wifi: Init lldesc rx ampdu entry mblock:4
    I (12219) wifi: wifi power manager task: 0x3ffe9ac0 prio: 21 stack: 2560
    I (12787) wifi: wifi timer task: 3ffeab28, prio:22, stack:3584
    I (12877) wifi: mode : softAP (30:ae:a4:27:de:55) 
Is there any way to close these logs? Or even all the logs that the idf out?

igrr, 2017-10-18:
See
http://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html#_CPPv217esp_log_level_setPKc15esp_log_level_t
#508 
#522

FFtust, 2017-10-18:
Thank you for your help!

closed 2017-10-19 by FayeY