espressif/esp-idf #1131: No custom log output on UART1 GPIO16,17 after boot

Curclamas, 2017-10-18:
## Expected Behavior
When selecting custom log output UART1, RX: GPIO16, TX: GPIO17, all stuff loged via ESP_LOG macros shall be visible via serial.

## Observed behavior
Since a couple of weeks (sorry, can't pinpoint it), after pulling latest idf: All boatloader messages are visible, then when the app starts, no log output is visible. There are three ways to mitigate this right now:

- choose UART0 (not an option)
- On native esp-idf run
```cpp
void app_main(void)
{
    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_UART1_CLK_EN);
    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_UART1_RST);
    // ....
}
```
- On Arduino run 
```cpp
    HardwareSerial serialUART1(1);
    serialUART1.begin(115200, SERIAL_8N1, 16, 17);
    serialUART1.println("Hello on UART1");
    serialUART1.flush(); 
```

Both of the last hacks work, but it would be nice to find the root cause of this. IMHO ESP_LOG should always work when the correct settings in `sdkconfig `are set.

## Steps to Reproduce

- In `sdconfig `set:
```bash
CONFIG_CONSOLE_UART_CUSTOM=y
CONFIG_CONSOLE_UART_CUSTOM_NUM_1=y
CONFIG_CONSOLE_UART_NUM=1
CONFIG_CONSOLE_UART_TX_GPIO=17
CONFIG_CONSOLE_UART_RX_GPIO=16
CONFIG_CONSOLE_UART_BAUDRATE=115200
```
- Compile, flash an reboot device
- No log output from app visible
- Try hacks from [Observed behavior](#observed-behavior)
- No it works again

## Notes
This is also mentioned in https://www.esp32.com/viewtopic.php?f=13&t=3166

negativekelvin, 2017-10-18:
https://github.com/espressif/esp-idf/commit/4776f73ca4651a5bb65b4db2407205c0dd39925e#diff-b1df72eee4bf4584db9a57c5b0489832

Curclamas, 2017-10-18:
This looks indeed like the root cause here. Where should this be fixed? I mean: when selecting UART1 as log output, this peripheral clock should either not be disabled or enabled again right in place, right?

igrr, 2017-12-08:
Closing as fixed.

closed 2017-12-08 by igrr