espressif/esp-idf #1132: Spurious crash, then app won't start: "W (32) rtc_clk: Bogus XTAL frequency: 0 MHz"

Curclamas, 2017-10-18:
## Expected Behavior
The ESP32 can run basically infinite (when there is no crash in the application). When the application crashes, it shall reboot and run the application without any lockup.

## Observed Behavior
The app crashes after a day or so. Cause: `StoreProhibited` by: ` _gettimeofday_r`. Interesting enough though this function is not called in that situation. It seams the code enters an if branch, that it could never enter in that situation.

However, even if that is a bug on my code: After the crash a normal reboot should occur. Instead the application won't start. First I see:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:7772
load:0x40078000,len:0
load:0x40078000,len:13760
entry 0x40079384
W (32) rtc_clk: Bogus XTAL frequency: 0 MHz
```

then:
```
>>>...<<<
I (730) cpu_start: Pro cpu up.
I (733) cpu_start: Starting app cpu, entry point is 0x40080f8c
I (740) cpu_start: App cpu up.
I (744) heap_init: Initializing. RAM available for dynamic allocation:
I (751) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (757) heap_init: At 3FFBAB60 len 000254A0 (149 KiB): DRAM
I (763) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (770) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (776) heap_init: At 40091000 len 0000F000 (60 KiB): IRAM
I (782) cpu_start: Pro cpu start user code

>>>>Log ends here, application is not executed<<<<
```

## Steps to Reproduce
Cannot be easily reproduced, occurs only intermittently.

## Notes
Error occurs on a custom board with 40 MHz XTAL ( FL4000157Z with 2x22pF Load Caps ) and 32 kHz RTC-XTAL ( ECS-.327-12.5-34B-C-TR with 2x18pF Load Caps ). When the error does not occur it still prints:

`W (32) rtc_clk: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 40 MHz.`

Could this be somehow a thermal issue? Usually the error would occur, when the ESP32-Board is outside on a cold night. If thermal stability is the concern here, I could switch for a TCXO, but still, since the Datasheet specifies a XTAL, this should also work.

igrr, 2018-01-08:
Several issues related to restart process and clock switching have been fixed in abacf8d2a0347a4bf95315cdb227e9b8b1c93794. Please update ESP-IDF and give it a try.

Curclamas, 2018-01-08:
Thanks for your reply @igrr . We are most of the time developing on the actual master of esp-idf. Since this error has not occurred for a few weeks/months I think this error was fixed in idf and this issue can be closed.

closed 2018-01-08 by Curclamas