espressif/esp-idf #1133: When I2C hangs restarting ESP doesn't usually help

chanibal, 2017-10-18:
Some background: I'm integrating a SSD1306 128x64 OLED driver and I2C seems to be randomly failing on `i2c_master_cmd_begin` with error 263 (ESP_ERR_TIMEOUT). This seems to be related to #680.

The real issue is this is not fixable by a restart of the ESP - the error persists. I have tried four ways of restarting the ESP with different results:

1. using `esp_restart()` - does not help, error remains after restart
2. hanging the ESP32 on a critical section and waiting for the watchdog to restart - does not help, error remains after restart
3. using idf_monitor's reset functionality (ctrl+R) - does help
4. going to deep sleep and waking up with a gpio attached button - does help, but the I2C seems to hangs again quicker than with method 3

A bit more about the I2C hang: it happens randomly while updating the screen, can be easily triggered by restart method 2 mentioned above (hang in a critical section, be restarted by watchdog).
[image]
[image]
Reconnecting the I2C peripherial doesn't help.

IDF version v3.0-dev-396-g6d2a597  
WROOM32 ESP32D0WDQ6 (revision 0)

me-no-dev, 2017-10-18:
We have [this](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c#L473) to coop with that issue in Arduino. I did get reports that it does not help 100% of the time though...

FayeY, 2017-10-20:
Hi @chanibal , did the above suggestion help you resolve the issue?

chanibal, 2017-10-24:
Temporarily reverted to IDF version from top comment because of the issues in newer one (still haven't lost hope, just will return to upgrade at a later time).

Using the workaround @me-no-dev mentioned doesn't help by itself, but in combination with power cycling (disconnecting VCC) of the I2C OLED it works. Double checked that without disabling/enabling the peripherial, power cycling is useless.

And for completeness, a few other things I tried, without success:
* Just upgrading to WROVER32 ESP32D0WDQ6 (revision 1) did not seem to have an effect.
* Also tried `i2c_reset_tx_fifo` and `i2c_reset_rx_fifo`, but they do not seem to have any effect on the issue.

Still, resetting the ESP from external programmer does reset I2C properly without power cycling the OLED and the same OLED itself worked well for a long time on a arduino nano - so I don't think this is the OLED's fault.

Will not repeat tests on WROOM32, but I'll try to update the IDF and run them again.

negativekelvin, 2017-10-24:
https://github.com/espressif/esp-idf/commit/01befe4d6a2e94bfc7db6484ff586470edea3086

chanibal, 2017-10-24:
@negativekelvin will try patch as soon as I get newest IDF working.

In an question unrelated to this issue, if I make a pull request to esp-idf, should I create an issue describing it separately or is the pull request itself sufficient?

chanibal, 2017-10-24:
Tried with esp-idf `v3.0-dev-944-gba971a3` and it works with power cycling.
Tried with esp-idf `v3.0-dev-1028-g2c95a77` (that contains 01befe4) and it seems to work without power cycling - but will leave for the night to be sure.

chanibal, 2017-10-25:
Left it for the night, it still works in the morning with some timeout errors, but nothing breaking.

Added some stats on how frequently the timeout error occurs:
```
count:          64
mean:          686 sec. = 0.1906 hour
Q1:            164 sec. = 0.0455 hour
median:        420 sec. = 0.1167 hour
Q3:            921 sec. = 0.2558 hour
```

The display was updating all the time with around 70% bandwidth being used:
[image]

With this fixed, this issue probably doesn't make sense anymore.

closed 2017-11-22 by FayeY