espressif/esp-idf #1134: Feature Request: Allow user to change I2C receive timeout

arsinio, 2017-10-18:
I'm working with a sensor (SI7050) that uses clock stretching to delay the I2C transaction while the actual measurement/conversion is being performed.

From this forum post, I found where to modify this timeout:
https://www.esp32.com/viewtopic.php?f=2&t=1226

_driver/i2c.c - i2c_param_config:_
```
//set timeout of receving data
I2C[i2c_num]->timeout.tout = 200000;
```

The default `200000` winds up being ~ 2.5mS. According to the SI7050 datasheet, the maximum conversion time is 10.8ms. So my two recommendations are:

1. it would be nice to have an accessor function to set this value or perhaps include it as a parameter in the `i2c_set_data_timing` function

2. the `.tout` struct field appears to be a maximum of 20-bits, which yields a maximum value of 1048575 or 13.1ms...somewhat close for comfort. I'm not sure if this is a hardware or software limitation, but if it's software, it might be good to up this maximum value a little.

For reference, increasing this value _does_ allow the transaction to complete.

negativekelvin, 2017-10-24:
https://github.com/espressif/esp-idf/commit/01befe4d6a2e94bfc7db6484ff586470edea3086#diff-e0bba96861e55f5b2b25116c62fed912R484

arsinio, 2017-10-24:
Looks good...I'll try it out later this week to make sure!

closed 2017-11-22 by FayeY

arsinio, 2017-12-31:
Sorry it took me so long to get back on this...I went down a number of other rabbit holes in the mean time...

Anyways, I updated my repo to `v3.0-rc1` and this appears to be broken in a few ways:

**`i2c_set_timeout()` parameter checking:** 
The timeout parameter to this function is checked via:
`I2C_CHECK((timeout <= I2C_SDA_SAMPLE_TIME_V) && (timeout > 0), I2C_TIMEING_VAL_ERR_STR, ESP_ERR_INVALID_ARG);`

For my case, I need `I2C[i2c_num]->timeout.tout` to be `1048575` which is well outside the `I2C_SDA_SAMPLE_TIME_V` maximum value...but it's not just my particular case...even the defaut value returned by i2c_get_timeout() (6400) will return an error.

**`i2c_set_timeout()` doesn't seem to work:**
This may be an order of operations issue...by necessity we must call `i2c_set_timeout()` after `i2c_param_config()` since the later overwrites `I2C[i2c_num]->timeout.tout` with a default value. However, `i2c_set_timeout()` doesn't seem to have any actual affect...my reads still do not work correctly. I _have_ verified that `I2C[i2c_num]->timeout.tout` is set properly by a subsequent call to `i2c_get_timeout()`. I have also verified that if I modify the default `I2C[i2c_num]->timeout.tout` value set within `i2c_param_config()` my read _is_ successful.

Unfortunately I don't see any smoking guns here...any ideas?

negativekelvin, 2018-01-01:
https://github.com/espressif/esp-idf/commit/9075b507b54a99ce732fc2819ba17274d9a69ab5

negativekelvin, 2018-01-03:
Not sure. I2C[i2c_num]->timeout.tout should write directly to the register