espressif/esp-idf #1135: Clarification on BT GATTC MTU size

rojer, 2017-10-18:
`esp_ble_gattc_send_mtu_req` sets MTU per connection but does not actually take MTU size as parameter.
instead, it instructs us to use `esp_ble_gatt_set_local_mtu`, however, that one is global - does not take connection id.

since i expect to be talking to potentially different devices, requiring different MTU sizes, how can i use different MTU on different connections?

Weijian-Espressif, 2017-10-23:
@rojer esp_ble_gatt_set_local_mtu() set the local support maximum MTU value, you can set the maximum MTU size of slave, the connection will use the Minimum value of two .

rojer, 2017-10-23:
that's clear enough. the question is: how to set different MTUs for different connections?
i want the esp32 device to have 517 MTU (maximum possible), but not all the clients i connect to may support MTU this big.

negativekelvin, 2017-10-24:
@rojer I believe esp_ble_gattc_send_mtu_req will negotiate the MTU and automatically reduce it per connection if necessary

rojer, 2017-10-24:
no, what it does is it sends mtu request for the value set by `esp_ble_gatt_set_local_mtu`. which makes sense, except that it is global, so if i have to set different values for different connections... what then?
if `esp_ble_gatt_set_local_mtu` took connection id as parameter and only acted on that connection, it would make more sense.

negativekelvin, 2017-10-24:
It will send request for global size but if server will respond with lower value then mtu will be reduced.  But you are saying you want to self-limit mtu on different connections independent of what the connected device says?

Weijian-Espressif, 2017-10-24:
esp_ble_gatt_set_local_mtu only set the maximum MTU value supported by the device. The real use MTU size of every connection  is the minimum value that the two devices support.

for example:
one client  set  esp_ble_gatt_set_local_mtu(500)  , one server set esp_ble_gatt_set_local_mtu(300), The real MTU value is 300 after connection. Another server set esp_ble_gatt_set_local_mtu(27), The real MTU value is 27 after connection.

rojer, 2017-10-24:
ok, if that's how it works, then it's fine. thanks!

closed 2017-10-24 by rojer