espressif/esp-idf #1136: `make monitor` Error 1

sigwinch28, 2017-10-18:
After following the install guide for Windows on a Windows 10 64-bit machine, I receive an error when attempting to run `make monitor` after flashing the `hello_world`.

I installed the IDF via the "from scratch" method for Windows and the monitor was _previously working_, but I can't figure out what's changed:

```
$ make monitor
MONITOR
--- idf_monitor on COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (S
Traceback (most recent call last):
  File "C:/msys32/home/Joe/esp/esp-idf/tools/idf_monitor.py", line 568, in <module>
    main()
  File "C:/msys32/home/Joe/esp/esp-idf/tools/idf_monitor.py", line 502, in main
    monitor.main_loop()
  File "C:/msys32/home/Joe/esp/esp-idf/tools/idf_monitor.py", line 269, in main_loop
    self.handle_serial_input(data)
  File "C:/msys32/home/Joe/esp/esp-idf/tools/idf_monitor.py", line 300, in handle_serial_input
    self.console.write_bytes(b)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/tools/miniterm.py", line 63, in write_bytes
    self.byte_output.write(byte_string)
  File "C:/msys32/home/Joe/esp/esp-idf/tools/idf_monitor.py", line 560, in write
    self.output.write(b)
IOError: [Errno 0] Error
make: *** [/home/Joe/esp/esp-idf/components/esptool_py/Makefile.projbuild:96: monitor] Error 1
```

projectgus, 2017-10-19:
The IOError is a Windows-level OS error. I get similar errors if the USB connection is flaky, or the USB/Serial chip is browning out and resetting.

If the same board has previously worked in the same situation, try swapping out the USB cable.

sigwinch28, 2017-10-19:
Tried a different board, different USB cable, and different USB port. The problem still occurs with different boards, too.

adamt80, 2017-10-24:
Same issue here on Win 10 64bit.  Worked fine about a week ago with the same board, USB port, and cable. Now I get this error 100% of the time with any code for make monitor.  The serial monitor in Arduino IDE and Putty work fine for monitoring.

sigwinch28, 2017-10-25:
@adamt80 Have you done the Fall Creator's Update on that machine? My machines have the update.

adamt80, 2017-10-25:
@sigwinch28 Yes. Both my machines have the update as well. Probably related because I undid the last couple monitor changes and the issue persists.

logmanPL, 2017-11-01:
Windows update is definitely responsible for this. I have no issues just before it and after I have same error as you. Quick and ugly fix:
in idf_monitor.py line 560 change

```
					self.output.write(b)
					self.matched = b''
```

to 

```
					try:
						self.output.write(b)
						self.matched = b''
					except:
						pass
```

zachjacobs, 2017-11-02:
I came here to say that I had the same issue after installing the Fall Creator's Update and @logmanPL 's workaround allowed me to debug.  Thanks!

dconyers, 2017-11-09:
Just another quick confirmation that I encountered the issue immediately upon installing the Fall Creator's Update and @logmanPL 's post saved the day for me as well, thank you!

baoshi, 2017-11-18:
I have this issue too after Win10 update. simple_monitor gives me same error occasionally too. Wonder esp-idf is to blame or pyserial.

projectgus, 2017-11-20:
Hi folks,

Sorry for the slow turnaround on this. I (finally) got my Windows 10 box updated to Fall Creators Update and saw the exact same issue.

The bug is, oddly enough, in the console display output handling rather than anything to do with serial. It appears to be timing related, somehow.

Fix coming shortly (very similar to @logmanPL's above, only I noticed that retrying the write() always seems to succeed so doing this instead.)

projectgus, 2017-12-13:
> Fix coming shortly (very similar to @logmanPL's above, only I noticed that retrying the write() always seems to succeed so doing this instead.)

Seems like this is not always the case:
https://esp32.com/viewtopic.php?f=13&t=3920&p=17795#p17795

closed 2017-12-18 by igrr