espressif/esp-idf #1137: Getting Error in make menuconfig

mohinishnaik, 2017-10-19:
When I tried to run run the make menuconfig, I am getting following error

$ make menuconfig
C:/msys32/home/pod/esp/esp-idf/make/project.mk:53: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
make[1]: Entering directory '/home/pod/esp/esp-idf/tools/kconfig'
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
Cannot open output file 'zconf.hash.c'
make[1]: *** [Makefile:308: zconf.hash.c] Error 1
make[1]: Leaving directory '/home/pod/esp/esp-idf/tools/kconfig'
make[1]: Entering directory '/home/pod/esp/esp-idf/tools/kconfig'
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
Cannot open output file 'zconf.hash.c'
make[1]: *** [Makefile:308: zconf.hash.c] Error 1
make[1]: Leaving directory '/home/pod/esp/esp-idf/tools/kconfig'
make: *** [/home/pod/esp/esp-idf/make/project_config.mk:20: /home/pod/esp/esp-idf/tools/kconfig/mconf] Error 2

I had cloned  the esp-idf using command --> git clone --recursive https://github.com/espressif/esp-idf.git

Please try to help me........

projectgus, 2017-10-19:
Hi @mohinishnaik ,

The error at the top (`C:/msys32/home/pod/esp/esp-idf/make/project.mk:53: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.`) suggests that you didn't run mingw32.exe to create the terminal window for working ESP-IDF. Can you please check?

The full instructions can be found here:
http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html#check-it-out

Angus

mohinishnaik, 2017-10-19:
I have followed the esp-idf setup process using MINGW32.exe.

JetstreamRoySprowl, 2017-12-07:
@mohinishnaik I had the same problem and found I had somehow gotten routed to an old toolchain. I switched to the toolchain dated 2017-11-23 and the problem went away.

projectgus, 2019-03-01:
Closing due to lack of updates. @mohinishnaik , if this is still not working in recent ESP-IDF then please open a new issue with details and we'll chase it up again.

closed 2019-03-01 by projectgus