espressif/esp-idf #1139: Getting error in make menuconfig

lBenji, 2017-10-19:
I have done all steps from the get started page.

But I get an error in make menuconfig

Error:
`$ make menuconfig
make[1]: Verzeichnis „/home/benjamins/esp/esp-idf/tools/kconfig“ wird betreten
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o zconf.tab.o zconf.tab.c
In file included from zconf.tab.c:2450:0:
confdata.c: In Funktion »conf_get_default_confname«:
confdata.c:113:23: Fehler: »PATH_MAX« nicht deklariert (erste Benutzung in dieser Funktion)
  static char fullname[PATH_MAX+1];
                       ^~~~~~~~
confdata.c:113:23: Anmerkung: jeder nicht deklarierte Bezeichner wird nur einmal für jede Funktion, in der er vorkommt, gemeldet
confdata.c: In Funktion »conf_write«:
confdata.c:741:15: Fehler: »PATH_MAX« nicht deklariert (erste Benutzung in dieser Funktion)
  char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
               ^~~~~~~~
confdata.c: In Funktion »conf_split_config«:
confdata.c:837:12: Fehler: »PATH_MAX« nicht deklariert (erste Benutzung in dieser Funktion)
  char path[PATH_MAX+1];
            ^~~~~~~~
make[1]: *** [<eingebaut>: zconf.tab.o] Fehler 1
make[1]: Verzeichnis „/home/benjamins/esp/esp-idf/tools/kconfig“ wird verlassen
make[1]: Verzeichnis „/home/benjamins/esp/esp-idf/tools/kconfig“ wird betreten
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o zconf.tab.o zconf.tab.c
In file included from zconf.tab.c:2450:0:
confdata.c: In Funktion »conf_get_default_confname«:
confdata.c:113:23: Fehler: »PATH_MAX« nicht deklariert (erste Benutzung in dieser Funktion)
  static char fullname[PATH_MAX+1];
                       ^~~~~~~~
confdata.c:113:23: Anmerkung: jeder nicht deklarierte Bezeichner wird nur einmal für jede Funktion, in der er vorkommt, gemeldet
confdata.c: In Funktion »conf_write«:
confdata.c:741:15: Fehler: »PATH_MAX« nicht deklariert (erste Benutzung in dieser Funktion)
  char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
               ^~~~~~~~
confdata.c: In Funktion »conf_split_config«:
confdata.c:837:12: Fehler: »PATH_MAX« nicht deklariert (erste Benutzung in dieser Funktion)
  char path[PATH_MAX+1];
            ^~~~~~~~
make[1]: *** [<eingebaut>: zconf.tab.o] Fehler 1
make[1]: Verzeichnis „/home/benjamins/esp/esp-idf/tools/kconfig“ wird verlassen
make: *** [/home/benjamins/esp/esp-idf/make/project_config.mk:20: /home/benjamins/esp/esp-idf/tools/kconfig/mconf] Fehler 2
`

igrr, 2017-10-19:
Are you running OS X 10.13? If yes, then the issue has been already reported here: https://github.com/espressif/esp-idf/issues/712#issuecomment-334700604

lBenji, 2017-10-19:
Windows

projectgus, 2017-10-19:
Hi @lBenji ,

I have a couple more questions:

Did you download esp32_win32_msys2_environment_and_toolchain-20170918.zip or did you follow some of the alternative steps in Getting Started?

Did you launch the terminal you are using via mingw32.exe or via another method?

What version of Windows are you using?

Angus

lBenji, 2017-10-19:
@projectgus 

I have donwloaded the sp32_win32_msys2_environment_and_toolchain-20170918.zip
and I launch the terminal via mingw32.exe.

Windows Version = Win7 Professional

projectgus, 2017-11-15:
Hi @lBenji ,

Sorry for the delay in getting back to you. I'm not able to reproduce this here, PATH_MAX should be defined in the headers included with msys32.

If updating IDF to the current master and running `make clean` doesn't help, can you please try running the preprocessor and sending us what it outputs. Something like the following commands:

```
cd /home/benjamins/esp/esp-idf/tools/kconfig
cc -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD -E zconf.tab.c > zconf_tab_preproc.c
```

Then attach the output zconf_tab_preproc.c file here (or post it to gist.github.com or pastebin.com).

lBenji, 2017-11-15:
After a update make menuconfig is running successfully

closed 2017-11-15 by lBenji

tangqingcai, 2019-03-25:
> After a update make menuconfig is running successfully

IBenji,you say you update ESP-IDF source code,then make menuconfig is running successfully?
i met your similar error;
Could you tell me how to fix it ?Just update source code ?