espressif/esp-idf #1140: Watchdog triggers on IDLE task (because ieee80211_sta.c 661?)

CogPy, 2017-10-19:
Hi,

with the latest master (3a271a4a), we see the following watchdog-error after this WIFI error has triggered.

```shell
E (250084) wifi: ieee80211_sta.c 661

Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
```

The wifi error isn't repeated, but the watchdog keeps triggering.

liuzfesp, 2017-11-26:
Hi @CogPy could you let me know how to reproduce this issue?

CogPy, 2017-11-26:
Unfortunately, I don't have access to the HW anymore.

liuzfesp, 2018-01-15:
HI @CogPy, we can't reproduce this issue, do you still remember how to reproduce this issue?

CogPy, 2018-01-15:
Nope, given it was a development branch and problem went away with a different configuration/commit, we never bothered to follow up. I guess we can close the issue.

closed 2018-01-22 by FayeY