espressif/esp-idf #1141: Problem with sockets

karawin, 2017-10-19:
On SDK v3.0-dev-942-g2e8441df-dirty
Sockets are allocated from 4096. 
This break the FD_xxxx and far from FD_SETSIZE
So no select() can work.

Why this change?

igrr, 2017-10-19:
LwIP sockets start at `LWIP_SOCKET_OFFSET`, and `FD_SET`/`FD_GET` macros take this into account when operating on fd sets: `LWIP_SOCKET_OFFSET` is subtracted from fd number, and the resulting value as used as bit offset in fd set.

Previously, `LWIP_SOCKET_OFFSET` was always zero. With the latest changes, `LWIP_SOCKET_OFFSET` is defined to `lwip_socket_offset` global variable. When LwIP is initialized, the value of `lwip_socket_offset` is determined dynamically.

It is unlikely, but if you happen to do operations on fd sets before LwIP is initialized, you may get incorrect behavior. Otherwise, `lwip_socket_offset` should be taken into account correctly.

Perhaps you could share the code which you are having an issue with?

karawin, 2017-10-19:
The code is at https://github.com/karawin/Ka-Radio32
main/servers.c

It works well on esp8266 and worked well in esp32 before the last release.

How can i know if lwip_socket_offset is initialized?

karawin, 2017-10-19:
Something is wrong in FD_XXXX because
I (8999) servers: telnetServer_sock socket: 4096, errno: 0
I (9009) servers: Webserver socket: 4097, errno: 0
...
V (54889) servers: Activity 1, max_fd: 4097
V (54889) servers: Server web accept, socket: 4097.
V (54909) servers: telnetServer_sock accept. Socket: 4096
 Two FD_ISSET for one activity. And i am sure that the telnetServer_soc socket is not awaked.

karawin, 2017-10-19:
Succeeded with these modifications in lwip/sockets.h
Not clean. Please fix it.

/* FD_SET used for lwip_select */
//#ifndef FD_SET
#ifdef LWIP_SOCKET_OFFSET
#undef FD_SET
#undef FD_CLR
#undef FD_ISSET
#undef FD_ZERO
#undef _types_fd_set
#undef fd_set

projectgus, 2017-10-20:
Thanks for reporting this @karawin.

There was a second set of FD_SET/etc macros in newlib sys/types.h that were unexpectedly overriding the LWIP ones. This is why the workaround you have is working as well.

A fix to the newlib header is on the way, in the mean time the workaround you have will fix the problem.

karawin, 2017-10-20:
Great, Thanks

closed 2017-10-23 by igrr