espressif/esp-idf #1143: GAP / GATTS callback setting priority

bmetman, 2017-10-19:
Dear Espressif, 

I had a question concerning the functions **esp_ble_gap_register_callback** function found in **esp_gap_ble_api.h**, and the **esp_ble_gatts_register_callback** function found in **esp_gatts_api.h**. 

I'm trying to build FreeRTOS firmware using BLE. In this firmware I have created many tasks with different priorities, which are responsible for on-board components ranging from power switches to motors. In the ESP32 GATT server example callbacks are registered using the above functions, which are triggered when either GATT or GAP events are coming in. 

When I'm flooding the Bluetooth antenna with commands, this means I can block the device out of operation, as the priority of the callback is higher than any of my tasks. I would like to lower the priority of the callbacks. Is it possible to do this? Can I check which priority they have? My guess is that they have the highest priority, which is my other tasks are blocked. 

Thank you for your help.

Yulong-espressif, 2017-11-04:
@bmetman  The esp_ble_gap_register_callback and the esp_ble_gatts_register_callback run in the btc task, the priory of the btc task is define in the thread.h file with the BTC_TASK_PRIO macro, the vaule is (configMAX_PRIORITIES - 6). We do not agree with you to modify the priority of this task, if you change the priority of this task is lower than the task on the APP, there may be a lot of anomalies.

bmetman, 2017-11-06:
Thank you for your answer, it is clear to me what I need to do now. I will choose not to configure the priorities but rather de-register the callbacks when the time difference between incoming BLE commands gets too short. I probably won't start messing with the priorities as this can result in anomalies.

closed 2017-11-06 by bmetman