espressif/esp-idf #1144: EXTRA_LDFLAGS does not allow to add pre-compiled libraries

pimarco, 2017-10-20:
The EXTRA_LDFLAGS of the make/project.mk does not allow to add pre-compiled librairies, this can be fix easily by adding the following EXTRA_LIB_LDFLAGS in the file.

# Set default LDFLAGS
EXTRA_LDFLAGS ?=
EXTRA_LIB_LDFLAGS ?=
LDFLAGS ?= -nostdlib \
	-u call_user_start_cpu0	\
	$(EXTRA_LDFLAGS) \
	-Wl,--gc-sections	\
	-Wl,-static	\
	-Wl,--start-group	\
	$(COMPONENT_LDFLAGS) \
	$(EXTRA_LIB_LDFLAGS) \
	-lgcc \
	-lstdc++ \
	-lgcov \
	-Wl,--end-group \
	-Wl,-EL

projectgus, 2017-10-20:
Hi @pimarco ,

The recommended approach for this to add the specific LDFLAGS to a component, via the COMPONENT_ADD_LDFLAGS variable in a component.mk file. Then the flags will be added to the COMPONENT_LDFLAGS at project level, and these are passed in the right place.

If it's useful, this also allows you manage building a library from some external build process as part of the component.mk as well. For more details about this, see https://esp-idf.readthedocs.io/en/latest/api-guides/build-system.html#fully-overriding-the-component-makefile

FayeY, 2017-10-24:
Hi @pimarco , did the above suggestion help you resolve the issue?

pimarco, 2017-10-25:
Yes, I make the following modification in my Makefile to use a component.mk file in my $(BUILD_TOPDIR)/pkg-build:

PROJECT_NAME := esp-abc
BUILD_TOPDIR = $(subst /Makefile,,$(abspath $(realpath Makefile)))
EXTRA_COMPONENT_DIRS += $(BUILD_TOPDIR)/pkg-build
include $(IDF_PATH)/make/project.mk

My $(BUILD_TOPDIR)/pkg-build has an include and a lib directory, so my component.mk file look like this:

COMPONENT_ADD_INCLUDEDIRS := include
COMPONENT_ADD_LDFLAGS := -L$(PROJECT_PATH)/$(COMPONENT_NAME)/lib -labc

However, with my proposed EXTRA_LIB_LDFLAGS. I was having only that in my Makefile:

PROJECT_NAME := esp-abc
BUILD_TOPDIR = $(subst /Makefile,,$(abspath $(realpath Makefile)))
CFLAGS += -I$(BUILD_TOPDIR)/pkg-build/include
EXTRA_LIB_LDFLAGS = -L$(BUILD_TOPDIR)/pkg-build/lib -labc
include $(IDF_PATH)/make/project.mk

and my $(BUILD_TOPDIR)/pkg-build directory does not contain any specific file for ESP-IDF, which I think is more clean. So adding the EXTRA_LIB_LDFLAGS just give more possibilities to user, but I also  understand that you may have some other reasons to not add it (like updating the ESP-IDF build documentation).

Regards

closed 2017-11-08 by FayeY