espressif/esp-idf #1145: The iperf example has a problem, leading to the continuous reset of esp32

L-eeX, 2017-10-20:
The iperf example has a problem, leading to the continuous reset of esp32

L-eeX, 2017-10-20:
These lines of code cause the esp32 reset to restart：

ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_NULL) );
ESP_ERROR_CHECK( esp_wifi_start() );

projectgus, 2017-10-20:
Do you have an example of the log output from the continuous reset?

Usually a continuous reset while starting WiFi indicates a power stability problem with your hardware. If it's USB powered, try a different USB cable. If it's different hardware, check the power supply has enough capacitance.

igrr, 2017-10-20:
Also note that iperf example was missing initialization of NVS.
The fix for this will be on Github shortly. You can add `nvs_flash_init()` call before WiFi initialization in the meantime, similar to other WiFi examples.

L-eeX, 2017-10-20:
@igrr : All of the print is 0, I don't know how to set it

  87-  90 sec       0.00 Mbits/sec
  90-  93 sec       0.00 Mbits/sec
  93-  96 sec       0.00 Mbits/sec
  96-  99 sec       0.00 Mbits/sec
  99- 102 sec       0.00 Mbits/sec
 102- 105 sec       0.00 Mbits/sec
 105- 108 sec       0.00 Mbits/sec
 108- 111 sec       0.00 Mbits/sec
 111- 114 sec       0.00 Mbits/sec
 114- 117 sec       0.00 Mbits/sec

yrh79, 2017-10-21:
I've tried to update the code with  ESP_ERROR_CHECK( nvs_flash_init() ); included and the problem is fixed! @igrr  thanks a lot!!!

```
#include "esp_vfs_dev.h"
#include "nvs_flash.h" <--- add this

void app_main(void)
{
    ESP_ERROR_CHECK( nvs_flash_init() ); <<--- add this
    initialise_wifi();

```

yrh79, 2017-10-21:
```
iperf  [-sua] [-c <ip>] [-p <port>] [-i <interval>] [-t <time>]
  iperf command
  -c, --client=<ip>  run in client mode, connecting to <host>
  -s, --server  run in server mode
     -u, --udp  use UDP rather than TCP
  -p, --port=<port>  server port to listen on/connect to
  -i, --interval=<interval>  seconds between periodic bandwidth reports
  -t, --time=<time>  time in seconds to transmit for (default 10 secs)
   -a, --abort  abort running iperf

esp32> sta venus WontLetUKnow
I (254636) iperf: sta connecting to 'venus'
I (258769) event: sta ip: 192.168.1.136, mask: 255.255.255.0, gw: 192.168.1.1
esp32> query
I (266797) iperf: sta mode, connected venus
esp32> iperf -c 192.168.1.135 -p 4000
I (343542) iperf: mode=tcp-client sip=192.168.1.136:5001, dip=192.168.1.135:4000, interval=3, time=30
esp32> 
        Interval Bandwidth
   0-   3 sec       8.74 Mbits/sec
   3-   6 sec       10.22 Mbits/sec
   6-   9 sec       10.49 Mbits/sec
   9-  12 sec       10.84 Mbits/sec
  12-  15 sec       10.84 Mbits/sec
  15-  18 sec       10.92 Mbits/sec
  18-  21 sec       10.75 Mbits/sec
  21-  24 sec       10.14 Mbits/sec
  24-  27 sec       9.96 Mbits/sec
  27-  30 sec       10.22 Mbits/sec
   0-  30 sec       10.31 Mbits/sec
I (376340) iperf: iperf exit
esp32> 
```

liuzfesp, 2017-10-30:
The issue is fixed by commit ID: ae1ac2ab0e0d6a46e1f0592f7b27f5bafbffbd4c, the latest idf already include it.

closed 2017-10-31 by FayeY