espressif/esp-idf #1149: I2S: No clock generated for PDM Microphone

akofoed, 2017-10-21:
I have a PDM microphone connected to the ESP32 and using the following config. I want to use the I2S PDM function to get the ESP32 to read the PDM microphone and convert the PDM to PCM.

I can see on my logic analyser that the ESP32 is not generating any clock signal:
```
i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM,
        .sample_rate = SAMPLE_RATE,
        .bits_per_sample = BITS_PER_SAMPLE,
        //.channel_format = I2S_CHANNEL_FMT_ALL_RIGHT,
        //.communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB,
        .dma_buf_count = 8,
        .dma_buf_len = 64,
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1
    };
    i2s_pin_config_t pin_config = {
        .bck_io_num = 25,
        .ws_io_num = I2S_PIN_NO_CHANGE,
        .data_out_num = I2S_PIN_NO_CHANGE, // Not used
        .data_in_num = 26
    };
    printf("Installing driver...\n");
    i2s_driver_install(I2S_NUM, &i2s_config, 0, NULL);
    i2s_set_pin(I2S_NUM, &pin_config);
...
    samples_read = i2s_read_bytes(I2S_NUM, samples_data, SAMPLE_LENTGH_BYTES, 1000);
```
If I exclude the I2S_MODE_PDM from the .mode, it does generate a clock signal. But as I read the documentation, the ESP32 should be able to clock and read the PDM mic and convert it to PCM?

Hesquibet, 2017-11-27:
Hey akofoed,
I'm running into the same issue, could you let me know if you found a solution ?
Thanks

akofoed, 2017-11-27:
Nope - no solution. Guessing it is an issue with the driver (SDK).

kewalmshah, 2017-12-14:
Hello @akofoed PDM works only on I2S_PORT_0, as mentioned in TRM in section 11.1, page 233. Could you please help to confirm if you are using i2s port 0 ?

FayeY, 2017-12-21:
Hi @akofoed , did the above suggestion help you resolve the issue? Please Note: PDM is supported on I2S PORT 0 only. Also the clk is generated on WS line only.  Thanks.

akofoed, 2017-12-21:
Sorry, I have not yet have time to test it. I will write again when I have.
Thanks

FayeY, 2017-12-21:
Got it. Please feel free to reopen it, if this is still an issue after testing. Thanks for your interest in ESP32!

closed 2017-12-21 by FayeY

FlorisColpaert, 2018-03-19:
Incorrect sample data in PCM.. When there is no sound I get a value aroud 976. When there is sound I detect data change but way more data in the positive value than negative. Is there anything wrong?

 i2s_config_t i2s_config =
  {
     .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM),
     .sample_rate = 44100,
     .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
     .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,
     .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_PCM ),
     .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, // lowest interrupt priority
     .dma_buf_count = 8,
     .dma_buf_len = 68,
     .use_apll = 0
  };

  // // Clock is internal // // No data out
  i2s_pin_config_t pinConfig = {
     .bck_io_num = I2S_PIN_NO_CHANGE,
     .ws_io_num = clkOut->GetPin(), // WS is connected to clock for external clock
     .data_out_num = I2S_PIN_NO_CHANGE,
     .data_in_num = pinRx->GetPin() // Data input of mic
  };

  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);  //install and start i2s driver
  i2s_set_pin(I2S_NUM_0, &pinConfig);

.....

while ((read_count_ < 8000))
  {
    if ( (i2s_pop_sample(I2S_NUM_0, (char *)&sample, (TickType_t)portMAX_DELAY)) >= 2)
    {
      // sample = ((sampleBfr[1] << 8 & 0xFF00) + (sampleBfr[0] & 0x00ff));
      rms_ += pow((double)sample/32767, 2);
      read_count_++;
    }
  }

  if( read_count_ >= 8000)
  {
    double db;
    if (rms_ != 0){
      db = 20*log10(sqrt((double)rms_/(read_count_)));
    }
    else
      db = 0;

    successHandler((float)db);

  }else{
    failHandler();
  }