espressif/esp-idf #1151: libnet80211.a build error：undefined reference to `wpa2_machine_start' and 'wps_start'

YanMinge, 2017-10-22:
log:

G:/project/micropython/myan/esp32_pro/esp-idf/components/esp32/lib\libnet80211.a(wl_cnx.o):(.text.cnx_do_handoff+0x0): undefined reference to `wpa2_machine_start'
G:/project/micropython/myan/esp32_pro/esp-idf/components/esp32/lib\libnet80211.a(wl_cnx.o):(.text.cnx_do_handoff+0x4): undefined reference to `wps_start'
make: *** [GNUMakefile:944：build/application.elf] 错误 1

projectgus, 2017-10-23:
You probably have out of date WiFi libraries. Run "git submodule update --init --recursive".

There is probably a warning about this in the IDF build logs if you are using IDF to build the project.

projectgus, 2017-10-23:
Closing as a duplicate of #1099. If you still get errors then please post on that issue.

closed 2017-10-23 by projectgus

YanMinge, 2017-10-23:
I'm sure I have done the command "git submodule update --init --recursive", and then I post the issue.
please try it.