espressif/esp-idf #1152: I cant compile hello_world example

efiorello, 2017-10-22:
I cant compile hello_world example.
I followed the entire installation guide but I can not seem to fix this error. can anybody help me?

`eze@samsung:~/esp/hello_world$ make flash
CC build/bootloader/bootloader_support/src/bootloader_flash.o
as: unrecognized option '--longcalls'
/home/eze/esp/esp-idf/make/component_wrapper.mk:242: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 1
/home/eze/esp/esp-idf/make/project.mk:434: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/home/eze/esp/esp-idf/components/bootloader/Makefile.projbuild:40: recipe for target '/home/eze/esp/hello_world/build/bootloader/bootloader.bin' failed
make: *** [/home/eze/esp/hello_world/build/bootloader/bootloader.bin] Error 2
`

projectgus, 2017-10-23:
> as: unrecognized option '--longcalls'

This suggests the toolchain version is somehow wrong. Can you post the output of the following two commands please (you may want to use a service like pastebin.com or gist.github.com for the second one):

```
xtensa-esp32-elf-gcc -v
make V=1
```

efiorello, 2017-10-23:
Sure:
https://gist.github.com/efiorello/3b2b532eb361df6a86b166d093965ec0
https://gist.github.com/efiorello/2c01f4054128ab93c826fbfe7f91c1b0

efiorello, 2017-10-24:
I delete all my esp directory and start again. was solved

closed 2017-10-24 by efiorello