espressif/esp-idf #1153: SPI SLAVE DEMO ：There are data errors in transmission

snowQY, 2017-10-23:
because I will use esp32 spi slave receive arm's datas ,so I use the official examples  in esp/esp-idf/examples/peripherals/spi_slave , there are two demos ,one sender ,another receiver .so I use two esp32 to test the demo ,and I have change nothing  about .c file. and I make sure the pin is connected  rightly  accord  to README.md . But in transmission ,the data always has error . the data seems can't be received  completely ,eg : the send data is ,{0x01,0x02,0x03,0x04,0x05,0x06,0x07,0x08} , the receive data  sometimes is ok (the start time always is ok),but in most times the receive data maybe is {0x01,0x02,0x03,0,0,0,0,0,0} (some time late ,the error occur), I guess maybe the timing has problem. so how to change the .c file to solve the problem?

the send  codes:
/* SPI Slave example, sender (uses SPI master driver)

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"
#include "lwip/igmp.h"

#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "soc/rtc_cntl_reg.h"
#include "rom/cache.h"
#include "driver/spi_master.h"
#include "esp_log.h"
#include "esp_spi_flash.h"

#include "soc/gpio_reg.h"
#include "driver/gpio.h"
#include "esp_intr_alloc.h"

/*
SPI sender (master) example.

This example is supposed to work together with the SPI receiver. It uses the standard SPI pins (MISO, MOSI, SCLK, CS) to 
transmit data over in a full-duplex fashion, that is, while the master puts data on the MOSI pin, the slave puts its own
data on the MISO pin.

This example uses one extra pin: GPIO_HANDSHAKE is used as a handshake pin. The slave makes this pin high as soon as it is
ready to receive/send data. This code connects this line to a GPIO interrupt which gives the rdySem semaphore. The main 
task waits for this semaphore to be given before queueing a transmission.
*/

/*
Pins in use. The SPI Master can use the GPIO mux, so feel free to change these if needed.
*/
#define GPIO_HANDSHAKE 2
#define GPIO_MOSI 12
#define GPIO_MISO 13
#define GPIO_SCLK 15
#define GPIO_CS 14

//The semaphore indicating the slave is ready to receive stuff.
static xQueueHandle rdySem;

/*
This ISR is called when the handshake line goes high.
*/
static void IRAM_ATTR gpio_handshake_isr_handler(void* arg)
{
[... 205 lines cut ...]

    while(1) {
        //Clear receive buffer, set send buffer to something sane
        memset(recvbuf, 0xA5, 129);
        sprintf(sendbuf, "This is the receiver, sending data for transmission number %04d.", n);

        //Set up a transaction of 128 bytes to send/receive
        t.length=128*8;
        t.tx_buffer=sendbuf;
        t.rx_buffer=recvbuf;
        /* This call enables the SPI slave interface to send/receive to the sendbuf and recvbuf. The transaction is
        initialized by the SPI master, however, so it will not actually happen until the master starts a hardware transaction
        by pulling CS low and pulsing the clock etc. In this specific example, we use the handshake line, pulled up by the
        .post_setup_cb callback that is called as soon as a transaction is ready, to let the master know it is free to transfer
        data.
        */
        ret=spi_slave_transmit(HSPI_HOST, &t, portMAX_DELAY);

        //spi_slave_transmit does not return until the master has done a transmission, so by here we have sent our data and
        //received data from the master. Print it.
        printf("Received: %s\n", recvbuf);
        n++;
    }

}

ESP-Coco, 2017-10-23:
Hi, the clock speed of this example is 5MHz. To avoid signal loss,please solder the SPI pins (MISO, MOSI, SCLK, CS) of the master and the slave to each other using shorter cables. Use standard Dupont lines for transmitting handshake signals.

snowQY, 2017-10-24:
Hello,now it is ok after I changed the mode 0 to mode 2,but when I want to increase the clock speed ,there is a new problem, I found the maximum speed is 10Mhz ,if   clock>10MHZ and clock<25Mhz  the receiver data will left shift one bit in full ,and if clock is larger than 25 Mhz,the receiver will not get anything. SO ,I want to know what is the actual maximum clock frequency  for esp32 spi? It is very important to me , you know, the clock means bandwidth. If I want to use a higher SCLK speed , what should I set the set the master and the slave?

ESP-Coco, 2017-10-24:
Hi @snowQY , we have not tested the actual maximum clock frequency. The actual maximum clock is related to the circuit diagram design. Because the theoretical maximum SCLK speed is up to 80Mhz, optimize the wiring between master and slave will help you get a higher speed.

FayeY, 2017-10-28:
Hi @snowQY , did the above suggestion help you resolve the issue?

snowQY, 2017-11-01:
i am so sorry  to reply  lately , beacause  i am so busy in a few days ago  , I tested maximum clock is 10 Mhz with esp32-wroom-32 board at present .But the esp32 will work on our pcb board in the end , and i have not test it bacause i need to design the codes for  fpga and arm in the first . I will ask for support if i test our board  that has error occur. In fact , if it can reach 10Mhz  really ,it is ok too. But the faster ,the better . Thank you very much  for your advices.

closed 2017-11-02 by FayeY

hpwit, 2018-02-13:
hello
I don't seem to be able to go faster than 5Mhz whatver the clock speed
when i put
getClycleCount() to measure the time of ret=spi_device_transmit(handle, &t);
do you have anyidea why ?
thank you

MasterLivens, 2022-11-15:
I got this issue too. I tried ESP32, ESP32-S2, ESP32-C3 at 5Mhz. And  ESP32-C3  is good for all tests, with no data errors.