espressif/esp-idf #1154: After esp_restart chip won't restart

lollizze, 2017-10-23:
We are running tests on ESP32 (IDF v2.1) and after a while when we do a `esp_restart` the module never restarts unless we do a HW reset. It is hard to reproduce but we have made some observation. The tests where it happens after a while are usually short, doing something then restarting the module, some tests are writing to one label in nvs and every test are erasing that label just before the restart. 

What we have tested:
1. Toggling some pins in `esp_restart_noos` procedure it seems like it never gets to the end (seems to hang somewhere around `esp_dport_access_int_deinit`).
2. Adding a delay between `esp_cpu_stall` and `esp_dport_access_int_deinit` made it take longer before it happened, so long we thought it was fixed but then we got a problem with tests that used wifi, see point 3
3. Doing a `esp_wifi_stop` in the before calling `esp_restart` made it take longer before it happened, made us think the delay and this where the solution until it happened again
4. The power consumption is normal when it happens
5. The problem isn't completely solved by doing a HW reset, when it has booted up after HW reset we can reproduce the problem quicker. Only a complete wipe of flash restores the devices to normal.
6. Swapped out the chip for new ones, problem still exists.

igrr, 2017-10-26:
Looking into similar reports for latest git version. There are three related issues here, i am not sure which one you are hitting:

1. `esp_dport_access_int_deinit` will wait on a critical section. Depending on what the other CPU is doing, this wait may never complete. In master branch there is a new function, `esp_dport_access_int_abort`, which will not wait for critical section and will just disable DPORT register access protection.

2. There is a small possibility that when `esp_cpu_stall(other_core_id);` is called, deadlock will happen. Other CPU may become stalled at the time when there is a D(or I)-bus transaction active. This may prevent current CPU from accessing the same bus, effectively stopping the program execution. The current solution (which we are still verifying) is to reset the other CPU before stalling it, e.g.

```c++
    SET_PERI_REG_MASK(RTC_CNTL_OPTIONS0_REG,
            other_core_id == 0 ? RTC_CNTL_SW_PROCPU_RST_M : RTC_CNTL_SW_APPCPU_RST_M);
```

3. Finally, there is an issue that `rtc_clk_cpu_freq_set` may disable the PLL before the switch from PLL to XTAL clock has completed. We're running more tests to be sure we choose the right value this time, but you may try increasing `DELAY_CPU_FREQ_SWITCH_TO_XTAL_WITH_150K` value from 20 to 40 (that's in rtc_clk.c). It is unlikely that you are hitting this case though, as by the time when `rtc_clk_cpu_freq_set` is called, RTC watchdog is already enabled.

Finally, moving the RTC watchdog init procedure to the top of esp_restart_nonos will help recover the device if issues 1 or 2 happen. (This is also something we are testing now.)

lollizze, 2017-10-26:
Thanks @igrr !! 
We have confirmed that at least number one is happening, we also notice that configuring our flash with 3.3V instead of 1.8V makes it easier to reproduce.
Is it safe to call `esp_dport_access_int_abort` instead of` esp_dport_access_int_deinit` in `esp_restart_noos` and ignore where the other CPU is stalled

igrr, 2017-10-30:
Sorry, i have misread your last sentence, it was in fact a question.

Yes, it is safe to call `esp_dport_access_int_abort` provided that the other CPU is stalled. As mentioned in point 2, before stalling the other CPU you have to use the `SET_PERI_REG_MASK...` bit from my previous comment before stalling the other CPU, to make sure that the other CPU is not in the middle of s32ci1 instruction.

lollizze, 2017-10-30:
@igrr no worry apparently i forgot the question mark

closed 2017-11-14 by FayeY