espressif/esp-idf #1156: Panic in uart tx ringbuffer

lollizze, 2017-10-23:
We have a uart with flow control at default baudrate, if the driver is under heavy load in both direction eventually we get a panic in ringbuf, we have seen three different places for the assert but all from tx ringbuffer
idf v2.1

**esp-idf/components/freertos/ringbuf.c:345 configASSERT((hdr->flags & iflag_free)==0);**
*0  0x40087278 in invoke_abort ()
    at esp-idf/components/esp32/panic.c:125
*1  0x40087346 in abort ()
    at esp-idf/components/esp32/panic.c:134
*2  0x4008689a in returnItemToRingbufDefault (rb=0x3ffdb158, item=0x3fff50a8)
    at esp-idf/components/freertos/ringbuf.c:345
*3  0x40086efa in vRingbufferReturnItemFromISR (ringbuf=0x3ffdb158,
    item=0x3fff50a8, higher_prio_task_awoken=0x3ffb0bc8)
    at esp-idf/components/freertos/ringbuf.c:623
*4  0x400fe37d in uart_rx_intr_handler_default (param=0x3ffdaf0c)
    at esp-idf/components/driver/uart.c:543
*5  0x40081919 in _xt_lowint1 ()

**esp-idf/components/freertos/ringbuf.c:272 configASSERT((hdr->len < rb->size) || (hdr->flags & iflag_dummydata));**
*0  0x40087288 in invoke_abort ()
    esp-idf/components/esp32/panic.c:125
*1   0x40087288 in invoke_abort ()
    esp-idf/components/esp32/panic.c:125
*2   0x4008696a in getItemFromRingbufDefault (rb=0x3ffdb1a8, length=0x3ffb0bcc,
    wanted_length=0)
    at esp-idf/components/freertos/ringbuf.c:272
*3   0x40086ec3 in xRingbufferReceiveFromISR (ringbuf=0x3ffdb1a8,
    item_size=0x3ffb0bcc)
    at esp-idf/components/freertos/ringbuf.c:582
*4  0x400fe3ad in uart_rx_intr_handler_default (param=0x3ffdaf5c)
    at esp-idf/components/driver/uart.c:530
*5  0x40081919 in _xt_lowint1 ()

**esp-idf/components/freertosingbuf.c:272 (getItemFromRingbufDefault)- assert failed!**

0x40086972: getItemFromRingbufDefault at esp-idf/components/freertos/ringbuf.c
0x40086f2f: vRingbufferReturnItemFromISR at esp-idf/components/freertos/ringbuf.c
0x400fe3aa: uart_rx_intr_handler_default at esp-idf/components/driver/uart.c:919
0x40081916: _xt_lowint1 at xtensa_vectors.o:?

koobest, 2017-11-06:
Hi, lollizze, I want to reproduce the problems you encountered. can you provide more details , such as test code?

lollizze, 2017-11-06:
Hi, here is how i configure the uart
```
uart_config_t config;
config->baud_rate = 115200,
config->data_bits = UART_DATA_8_BITS;
config->parity = UART_PARITY_DISABLE;
config->stop_bits = UART_STOP_BITS_1;
config->flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS;
config->rx_flow_ctrl_thresh = 100;
espStatus = uart_param_config(0, &uart_config);
espStatus = uart_set_pin(0, GPIO_NUM_1, GPIO_NUM_3, GPIO_NUM_22, GPIO_NUM_19);
uart_driver_install(0,  2048 * 2,  0, 50, uartIsrEventQueue, 0);
```
the tx task
```
void uartTxTask()
{
    QMessage msg;
    cbUART_LOG("%s %s\n", __FUNCTION__, " task started");
    while (1) 
    {        
        if ( xQueueReceive(txQueue, &msg, (portTickType) portMAX_DELAY) )
        {
            uart_write_bytes(0, (cb_char*)msg.buf, msg.size);
        }
    }
    vTaskDelete(NULL);
}
```

we found this when sending/receiving tcp data

koobest, 2017-11-08:
Hi, lollizze,I Wrote a simple test code and used tcp too, but didn't reproduce your problem. What makes me curious is why this error occurs because you did not use rx_ringbuffer in uart, it mabe a driver's error.  Is your program want to achieve TCP to UART and where is the definition of  uartIsrEventQueue？

lollizze, 2017-11-08:
@koobest  yes you are right, i took the wrong version. When experience this i had 2048 as tx ringbuffer size,

Here is more info of how the data is sent
 * computer sends 1024 bytes at a time 
 * Other side sends tcp data of size 512
 1. module reads all available data from rx ringbuffer but max 2048 bytes
 2. chunks it up into max 512 bytes and sends it on tcp
 3. repeat 2

koobest, 2017-11-08:
So, is the wrong version of IDF that caused this problem and will this problem appear now if use a new version of IDF?

lollizze, 2017-11-08:
This is from the latest release so I wouldn't say it is the wrong version of the IDF

uartIsrEventQueue is of type QueueHandle_t

InfiniteYuan1, 2017-11-09:
@lollizze  hi  lollizze , It is a bit difficult for us to reproduce this problem. If you provide your test code, perhaps this problem is easy to reproduce and quickly resolved.

lollizze, 2017-11-09:
I don't have a small reproducibly code only my whole application

FayeY, 2018-01-08:
Closing as can't reproduce. Please reopen if the issue persists in the latest IDF and you can supply more details. Thanks.

closed 2018-01-08 by FayeY