espressif/esp-idf #1157: nvs_get_str() seams to causes lock (reproducable on arduino-esp32)

Curclamas, 2017-10-23:
## Description
This error is reproducible on  custom boards as well as ESP32-DevBoard:
When nvs_get_str() is called and there is another task that runs on higher priority, the ESP32 can inadvertently lock up.

## Reproduction
The error is described and reproduced in arduino-esp32-issue: https://github.com/espressif/arduino-esp32/issues/740
It seams to be rooted in IDF though.

igrr, 2017-10-23:
I think this was fixed in https://github.com/espressif/esp-idf/commit/b013f5d49066249ecb9aebaf250fdd83d996998b, but I am not sure which IDF commit you are using. Can you please mention the commit ID?

Curclamas, 2017-10-24:
@igrr Now I've tried it with the latest IDF from arduino-esp32 which is 9274814 (should be after b013f5d). The device still freezes...

Curclamas, 2017-10-24:
One additional piece of Information: after several crashes it appears that the ESP32 does not boot. Only the following can than be seen in the log-output:
```
preferences.getString:  <<<---- Here it crashes
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, ⸮  <<<---- Nothing below here
```

igrr, 2017-10-30:
Could you please try the attached patch?

Based on my testing, this solves the issue you have found. There are still periods of time when loop_task doesn't run because coreTask is taking all the CPU time (for a few seconds), but there is no more deadlock, eventually TLS connection fails in coreTask, and loop_task runs again.

```patch
From 0e65e67480b5f5a6f452aa785a6ec64764a78866 Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Mon, 30 Oct 2017 18:53:39 +0800
Subject: [PATCH] spi_flash: raise priority of the task performing spi_flash
 operation

Fixes https://github.com/espressif/arduino-esp32/issues/740
---
 components/spi_flash/cache_utils.c | 7 +++++++
 1 file changed, 7 insertions(+)

diff --git a/components/spi_flash/cache_utils.c b/components/spi_flash/cache_utils.c
index 53a0caf8e..cc1f56929 100644
--- a/components/spi_flash/cache_utils.c
+++ b/components/spi_flash/cache_utils.c
@@ -110,6 +110,11 @@ void IRAM_ATTR spi_flash_disable_interrupts_caches_and_other_cpu()
         assert(other_cpuid == 1);
         spi_flash_disable_cache(other_cpuid, &s_flash_op_cache_state[other_cpuid]);
     } else {
+        // Temporarily raise current task priority to prevent a deadlock while
+        // waiting for IPC task to start on the other CPU
+        TaskHandle_t self = xTaskGetCurrentTaskHandle();
+        int old_prio = uxTaskPriorityGet(self);
+        vTaskPrioritySet(self, configMAX_PRIORITIES - 1);
         // Signal to the spi_flash_op_block_task on the other CPU that we need it to
         // disable cache there and block other tasks from executing.
         s_flash_op_can_start = false;
@@ -121,6 +126,8 @@ void IRAM_ATTR spi_flash_disable_interrupts_caches_and_other_cpu()
         }
         // Disable scheduler on the current CPU
         vTaskSuspendAll();
+        // Can now set the priority back to the normal one
+        vTaskPrioritySet(self, old_prio);
         // This is guaranteed to run on CPU <cpuid> because the other CPU is now
         // occupied by highest priority task
         assert(xPortGetCoreID() == cpuid);
-- 
2.13.5 (Apple Git-94)

```

Curclamas, 2017-10-30:
@igrr after this patch the board no longer freezes. But it seams that the HTTPS-Task eventually ceases to run, I have yet to find out where.

igrr, 2017-12-08:
The scheduler needs to be disabled to prevent switching to other tasks during flash operations. Other tasks may be (and probably are) running code from cache, but the cache is disabled during flash operation. So a task switch during flash operation would likely cause an immediate crash.

Effectively, we want the task doing the flash operation to be temporary elevated to highest priority. There is no "normal" way to do this in FreeRTOS, using locks and critical sections only.

Curclamas, 2017-12-08:
Thanks for the insights Ivan!

That patch you've posted workes fine now in production for weeks now. From
my side it's fine. If you have no objections I'd love to see that fix
upstream!

Curclamas, 2018-06-22:
@igrr While comparing my branch with master: this still does not seam to be upstream right? 
Any news on this issue, because I'd love to keep the differences between my branch and master small. Especially since I want to move to a long time support / release version soon.

igrr, 2018-06-25:
Sorry for the delay @Curclamas, I've pushed this patch for review now. Should be in IDF master branch soon.

Curclamas, 2018-06-25:
Thank you very much!

closed 2018-07-04 by igrr

kealist, 2018-08-02:
I know I am resurrecting this, but I don't see this change propageted into esp-idf and I am experiencing the same issue with the above fix resolving the issue.