espressif/esp-idf #1159: Spiffs crash

llewellynnu, 2017-10-25:
Hi, 
After mounting the spiffs I run the following command and the code crashes with the error trace shown below

stat("/spiffs/swap.txt", &st) == 0)

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x401252c4  PS      : 0x00060034  A0      : 0x400828c0  A1      : 0x3ffb05c0  
0x401252c4: __floatundidf at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2177

0x400828c0: _xt_lowint1 at /Users/llewellyn/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

A2      : 0x00000001  A3      : 0x3ff5f024  A4      : 0x00000000  A5      : 0x0000000e  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x80082fe0  A9      : 0x3ffb327c  
A10     : 0x0000000e  A11     : 0x00000000  A12     : 0x8008fa49  A13     : 0x3ffca2a0  
A14     : 0x00000000  A15     : 0x3ffca2d0  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x401252c4:0x3ffb05c0 0x400828bd:0x3ffb0610 0x4008fb45:0x00000000
0x401252c4: __floatundidf at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2177

0x400828bd: _xt_lowint1 at /Users/llewellyn/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

0x4008fb45: esp_rom_spiflash_read_data at /Users/llewellyn/esp/esp-idf/components/spi_flash/./spi_flash_rom_patch.c:412

I have disabled the cache for spiffs and tried it with no luck. 

Regards

igrr, 2017-10-25:
Does the application only work with SPIFFS, or are there any other
components in use? Interrupt handlers, perhaps?

llewellynnu, 2017-10-25:
The applications has, 

WIFI
TIMER(Timer ISR)
ADC

Regards

igrr, 2017-10-25:
Please post the line which you use to initialize TIMER ISR.
Thanks.

llewellynnu, 2017-10-25:
hi @igrr 
timer_isr_register(timer_group, timer_idx, Timer1ISR, (void*) timer_idx,
			ESP_INTR_FLAG_IRAM, NULL);

The above line is used to register the Timer ISR

The timer ISR is declared as below
void IRAM_ATTR Timer0ISR(void *para)

Regards

igrr, 2017-10-25:
The issue is that the ISR is placed into IRAM (ESP_INTR_FLAG_IRAM), but it
does some floating point manipulation which uses a function not placed into
IRAM (__floatundidf). If you remove ESP_INTR_FLAG_IRAM flag, then
everything should work fine.

llewellynnu, 2017-10-25:
@igrr  

Tested it out. Works perfectly now.

Regards

closed 2017-10-27 by igrr