espressif/esp-idf #1160: How to checkout ESP-IDF's commit with specified submodules' commits

Gfast2, 2017-10-25:
Hi ESP-IDF,

I love my ESP32. Right now I did a lot development here. But I did encounter a practical problem.

Along with the active development of this framework. Many header files, API are deprecated. When I checkout someone's example that a little bit old (say, [***this one*** (from pcbreflux)](https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_freertos_eventgroup)). I just can't without checking out the older version of ESP-IDF repo in order to let the example compiled.
Right now I can do the clean up job without any extra effort. But if I would like to check out a really really huge & old repo in the near futur, which needs not only check out the older ESP-IDF main repo, but also check out many of its `git submodule` to their specified commits. What should I do?
Will the main git repository's commits record (say in file `.gitsubmodule`) which commits are its submodules using automatically? Or I've gethering all check points of all `submodule` and write them down & check in for the problem.

I did investigate sometime on the Internet. 🍡  But I think there is still not good answers out there.

jakkra, 2017-10-25:
`git submodule update --recursive` will update all sbmodules to the correct commit after you checkout out the main repo.

Gfast2, 2017-10-25:
Hi @jakkra ,

Thanks für your answer! Super cool. Since I don't really sure about this.
Right now I've tested your answer:
```
gfast2@gfast2-VirtualBox:~/esp/esp-idf$ git submodule update --recursive
Submodule path 'components/bt/lib': checked out '9a4bb1d5287572664f170f9df4dbfd71babdfc68'
Submodule path 'components/esp32/lib': checked out 'e8743355806ed3671f7b0ef8bac1ea63909daaf3'
Submodule path 'components/esptool_py/esptool': checked out 'e9e9179f6fc3f2ecfc568987d3224b5e53a05f06'

```

It's just ***PURE MAGIC*** 🌟 

Thanks!

closed 2017-10-25 by Gfast2