espressif/esp-idf #1161: SPIFFS format triggers WDT

forthlightning, 2017-10-25:
Running esp-idf with arduino as a component. 

When I format with `esp_err_t ret = esp_vfs_spiffs_register( &conf );` The idle task triggers the watchdog due to being starved of cycles. 

If I add the following code at line 416 in spiffs_nucleus.c, along with initializing `static int count = 0' at 364

```
    count++;
    if( count == 20 )
    {
      pet_the_dog(); // just a 'vTaskDelay(50)'
      count = 0;
    }
```

The idle task WDT reset goes away and I can format SPIFFS no problem. This feels like a weird way to do it, but this is something I had to add in the old SPIFFS library ripped from the lua-RTOS project. Is there a better way to workaround?

luc-github, 2017-10-26:
Uncheck this setting (suggested by @me-no-dev)  solve issue also : 
[image]

forthlightning, 2017-10-26:
I have seen that that setting also fixes the problem. It seems like a risky thing to do, disabling safeguards? Is it possible to suspend the scheduler or suspend the idle task watchdog for some period of seconds?

forthlightning, 2017-10-27:
from an administrative point of view it is much easier to just change the idle task wdt setting. Is there a reliability impact in this situation?

krzychb, 2018-12-10:
I see the same issue on master with plain `examples/storage/spiffs` example:

```
I (238) cpu_start: Pro cpu up.
I (241) cpu_start: Application information:
I (246) cpu_start: Project name:     spiffs
I (251) cpu_start: App version:      v3.3-beta1-2-g19cf1bb7d-dirty
I (258) cpu_start: Compile time:     20:00:41
I (263) cpu_start: Compile date:     Dec 10 2018
I (268) cpu_start: ESP-IDF:          v3.3-beta1-2-g19cf1bb7d-dirty
I (275) cpu_start: Starting app cpu, entry point is 0x40081068
0x40081068: call_start_cpu1 at /home/krzysztof/esp/esp-idf/components/esp32/cpu_start.c:261

I (0) cpu_start: App cpu up.
I (286) heap_init: Initializing. RAM available for dynamic allocation:
I (293) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (299) heap_init: At 3FFB2EE0 len 0002D120 (180 KiB): DRAM
I (305) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (311) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (318) heap_init: At 400894A4 len 00016B5C (90 KiB): IRAM
I (324) cpu_start: Pro cpu start user code
I (6) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (8) example: Initializing SPIFFS
W (8) SPIFFS: mount failed, -10025. formatting...
E (4958) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (4958) task_wdt:  - IDLE0 (CPU 0)
E (4958) task_wdt: Tasks currently running:
E (4958) task_wdt: CPU 0: main
E (4958) task_wdt: CPU 1: IDLE1
E (9938) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (9938) task_wdt:  - IDLE0 (CPU 0)
E (9938) task_wdt: Tasks currently running:
E (9938) task_wdt: CPU 0: main
E (9938) task_wdt: CPU 1: IDLE1
I (11138) example: Partition size: total: 896321, used: 0
I (11138) example: Opening file
I (11218) example: File written
I (11268) example: Renaming file
I (11308) example: Reading file
I (11308) example: Read from file: 'Hello World!'
I (11308) example: SPIFFS unmounted
```
The WDT is triggered during formatting that takes couple of seconds and gone after that. Tested with SanDisk 2GB microSD card.

nopnop2002, 2019-10-03:
It does not occur by disabling [Watch CPU0 Idle Tak] in my environment
My environment:ESP-IDF version:v4.1-dev-369-g4dac7c7df

[image]

[image]

But it may not be the fundamental solution

igrr, 2020-09-22:
This has been fixed in https://github.com/espressif/esp-idf/commit/ae89f98d59c3391cf68cf4678e6e161fbac28d30 and https://github.com/espressif/esp-idf/commit/3cb655eaabd136700071c9225f33715ea4915917 (with backports in v4.2, v4.1, v4.0, v3.3 release branches).

closed 2020-09-22 by igrr