espressif/esp-idf #1162: esp_bt_controller_mem_release issue

chegewara, 2017-10-25:
With simple code:

```
int app_main(void) {
//    ESP_ERROR_CHECK( heap_trace_init_standalone(trace_record, NUM_RECORDS) );
    esp_bt_controller_enable(ESP_BT_MODE_BLE);
	esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
//	task_webserver(nullptr);
	return 0;
}
```
app crashes:
```
D (133) BTDM_INIT: Release DRAM [0x3ffae6e0] - [0x3ffaff10]
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x400d7282
0x400d7282: esp_bt_controller_mem_release at /home/esp32/esp/esp-idf/components/bt/./bt.c:494
file: "/home/esp32/esp/esp-idf/components/bt/./bt.c" line 413
func: esp_bt_controller_mem_release
expression: heap_caps_add_region(mem_start, mem_end)

Backtrace: 0x40089b20:0x3ffccdf0 0x40089e95:0x3ffcce10 0x400d7282:0x3ffcce30 0x400d2628:0x3ffcce60 0x400d0957:0x3ffcce80
0x40089b20: invoke_abort at /home/esp32/esp/esp-idf/components/esp32/./panic.c:553
0x40089e95: _esp_error_check_failed at /home/esp32/esp/esp-idf/components/esp32/./panic.c:565
0x400d7282: esp_bt_controller_mem_release at /home/esp32/esp/esp-idf/components/bt/./bt.c:494
0x400d2628: app_main at /home/esp32/ESP32_Explorer/main/./main.cpp:71
0x400d0957: main_task at /home/esp32/esp/esp-idf/components/esp32/./cpu_start.c:429
```

chegewara, 2017-10-30:
Still nothing in this matter?

chegewara, 2017-10-31:
Ok, ive found it. 
`	esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
`
can be used after `esp_bt_controller_enable`. Trying to release memory before enabling bt controller causing crash.

chegewara, 2017-11-01:
@projectgus Im able to free heap from BT stack when i dont want to use classic BT and it works great:
```
I (416) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 0, 0
E (416) BLEDevice: 128460   <<< before
D (416) BTDM_INIT: Release DRAM [0x3ffb2730] - [0x3ffb8000]

D (426) BTDM_INIT: Release DRAM [0x3ffbbb28] - [0x3ffbdb28]

E (426) BLEDevice: 159084  <<< after
```
But now when im trying to scan for with ble client im getting this heap corrupt (it is in address released by bt):
```
CORRUPT HEAP: multi_heap.c:370 detected at 0x3ffbdb14
abort() was called at PC 0x40089c63 on core 0
0x40089c63: multi_heap_assert at /home/esp32/esp/esp-idf/components/heap/./multi_heap.c:635

```
and this is full backtrack:
```
Backtrace: 0x4008a51f:0x3ffe0de0 0x4008a54b:0x3ffe0e00 0x40089c63:0x3ffe0e20 0x40089f87:0x3ffe0e40 0x4008a241:0x3ffe0e70 0x4008260c:0x3ffe0ea0 0x40082650:0x3ffe0ed0 0x40082d07:0x3ffe0ef0 0x40082d45:0x3ffe0f70 0x401adb35:0x3ffe0f90 0x401ade23:0x3ffe0fb0 0x401b35be:0x3ffe0fd0 0x401ae782:0x3ffe0ff0 0x4001791d:0x3ffe1010 0x401aec69:0x3ffe1030 0x401aee3a:0x3ffe1070 0x400188f5:0x3ffe10a0 0x4004d2ae:0x3ffe10e0 0x401b1975:0x3ffe1100 0x4004c94d:0x3ffe1120 0x4001a637:0x3ffe1140 0x40019d11:0x3ffe1170 0x40055b4d:0x3ffe1190 0x401ae2b3:0x3ffe11b0 0x401ae303:0x3ffe11d0
0x4008a51f: invoke_abort at /home/esp32/esp/esp-idf/components/esp32/./panic.c:553

0x4008a54b: abort at /home/esp32/esp/esp-idf/components/esp32/./panic.c:553

0x40089c63: multi_heap_assert at /home/esp32/esp/esp-idf/components/heap/./multi_heap.c:635

0x40089f87: multi_heap_malloc_impl at /home/esp32/esp/esp-idf/components/heap/./multi_heap.c:635

0x4008a241: multi_heap_malloc at /home/esp32/esp/esp-idf/components/heap/./multi_heap_poisoning.c:339

0x4008260c: heap_caps_malloc at /home/esp32/esp/esp-idf/components/heap/./heap_caps.c:136

0x40082650: heap_caps_malloc_default at /home/esp32/esp/esp-idf/components/heap/./heap_caps.c:136

0x40082d07: trace_malloc at /home/esp32/esp/esp-idf/components/heap/./heap_trace.c:324

0x40082d45: __wrap_malloc at /home/esp32/esp/esp-idf/components/heap/./heap_trace.c:392

0x401adb35: buffer_alloc at /home/esp32/esp/esp-idf/components/bt/bluedroid/hci/buffer_allocator.c:24

0x401ade23: host_recv_pkt_cb at /home/esp32/esp/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:264

0x401b35be: vhci_send at ??:?

0x401ae782: r_eif_send at ??:?

0x401aec69: hci_tx_start at hci_tl.c:?

0x401aee3a: r_hci_tl_send at ??:?

0x401b1975: hci_le_set_scan_param_cmd_handler at llm_hci.c:?

0x401ae2b3: r_rw_schedule at ??:?

0x401ae303: btdm_controller_task at ??:?

```
here is code >> https://github.com/chegewara/ESP32_Explorer/tree/heap_issue

chegewara, 2017-11-06:
I have more loggings about this issue and application to recreate this issue:

```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4018627d  PS      : 0x00060d30  A0      : 0x801c01f7  A1      : 0x3ffb6090  
0x4018627d: osi_sem_take at /home/esp32/esp/esp-idf/components/bt/bluedroid/osi/semaphore.c:64

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000014c  A5      : 0x3ffcb680  
A6      : 0x00000002  A7      : 0x00000014  A8      : 0xcccccccd  A9      : 0x3ffb6090  
A10     : 0x3ffb62bc  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4018627d:0x3ffb6090 0x401c01f4:0x3ffb60b0 0x4018fade:0x3ffb60d0 0x4019e7fd:0x3ffb6100 0x401a03a1:0x3ffb6130 0x401a0dde:0x3ffb6150 0x401a1004:0x3ffb6180 0x401a1222:0x3ffb61a0
0x4018627d: osi_sem_take at /home/esp32/esp/esp-idf/components/bt/bluedroid/osi/semaphore.c:64

0x401c01f4: fixed_queue_try_dequeue at /home/esp32/esp/esp-idf/components/bt/bluedroid/osi/fixed_queue.c:234

0x4018fade: btm_acl_resubmit_page at /home/esp32/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_acl.c:2481

0x4019e7fd: btm_sec_disconnected at /home/esp32/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_sec.c:5840 (discriminator 2)

0x401a03a1: btu_hcif_disconnection_comp_evt at /home/esp32/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_hcif.c:619

0x401a0dde: btu_hcif_process_event at /home/esp32/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_hcif.c:182

0x401a1004: btu_hci_msg_process at /home/esp32/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:559

0x401a1222: btu_task_thread_handler at /home/esp32/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:559
```
Steps to recreate:
1. application https://github.com/chegewara/ESP32_demo_project
2. app starts as a wifi AP, address 192.168.4.1 to setup local wifi ssid and password (IP, gateway and netmask are optional),
3. web interface to interact with application https://github.com/chegewara/ESP32_Explorer/tree/master/fatfs_data 
4. optional: this line need to be changed if esp32 got IP different than 192.168.1.99, https://github.com/chegewara/ESP32_Explorer/blob/master/fatfs_data/ESP32Explorer.js#L2
5. on tab BLE->server steps required to start:
- create server
- optional: add service
- optional: add characteristic
- start advertising
until now everything works fine
6. connect to esp32 with ble client, (android nRF in my case)
7. disconnect from server
8. app crash
8a. with this line commented, there is no crash and all works good https://github.com/chegewara/ESP32_demo_project/blob/master/components/cpp_utils/BLEDevice.cpp#L193

Hope its enough informations to recreate issue

TianHao-Espressif, 2017-12-11:
@chegewara , sorry for the long time delay, could tell me your git commit ID?

chegewara, 2017-12-11:
@TianHao-Espressif if i remember this repository is created only for this issue, so its last one commit.

About delay, dont worry. Issues im opening are not very important to myself, but i think it good to help community. If its somehow helpful its good, if not i can stop opening issues.

TianHao-Espressif, 2017-12-12:
@chegewara , I have fix the problem. It's due to the function calling order that he esp_bt_controller_mem_release() should be called before esp_bt_controller_init(). Because esp_bt_controller_init() have something reference with the memory region. The bug fix will be merged soon.

chegewara, 2017-12-12:
Good news.

closed 2017-12-12 by chegewara