espressif/esp-idf #1166: Boot look up in start_cpu0_default

mill1000, 2017-10-26:
The ESP32 occasionally locks up while starting the Pro CPU. Easiest way to reproduce is build an application that calls esp_restart() in the app_main()

Last log messages received are:

> I (72) cpu_start: Pro cpu up.
> I (73) cpu_start: Starting app cpu, entry point is 0x40080dd0
> 0x40080dd0: call_start_cpu1 at cpu_start.c:222
> I (48) cpu_start: App cpu up.
> I (74) heap_init: Initializing. RAM available for dynamic allocation:
> I (75) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (76) heap_init: At 3FFB29E8 len 0002D618 (181 KiB): DRAM
> I (76) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
> I (77) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (78) heap_init: At 40088454 len 00017BAC (94 KiB): IRAM
> I (79) cpu_start: Pro cpu start user code

Placing additional logs in start_cpu0_default()

> I (0) cpu_start: App cpu up.
> I (74) heap_init: Initializing. RAM available for dynamic allocation:
> I (75) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (75) heap_init: At 3FFB29E8 len 0002D618 (181 KiB): DRAM
> I (76) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
> I (77) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (78) heap_init: At 4008841C len 00017BE4 (94 KiB): IRAM
> I (79) cpu_start: Pro cpu start user code
> E (79) clk: esp_clk_init()
> E (87) clk: uart_tx_wait_idle()

Looks like a lock up might be occurring in esp_clk_init(). Can't tell exactly as the log function doesn't become available again until after the lock up point.

igrr, 2017-10-26:
We're tracking a somewhat similar issue here.

Could you please mention the development board you are using, and CPU frequency selected in sdkconfig?

Also, could you please try changing DELAY_CPU_FREQ_SWITCH_TO_XTAL_WITH_150K define in components/soc/esp32/rtc_clk.c to 40?

mill1000, 2017-10-26:
Developing on the ESP-WROOM-32 module. CPU frequency in sdkconfig is set to 240 MHz

mill1000, 2017-10-26:
Changing DELAY_CPU_FREQ_SWITCH_TO_XTAL_WITH_150K to 40 appears to be working well.

igrr, 2017-10-26:
Thanks for confirming, this means we have been seeing the same issue. The fix should be in master branch soon.

negativekelvin, 2017-10-27:
@igrr question about rtc_clk_init and esp_clk_init: are we potentially reconfiguring clock when we don't need to (bootloader already did it) or is it already optimized?

igrr, 2017-10-27:
@negativekelvin 
Bootloader enables 80MHz clock, and if the application is also configured for 80MHz clock then indeed we will be switching to XTAL and then back to PLL for no reason. That is a good point, and an optimization easy to implement. For the case when the application uses 160MHz, we can also get away with less amount of configuration. For 240MHz, switching to XTAL and then to a different PLL is required, so no issue there.
At this point we have a single function for all the cases because it is somewhat easier to test than several functions, or different paths within single function depending on the state. Extra time needed for this process (switch to XTAL, disable PLL, then re-enable PLL, switch to PLL) is on the order of 250us. Definitely something that can be improved, but not terribly inefficient.

There is a separate function (rtc_clk_cpu_freq_set_fast) used for dynamic frequency scaling, which doesn't disable PLL when switching to XTAL in order to save time.

mill1000, 2017-11-07:
Looks like this might have been resolved in commit abacf8d.

igrr, 2017-11-08:
Fixed in https://github.com/espressif/esp-idf/commit/6d4ed4ff6c4ca6e3b3bd4af3c7aeabe33570b540.

closed 2017-11-08 by igrr