espressif/esp-idf #1168: Inconsistent behavior with AP+STA mode

chmorgan, 2017-10-27:
I'm seeing some inconsistent behavior here with AP+STA mode.

I'm using a passphrase of >= 8 characters and esp-idf from master from about a week and a half ago (sha 3a271a4ae7df8).

It looks like the AP works much better when the STA is connected to my wifi router but that when the STA isn't connected the esp32's AP typically doesn't allow connections.

Here is a test sequence that represents what I'm seeing. Note that the wifi settings updates are occurring while the system is running using cli commands.

Are there known issues with AP+STA mode? If not I'd like to report that I'm seeing some.

Test sequence:
- I couldn't connect to the AP from iPhone
- Updated my STA settings to point at my router, it connected
- I was then able to connect to the device AP
- Both interfaces let me get to the internal http server and the
console prints reflected that as well

- Changed STA ssid to something silly, saw that it disconnected
- Forgot AP from iPhone, reconnected and was able to reconnect and load a page

- Restarted unit from cli
- Reconnected iPhone to AP
- Was able to reload pages even though the STA isn't connected.

- Power cycled
- Able to reconnect to AP from iPhone and load pages from http

- Power cycled
- Unable to connect to the AP from iPhone
- Set STA ssid and passphrase and able to connect to my router
- Able to connect to the AP from iPhone again

negativekelvin, 2017-10-27:
Is the AP on the same channel in both cases? Does it show in scan? How far does it get in connecting when it fails?

chmorgan, 2017-10-27:
@negativekelvin hmm, that is a good question. Let me add that to my output so I can check.

I'm not sure how to tell how far it got. I see a printout indicating that a station is attaching and then several seconds later leaving. On the phone/computer side it never completes the connection.

chmorgan, 2017-10-30:
@negativekelvin one data point. If the STA isn't connected I see 'scan in progress' messages in the output every several seconds.

It's almost as if the wifi scan is causing the AP to lose its stations, although I haven't initiated a scan here (at least as far as I can tell).

If I am connected to an STA I don't see these scan messages and I don't see these station join messages repeatedly showing up in the serial terminal output. It looks like a strong correlation between issues maintaining connection to the esp32 AP when the esp32 is not connected to an STA (when in AP+STA mode).

```
esp32> I (189344) wifi: n:6 0, o:6 1, ap:6 1, sta:0 0, prof:6
I (189344) wifi: scan in process, ignore set current channel
I (189344) wifi: station: 5c:e0:c5:99:0c:dc join, AID=1, g, 20
I (189354) wifi: station:5c:e0:c5:99:0c:dc join,AID=1

I (198534) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (198534) wifi: station: 5c:e0:c5:99:0c:dc join, AID=1, g, 20
I (198534) wifi: station:5c:e0:c5:99:0c:dc join,AID=1

I (233064) wifi: n:6 1, o:6 0, ap:6 1, sta:0 0, prof:6
I (233074) wifi: scan in process, ignore set current channel
I (233074) wifi: station: 5c:e0:c5:99:0c:dc join, AID=1, n, 40U
I (233084) wifi: station:5c:e0:c5:99:0c:dc join,AID=1

sel ret
SSL server create ......
OK
SSL server accept client ......
OK
Conn req from  192.168.4.2:31978, using pool slot 0
Sel add 3 (write 0)
Sel add listen 0
sel ret
SSL server create ......
Sel add 4 (write 0)
Sel add listen 0
I (245004) wifi: n:6 1, o:6 1, ap:6 1, sta:0 0, prof:6
I (245004) wifi: station: 5c:e0:c5:99:0c:dc join, AID=1, n, 40U
I (245004) wifi: station:5c:e0:c5:99:0c:dc join,AID=1

sel ret
URL = /index.html
Is url index 12
Heatshrink compressed file; decode parms = b4
Sel add 4 (write 1)
Sel add 4 (write 0)
Sel add listen 0
I (262914) wifi: n:6 1, o:6 1, ap:6 1, sta:0 0, prof:6
I (262914) wifi: station: 5c:e0:c5:99:0c:dc join, AID=1, n, 40U
I (262914) wifi: station:5c:e0:c5:99:0c:dc join,AID=1
```

negativekelvin, 2017-10-30:
https://github.com/espressif/esp32-wifi-lib/issues/4#issuecomment-299423735

chmorgan, 2017-10-30:
The channel looks valid in both cases. I was able to reproduce the issue and the computer only has trouble connecting to the esp32 AP when the esp32 STA isn't connected to an AP. As soon as the esp32 STA is connected the computer is able to connect and maintain connection to the esp32 AP.

```
Working case

esp32> wifi
AP
  mac: 30:ae:a4:0c:31:ec
  ssid: xxxxxxxxxxx
  passphrase: defaultpassword
  channel: 6
STA
  mac: 30:ae:a4:0c:31:ec
  ssid: yyyyyyyyyyyy
  passphrase: zzzzzzzzzz
  channel: 0
  connected to ssid: 'yyyyyyyyyyy'
  rssi: '-76'
esp32> 

Failing case

esp32> wifi
AP
  mac: 30:ae:a4:0c:31:ec
  ssid: xxxxxxxxxxxxx
  passphrase: defaultpassword
  channel: 6
STA
  mac: 30:ae:a4:0c:31:ec
  ssid: dummy
  passphrase: 
  channel: 0
  connected to ssid: ''
  rssi: '0'
esp32> 
```

chmorgan, 2017-10-30:
@negativekelvin note that I'm not explicitly setting channel for either the AP or STA. I'm using whatever is set by default.

chmorgan, 2017-10-30:
@negativekelvin and I can confirm that my Airport Extreme is on channel 6 for its 2.4GHz channel, channel 149 for its 5GHz channel.

negativekelvin, 2017-10-30:
Right but if the esp32 sta is scanning all channels trying to connect it is going to switch channels affecting the ap

chmorgan, 2017-10-30:
Doh. @negativekelvin that other thread looks exactly like the issue I’m seeing. What is the fix? Is ap+sta mode not useable when you aren’t connected to an sta?

negativekelvin, 2017-10-30:
Yes unless the wifi lib is fixed to allow fixed channel scan on sta connect (@TimXia) you would have to  only try to connect to sta once in a while to minimize impact

TimXia, 2017-10-30:
@negativekelvin @chmorgan The channel can not be fixed because if the Router changes its channel, station will never connects to it forever.

negativekelvin, 2017-10-30:
@TimXia yes you are right for default it should scan all channel but should be possible to override if programmer wants to.

chmorgan, 2017-10-30:
Slowing the scan down might probably solve the visible issue for users trying to connect to the ap. Right now there isn’t time for the computer to connect and do much on the ap. Yes it would slow connection to the sta but otherwise it does appear nearly useless to use ap+sta when not attached to the sta.

My intent was to use the ap for configuration of the sta and other internal admin settings of the product. I thought about using btle but didn’t want an app to be required for configuration.

chmorgan, 2017-11-02:
Any updates here? I'm getting a bit nervous that the approach I assumed would work for my production setup (having the AP always available to enable the user to configure the STA) may not work. Not being able to do this reliably means I'll have to come up with another approach and I'm not sure at the moment exactly what the next best option might be....

TimXia, 2017-11-03:
It may make ap connected if slowing down the scan. But it also takes a long time for station to get re-connected sometimes. If you do not care about the re-connect time, it is a solution for this issue. However, we suggest to use a button to control the network configuration. For example, when the button is pressed, station disconnects from the router. Then enable softap and smart phone connects to softap. After ssid and password are transmitted to softap, disable softap and station connects to the router.

TimXia, 2017-11-03:
There is another solution. When SYSTEM_EVENT_STA_DISCONNECTED is received, do not call esp_wifi_connect() immediately, wait for a while for smart phone to connect to softap.

negativekelvin, 2017-11-03:
You could also do manual single channel scan and look for ssid then only call connect when you see it. But it is still annoying that you can't restrict station to single channel.

chmorgan, 2017-11-04:
I tried the “slow down sta connections” approach and it doesn’t appear to work very cleanly.

Button is an option but I’d prefer no user device interaction was necessary.

I’m wondering if the better approach is to try the sta connection a few times and then stop trying until the sta information is changed by the user. What I’m not sure about is scanning. I was thinking to perform scanning when the user visits the WiFi page and then stop scanning but I’m concerned that this may not keep the AP list current and could disrupt the ap connection that the user is making use of to configure the sta ssid. Thoughts on an ideal approach if one exists?

TimXia, 2017-11-06:
You can use a LED to indicate that the station is scanning or not. If it is scanning, do not use smart phone to configure the sta ssid.

negativekelvin, 2017-11-06:
@TimXia that sounds like a terrible user experience

chmorgan, 2017-11-06:
I am actually looking at a periodic approach with led indicator. Something like:
- STA mode, attempt to connect for 30 seconds at boot
- If failed to connect to STA host AP only and re-attempt connecting to STA every 1 minute
- If AP connection is established re-attempt connecting to STA every 5 minutes or if STA information is altered (via user interfaction)
- If STA is connected then move into APSTA mode. If STA is lost after connection then move back into the first state above.

(This is a very cribbed version of the state machine and there are timeouts in there as well)

And then of course the requisite event handlers for STA lost, AP connection lost, gained etc.

It's not ideal but imo it should be usable and it handles the cases of the AP going away and coming back (albeit with a number of seconds of delay), and preserves the AP connection once it is established.

I was thinking about using two esp-wroom-32 modules to work around the issue, or an external low cost wifi module for the sta connection. Those seemed like a lot of work.

Would be great if the radio could support multiple channels in the future. It's such a great feature and its partially disrupted by the channel limitation due to the softap.

chmorgan, 2017-11-08:
@TimXia I'm at the point where I'm trying to test this approach and I'm running into an interesting issue.

What I have is a handler that I call from the wifi event loop, lets call it wifi_state_handler(). This handles the "wifi state" and timeouts. The issue is that I'd like to inject events so those timeouts can take effect.

I'm using:

    system_event_t timerEvent;
    timerEvent.event_id = (system_event_id_t)(SYSTEM_EVENT_MAX);
    esp_event_send(&timerEvent);

in a timer callback.

This is producing errors, because the event code is checking the event it:

```
W (22174) event: unexpected system event 24!
E (22174) event: mismatch or invalid event, id=24
E (22174) event: default event handler failed!
```

Thoughts on how to best handle this? I could have the timer call wifi_state_handler() directly, but then I worry that both the wifi code and the timer will be calling the same function and there is a chance that both will be inside of this function at the same time in two threads.

Another option is to create another thread and marshal the wifi events over to this thread. That would let me inject the timeout events into this new queue. I'd prefer not to have yet another thread.

Is there another approach? I think I require the timeouts because I'd like to reconnect to the STA occasionally and I'd really like to manage all of the state logic in a single handler vs. splitting it up.

Thoughts?

TimXia, 2017-11-09:
You can add another event like SYSTEM_EVENT_TIMEOUT to system_event_id_t instead of using SYSTEM_EVENT_MAX directly. SYSTEM_EVENT_MAX can not be used because it is checked in event handler. I suggest to use another thread to process these events unless you are quite familiar with system event handler.

chmorgan, 2017-11-09:
@TimXia I'd prefer not to modify esp-idf for the changes, if I do I'll have to carry them around the whole time.

It seems like a bit of a waste to create basically an entire clone of the wifi event handling loop and waste that extra stack space. I'll likely use the same xQueueReceive() and send functions and then pass the system_event_t values over to that thread. I mean its doable, it should work fine but ehh.

What about a #define like SYSTEM_EVENT_USER_EVENT_COUNT that would add to the SYSTEM_EVENT_MAX. Then the user would have a number of spots available for user defined events. Could add a Kconfig entry in the esp32 component for setting the value of SYSTEM_EVENT_USER_EVENT_COUNT and default it to zero.

igrr, 2017-11-09:
In that case we'd prefer run-time registration of events / handlers,
otherwise this system will not be usable by multiple independent components.

chmorgan, 2017-11-09:
@igrr any pointers on the implementation there? The event loop itself is pretty simple, the default handlers I don't understand as well but I can see where the user handler is called. Can't say anything comes to mind on a clean and efficient way to register at run-time right at the moment.

chmorgan, 2017-11-09:
@TimXia, @igrr well I can certainly help out with improving the event loop stuff if there is a bit more direction. For now I'm thinking I'll just use xQueueXXX and marshal over to my own separate handler thread, its easy and low impact on esp-idf. Let me know if you have some idea of what you'd like to see for the event registration, if you have any idea, and maybe I can help with that.

chmorgan, 2018-03-14:
Ended up writing a state machine to manage switching between states, a separate thread where I process most events and custom events like timeouts etc. It works ok. Closing this out as this doesn't look like the root cause is something fixable in software.

closed 2018-03-14 by chmorgan