espressif/esp-idf #1172: Bluetooth scanner not working

Arthedian, 2017-10-27:
I am using gatt_client from examples and even if I have two devices (phone, laptop) set to visible, it still cannt searched them?
```

I (111) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (172) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (442) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 0, 0
E (482) BT: btc_gattc_call_handler()
I (482) GATTC_DEMO: REG_EVT
I (482) GATTC_DEMO: scan start success
I (562) GATTC_DEMO: 78 bd bc 17 1b af
I (562) GATTC_DEMO: searched Adv Data Len 28, Scan Response Len 0
I (562) GATTC_DEMO: searched Device Name Len 0
I (572) GATTC_DEMO:

I (712) GATTC_DEMO: 78 bd bc 17 1b af
I (712) GATTC_DEMO: searched Adv Data Len 28, Scan Response Len 0
I (712) GATTC_DEMO: searched Device Name Len 0
I (722) GATTC_DEMO:

```
It says, that device name len is 0. Why?  And it looks only 1 device. Why?

Yulong-espressif, 2017-11-04:
@Arthedian When you set visible by the phone or laptop, it only for the classic bt not ble. If you when the phone can be found. You should use iphone with the lightblue app. You can find this APP in the app store. for the Android phone, the system must higher than 5.0 then can be act as a slave. Hope this answer can help you. Thanks.

Arthedian, 2017-11-05:
@Yulong-espressif Oh. Ok thank you for your answer. It helped me. But I have still few questions:
1) Why sometimes it has name len 0 and sometimes proper namelen? Why I see a name of BT device (BT earphones) on my Android phone, but on ESP32 it has 0 name len?

2) Can I scan classic BT? Because I didnt find anything in the docs.

pavestru, 2017-11-07:
@Yulong-espressif, it seems I have the same issue. I was trying to use both ESP_BT_MODE_BTM (dual mode) and ESP_BT_MODE_CLASSIC_BT (Classic BT mode), but neither of them were picking a phone with visible classic BT, only devices with BLE.

See below or here: https://github.com/espressif/esp-idf/blob/master/components/bt/include/bt.h#L64-L72
```c++
/**
 * @brief Bluetooth mode for controller enable/disable
 */
typedef enum {
    ESP_BT_MODE_IDLE       = 0x00,   /*!< Bluetooth is not running */
    ESP_BT_MODE_BLE        = 0x01,   /*!< Run BLE mode */
    ESP_BT_MODE_CLASSIC_BT = 0x02,   /*!< Run Classic BT mode */
    ESP_BT_MODE_BTDM       = 0x03,   /*!< Run dual mode */
} esp_bt_mode_t;
```

jonface, 2017-11-23:
@pavestru I have the same problem, doesn't matter what mode I use, I seem to only get BLE results. I checked out the esp-idf yesterday so I am up to date as far as I know.

Thanks

Yulong-espressif, 2017-12-15:
@Arthedian Can you update your IDF version, I think this is because the scan reponse issues.

FayeY, 2017-12-21:
Please feel free to reopen it, if this issue is still unsolved on the latest IDF.

closed 2017-12-21 by FayeY