espressif/esp-idf #1173: WiFi Alliance test against Krack fails in latest 2.1

lollizze, 2017-10-27:
We have run WiFi alliance tests on commit b6c91ce088ef64bd5b96a5af04885040b42b1816 and the latest on 2.1 but we get one failing test, with the output
"Received 5 unique replies to replayed broadcast ARP requests. Client is vulnerable to group key reinstallations in the 4-way handshake (or client accepts replayed broadcast frames)!"

TimXia, 2017-10-30:
@lollizze How did you run WiFi alliance tests? We tested the vulnerability with the tool provided by the author of the vulnerability. The test passed with the log 
"client DOESN'T seem vulnerable to group key installation in the 4-way handshake."

lollizze, 2017-10-30:
@TimXia we ran tests given by Wi-Fi Allience ([wi-fi-alliance-security-update](https://www.wi-fi.org/news-events/newsroom/wi-fi-alliance-security-update))
the test that fails on the test for replayed broadcast ARP requests

lollizze, 2017-10-30:
@TimXia Wi-Fi alliance published an updated test suite for this and the result from that one is much better

TimXia, 2017-10-30:
@lollizze The test has passed with the latest test suite?

lollizze, 2017-10-30:
@TimXia Yes, just ran there new suite and all tests are OK

closed 2017-10-31 by FayeY