espressif/esp-idf #1175: No easy way to know when RMT transmission has ended

chanibal, 2017-10-27:
From what I know, the only way to know when TX has ended at the moment is to completely replace RMT initialization (`rmt_driver_install`) with copy-pasted and slightly modified version. So it would load the, again slightly modified, `rmt_driver_isr_default` where finally you can tap into the event. 

You could also poll the RMT struct, but that will take time and resources.

I propose an easy fix for this solution adding a handler that will fire when transmission ends, see pull request https://github.com/espressif/esp-idf/pull/1174

koobest, 2017-11-09:
@chanibal   Hi, chanibal, we have API 
`esp_err_t rmt_wait_tx_done(rmt_channel_t channel, TickType_t wait_time)`  that you can call to see if the transmission has been finished.

chanibal, 2017-11-09:
But `rmt_wait_tx_done` needs to be polled from a task/timer, and it's not usable in interrupt context. And knowing immediately is very usefull for two-way communication using signals generated from RMT.

In my current implementation, I need only to check if RMT has finished transmitting and don't have to execute custom logic at end of transmission. But I can't do that either, because I need to know that from ISR context, and `rmt_wait_tx_done` is not something that I should use there, even with `wait_time=0`.

chanibal, 2017-11-09:
For details on what I need this message for, look at the diagram at https://github.com/espressif/esp-idf/issues/1179

koobest, 2017-11-19:
yes, you are right. `rmt_wait_tx_done` is not suitable in some cases. we do not currently have a good solution for this, but your fix is realy a good way to be taken.

closed 2017-11-22 by FayeY

phatpaul, 2018-09-23:
esp_err_t rmt_wait_tx_done(rmt_channel_t channel, TickType_t wait_time) could be used for polling (with wait_time==0), but it emits an error message.

E (54406) rmt: Timeout on wait_tx_done

How should I poll for tx busy without causing the error message?

chanibal, 2018-09-24:
@phatpaul : Don't. My question is almost a year old, [this has been fixed since](https://github.com/espressif/esp-idf/pull/1174). 
You should register a callback with [`rmt_register_tx_end_callback`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html#_CPPv228rmt_register_tx_end_callback15rmt_tx_end_fn_tPv).

phatpaul, 2018-09-24:
I see the new callback, and that will be very useful for when I need to wait for tx to be free.

But when I enter my CGI function, I would like to know immediately whether tx is busy.  
(I want to know if it is busy before calling rmt_write_sample(), because that function might yield if it is busy and I don't want my http server to be blocked.)

- If tx is not busy, I will start a RMT transmission from within the CGI function and return immediately.  I don't need to wait for completion in this case.

- If tx is busy, I will register the rmt_register_tx_end_callback callback from within the CGI function and return immediately with HTTPD_CGI_MORE to allow the http server to service other requests.  When rmt_register_tx_end_callback triggers, it will resume the http server connection/CGI function.

So again, I just want to call rmt_wait_tx_done(rmt_channel_t channel, TickType_t wait_time)  (with wait_time==0) to check if tx is busy, without generating an error message.

Do you think I should open a new issue for this?  Thanks.

koobest, 2018-09-25:
Hi, @phatpaul 
I added an API to get the status of the RMA channel. It will be merged soon. thanks.

koobest, 2018-10-09:
Hi, here is the description of each field of the status register:

```
struct {
            uint32_t mem_waddr_ex:   10; //[9:0]    The current memory read address of channel n
            uint32_t reserved10:            2; //[11:10]
            uint32_t mem_raddr_ex:     10;  //[21:12]  The current memory write address of channel n
            uint32_t reserved22:           2;  //[23:22]   
            uint32_t state:                     3;  //[26:24]  The channel n state machine status register:
0x0 : idle, 0x1 : send, 0x2 : read memory, 0x3 : receive, 0x4 : wait.
            uint32_t mem_owner_err:   1;  //[27]  When channel n is configured for receive mode, this bit will turn to high level if rmt_mem_owner register is not set to 1.
            uint32_t mem_full:              1;  //[28]  The menory full status bit for channel n turns to high level
 when mem_waddr_ex is greater than or equal to the configuration range.
            uint32_t mem_empty:         1;  //[29]  The menory empty status bit for channel n. in acyclic mode,
 this bit turns to high level when mem_raddr_ex is greater than or equal to the configured range.
            uint32_t apb_mem_wr_err: 1;  //[30]  The apb write menory status bit for channel 2 turns to
 high level when the apb write address exceeds the configuration range.
            uint32_t apb_mem_rd_err:  1; // [31]  The apb read menory status bit for channel 2 turns to high level when the apb read address exceeds the configuration range
    } status_ch_n;
```

we will update the TRM and the IDF soon.

thanks!!

phatpaul, 2018-10-09:
Thanks for the description of the mystery status register.

But please also take a look at my pull-request.  It adds some needed functions for me and probably for others too.
https://github.com/espressif/esp-idf/pull/2461

johannesmaibaum, 2018-10-25:
Hi @koobest ,

> ```c++
> ...
> uint32_t mem_waddr_ex: 10; //[9:0] The current memory read address of channel n
> uint32_t mem_raddr_ex: 10;  //[21:12] The current memory write address of channel n
> ...
> ```

Is this true? Or is read/write swapped in the comments?