espressif/esp-idf #1176: wifi_ap_config_t / wifi_sta_config_t inconsistencies

chmorgan, 2017-10-27:
I ran into a few things that seemed inconsistent about the ap and sta config structures. As the firmware for the wifi module is a binary I thought I'd report them here to see if what I saw was something that should be improved.

In particular:

- The password fields don't indicate that they require null termination, I assume they are because there isn't any other way I could think of them working, except in case of a 64 character password where there is no termination.

-There is an ssid_len field for the ap but not for the sta.

https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi_types.h#L185

typedef struct {
    uint8_t ssid[32];           /**< SSID of ESP32 soft-AP */
    uint8_t password[64];       /**< Password of ESP32 soft-AP */
    uint8_t ssid_len;           /**< Length of SSID. If softap_config.ssid_len==0, check the SSID until there is a termination character; otherwise, set the SSID length according to softap_config.ssid_len. */
    uint8_t channel;            /**< Channel of ESP32 soft-AP */
    wifi_auth_mode_t authmode;  /**< Auth mode of ESP32 soft-AP. Do not support AUTH_WEP in soft-AP mode */
    uint8_t ssid_hidden;        /**< Broadcast SSID or not, default 0, broadcast the SSID */
    uint8_t max_connection;     /**< Max number of stations allowed to connect in, default 4, max 4 */
    uint16_t beacon_interval;   /**< Beacon interval, 100 ~ 60000 ms, default 100 ms */
} wifi_ap_config_t;

typedef struct {
    uint8_t ssid[32];      /**< SSID of target AP*/
    uint8_t password[64];  /**< password of target AP*/
    wifi_scan_method_t scan_method;    /**< do all channel scan or fast scan */
    bool bssid_set;        /**< whether set MAC address of target AP or not. Generally, station_config.bssid_set needs to be 0; and it needs to be 1 only when users need to check the MAC address of the AP.*/
    uint8_t bssid[6];     /**< MAC address of target AP*/
    uint8_t channel;       /**< channel of target AP. Set to 1~13 to scan starting from the specified channel before connecting to AP. If the channel of AP is unknown, set it to 0.*/
    wifi_sort_method_t sort_method;    /**< sort the connect AP in the list by rssi or security mode */
    wifi_fast_scan_threshold_t  threshold;     /**< When scan_method is set to WIFI_FAST_SCAN, only APs which have an auth mode that is more secure than the selected auth mode and a signal stronger than the minimum RSSI will be used. */
} wifi_sta_config_t;

liuzfesp, 2017-11-26:
HI @chmorgan I think the ssid_len is a historical field, a lot of existing code will be impacted if we remove it. Anyway, we can remove it, at least add some comments about this field.

XinDeng11, 2018-01-05:
@chmorgan Hi， thank you for advise, as Zhifu said, it maybe a history story. However, we only change the structure in our big version release like 3.0 or 4.0, we will add it to our discussion list, thanks.

freedaun, 2018-09-06:
Hello. chmorgan suggested to include a comment on the wifi_ap_config_t and wifi_sta_config_t structs. Currently the password field says:

    `uint8_t password[64];       /**< Password of ESP32 soft-AP */`

But it should be something like:

    `uint8_t password[64];       /**< Password of ESP32 soft-AP, null terminated */`

This makes it clear there is no other option.

Alvin1Zhang, 2019-03-13:
@espxiehang Hi, would you help to take a look at this issue? Thanks.

closed 2019-03-13 by igrr