espressif/esp-idf #1177: mbedtls_pk_sign is twice as slower when using MPI acceleration

pieceofsummer, 2017-10-27:
Consider the following example:
```c
uint8_t hash[32];

const char *message = "Hello World!";
mbedtls_sha512((const unsigned char *)message, 12, hash, 0);

mbedtls_pk_context pk;
mbedtls_pk_init(&pk);

int ret = mbedtls_pk_parse_key(&pk, clientKey, sizeof_clientKey, NULL, 0);
if (ret < 0)
{
    printf("pk_parse_key -0x%x", -ret);
}
else
{
    uint8_t signature[256]; // big enough to hold signature
    size_t signature_length = sizeof(signature);

    uint32_t start_at = system_get_time();

    for (int i = 0; i < 10; i++)
    {
        ret = mbedtls_pk_sign(&pk, MBEDTLS_MD_SHA512,
                hash, 0, signature, &signature_length,
                NULL, NULL);
        if (ret < 0)
        {
            printf("pk_sign -0x%x", -ret);
        }
    }

    printf("10 signs in %u ms", (system_get_time() - start_at) / 1000);

    mbedtls_pk_free(&pk);
}
```

`clientKey` is an elliptic secp256r1 key (a test one, so there's no harm to post it):
```
-----BEGIN EC PARAMETERS-----
BggqhkjOPQMBBw==
-----END EC PARAMETERS-----
-----BEGIN EC PRIVATE KEY-----
MHcCAQEEIOfiGhATIqrbO269AeGOrJkBa3vRpM0tlhwwFTD9xGqSoAoGCCqGSM49
AwEHoUQDQgAE77tcIw7m3VDpmSYxT9pMTn3Pq2NFuAV9JrJDD+smeQLsBYZb9koC
JtrRDCMjxUvG3s9i/qA24qAU9HoLJFPMVw==
-----END EC PRIVATE KEY-----
```

Results:
- Without MPI acceleration: `10 signs in 1944 ms`
- With MPI acceleration: `10 signs in 2932 ms`
- With MPI acceleration (use interrupts): `10 signs in 3687 ms`

The above numbers are for Release build @ 240MHz.

pieceofsummer, 2017-10-27:
The results above were for v2.1, on master it gives somewhat better results (but still far from non-accelerated version):
- Without MPI acceleration: `10 signs in 1964 ms`
- With MPI acceleration: `10 signs in 2794 ms`
- With MPI acceleration (use interrupts): `10 signs in 3176 ms`

negativekelvin, 2017-10-28:
You mean SHA acceleration?
http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#mbedtls-hardware-sha

pieceofsummer, 2017-10-28:
@negativekelvin no, SHA acceleration works fine :)

I mean MPI acceleration, http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#mbedtls-hardware-mpi

negativekelvin, 2017-10-28:
Oh ok computing ECDSA signatures of a SHA hash?

negativekelvin, 2017-10-28:
Similar results single core mode @240MHz:

Without MPI acceleration: 10 signs in 1894 ms
With MPI acceleration: 10 signs in 2435 ms
With MPI acceleration (use interrupts): 10 signs in 2964 ms

projectgus, 2017-10-28:
Sorry about that, somehow we have notes in the config menu help for both AES & SHA but no note for hardware MPI.

There are at least three underlying issues contributing to this performance gap:

* CPU MPI operations are are competitively quick (as you've seen) , especially if the CPU is running at 240MHz. For slower CPU clock speeds, the balance can shift.

* A limitation in the MPI engine means that when either operand of an MPI multiplication is longer than 2048 bits, we have to make two hardware multiply operations and then combine the result (see `mpi_mult_mpi_overlong`). I don't have hard numbers, but this hurts EC more than RSA - depending on key length.

* In SMP mode, there's a rarely triggered hardware bug that means we have to suspend the other core when reading/writing the "DPORT" region of memory. Mostly this doesn't matter for performance because DPORT memory is rarely accessed, with the sole exception being the hardware cryptography registers. This is why V3.0 will be slower than V2.1, see esp_dport_access.h and related macros for some details. If you run in single core mode, this protection is disabled and the hardware MPI will run faster.

Due to this combination of factors, hardware MPI acceleration is disabled by default but can be enabled for users who are using single core mode and/or a slower clock speed.

We have some ideas for further optimisations, at least for the DPORT access accelerations. It's unlikely hardware MPI will beat the CPU running at 240MHz though, even with these.

pieceofsummer, 2017-10-28:
@projectgus probably it would be reasonable to have two separate config options for MPI acceleration: one for MPI_MOD_EXP and the other for MPI_MUL_MPI. So relatively short EC multiplications may execute in software, but expensive ModExp operations would be done by hardware.

I've also managed to speed things up a little (from 2794 to 2709 ms) just by optimizing `modular_inverse` method, and could even get to 2607 by adding multiplication shortcut when one of the operands takes only one limb.

projectgus, 2017-10-30:
> probably it would be reasonable to have two separate config options for MPI acceleration: one for MPI_MOD_EXP and the other for MPI_MUL_MPI

I agree this sounds reasonable. I'll do some benchmarking when I get a chance.

> I've also managed to speed things up a little 

Nice! Please consider sending PRs when you're done, it would be much appreciated. :)

Alvin1Zhang, 2019-04-19:
@pieceofsummer Hi, have you considered sending PRs? It will be much appreciated. Thanks.

projectgus, 2019-10-30:
We've made some optimisations so the numbers are closer now, but MPI acceleration is still slower for EC signing:

On IDF master, [using this code](https://gist.github.com/projectgus/7ab367479dfd3a5f4d34314e1906f38b), configuration updated for -O2 optimization , and hardware acceleration features.

* 240MHz software MPI+SHA -> 1573ms for 10 signatures
* Add Hardware SHA -> 1533ms
* Add Hardware MPI -> 1745ms