espressif/esp-idf #1178: espcoredump.py generates errors when used with OTA enabled

jwuavionix, 2017-10-27:
I have an application with OTA implemented and I'm looking to enable the coreDump to UART functionality.  I'm able to get everything working when the code is executing from the factory app partition and the output from espcoredump is as expected.  However, when I capture a coredump when running from either of the OTA partitions, I'm seeing the following error at the top of the output from espcoredump.

==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: struct value *value_of_register_lazy(struct frame_info *, int): Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: struct value *value_of_register_lazy(struct frame_info *, int): Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
GDB exited (None / None)!
Problem occured! GDB exited, restart it.
Reading symbols from /Users/thing20/esp/myApp/build/myApp.elf...done.

vonnieda, 2018-06-26:
This sounds like the same problem as described in https://github.com/espressif/esp-idf/issues/321#issuecomment-281243358. I'm wondering if there has been any update to that. @gerekon mentioned that they would distribute the ROM ELF in the future.

igrr, 2019-02-26:
Closing as https://github.com/espressif/esp-idf/issues/321 has been fixed.

closed 2019-02-26 by igrr