espressif/esp-idf #1179: Cannot switch between RMT and GPIO input with interrupt on same pin

chanibal, 2017-10-27:
I have a protocol that requires me to receive a synchronization signal (using a GPIO interrupt for this) and reply to it on the same line (using a RMT signal). The communication is performed by two interrupts interacting, in pseudocode:
```
IRAM_ATTR void rmt_tx_end_handler () {
   gpio_config(gpio as input with interrupt handler);
}
IRAM_ATTR void gpio_message_begin_isr() {
   rmt_config(rmt in tx mode);
   rmt_write_items(signal);
}
```

[image]

This setup works - but only for a while (minutes). After some time, it crashes, usually with:

```
(...)/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40086ad3 on core 0
0x40086ad3: xQueueGenericReceive at (...)/esp-idf/components/freertos/queue.c:2034

Backtrace: 0x400889c4:0x3ffc0470 0x40088ac3:0x3ffc0490 0x40086ad3:0x3ffc04b0 0x40083055:0x3ffc04f0 0x4013c261:0x3ffc0540 0x40083273:0x
3ffc05a0 0x4008489e:0x3ffc05f0 0x40082af1:0x3ffc0610 0x400850e2:0x00000000
0x400889c4: invoke_abort at (...)/esp-idf/components/esp32/panic.c:553

0x40088ac3: abort at (...)/esp-idf/components/esp32/panic.c:553

0x40086ad3: xQueueGenericReceive at (...)/esp-idf/components/freertos/queue.c:2034

0x40083055: esp_log_write at (...)/esp-idf/components/log/log.c:111

0x4013c261: rmt_config at (...)/esp-idf/components/driver/rmt.c:780

0x40083273: 🢂🢂🢂 gpio_message_begin_isr 🢀🢀🢀

0x4008489e: gpio_intr_service at (...)/esp-idf/components/driver/gpio.c:428

0x40082af1: _xt_lowint1 at (...)/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400850e2: vTaskEnterCritical at (...)/esp-idf/components/freertos/tasks.c:4751

Rebooting...
```

Usually followed by a boot loop in RMT setup:
```
( ...initializing RMT again... )

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4008464b  PS      : 0x00050031  A0      : 0x40082af4  A1      : 0x3ffc05e0
0x4008464b: rmt_driver_isr_default at (...)/esp-idf/components/driver/rmt.c:780

0x40082af4: _xt_lowint1 at (...)/esp-idf/components/freertos/xtensa_vectors.S:1105

A2      : 0x00000006  A3      : 0x00000240  A4      : 0x00000002  A5      : 0x00000000
A6      : 0x400831bc  A7      : 0x3ffdcb90  A8      : 0x00000006  A9      : 0x3ffdcbc0
0x400831bc: 🢂🢂🢂 rmt_tx_end_handler 🢀🢀🢀

A10     : 0x00000010  A11     : 0x3ffc05e0  A12     : 0x800883d0  A13     : 0x3ffcf4b0
A14     : 0x00000001  A15     : 0x3ffdcbd9  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000014  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x4008464b:0x3ffc05e0 0x40082af1:0x3ffc0610 0x4000bfed:0x00000000
0x4008464b: rmt_driver_isr_default at (...)/esp-idf/components/driver/rmt.c:780

0x40082af1: _xt_lowint1 at (...)/esp-idf/components/freertos/xtensa_vectors.S:1105
```

I guess that using `gpio_config` and `rmt_config` in ISR context is asking for trouble, but I don't see any other way to switch between the two modes, and I think the signal is too fast for using FreeRTOS synchronization methods.

Is there any way to switch between the two functions in a more elegant matter then reconfigure them all the time?

negativekelvin, 2017-10-28:
If you are in open drain mode you could probably just leave both input and output enabled and just disable the interrupt

chanibal, 2017-10-28:
@negativekelvin, thanks for the idea, it looks like this has helped, but after a day of work on this, I'm still not sure, so let's not close this yet.

Also, what is the friendliest way for na ISR to execute the RMT? Is `rmt_write_items` a good thing from ISR?

chanibal, 2017-10-28:
After some tests, it looks like both `rmt_write_items` and `rmt_tx_start` sometimes crash ESP when used from ISR context. Usually something like:

```
W (30) boot: PRO CPU has been reset by WDT.
W (30) boot: WDT reset info: PRO CPU PC=0x4008875e
0x4008875e: vListInsert at  (...)/esp-idf/components/freertos/list.c:188

W (31) boot: WDT reset info: APP CPU PC=0x40086b5b
0x40086b5b: uxPortCompareSet at (...)/esp-idf/components/freertos/port.c:373
 (inlined by) vPortCPUAcquireMutexIntsDisabled at (...)/esp-idf/components/freertos/portmux_impl.h:86
 (inlined by) vPortCPUAcquireMutex at (...)/esp-idf/components/freertos/port.c:332
```

For `rmt_tx_start`, execute it from ISR context, and earlier execute `rmt_write_items` from normal context to fill the memory.

pufstudio, 2017-10-29:
It's probably not a good idea to call setup and transmit functions from isr.
You should set up dedicated task and event queue and then call xQueueSendFromISR to start transmission.

If you have GPIOs to spare you could use one input to sense falling edge and then
disable sensing so you don't interrupt on your own messages. After transmission is done
enable it again. You can use volatile bool so you don't need to set up interrupt every time.

chanibal, 2017-10-29:
@pufstudio, currently I don't call setup from ISR, everything is setup normally and works on the same line thanks to some fiddling with the gpio matrix. The only thing that is left to be called in the ISR is checking if enough time has passed from the last signal and sending with the RMT (`rmt_write_items`). 

I have tried a queue from ISR before, but the timings I got were extremely unsatisfactory. While with an ISR I can get a 4us lag, with a queue it's around 1000x more. I can work with a lag of around 40us.
[image]

Here is how I set up the ISR and queue. This example does not have RMT, it's just a try to relay interrupt processing from interrupt context to the task one.
```
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/queue.h>
#include <driver/gpio.h>
#include <esp_log.h>

const auto pin = GPIO_NUM_19;
const auto output1 = GPIO_NUM_5;
const auto output2 = GPIO_NUM_18;

QueueHandle_t q;

IRAM_ATTR void isr(void*_) {
	BaseType_t xHigherPriorityTaskWoken = pdFALSE;
	uint32_t level = gpio_get_level(pin);

	// copying input to output1
	gpio_set_level(output1, level);

	// copying input to output2
	xQueueOverwriteFromISR(q, &level, &xHigherPriorityTaskWoken);
	//if (xHigherPriorityTaskWoken == pdTRUE)
	//	taskYIELD();	// no *FromISR version? this one crashes
}

void task(void*_) {
	uint32_t level = false;
	for (;;) {
		if (xQueueReceive(q, &level, portMAX_DELAY)) {
			gpio_set_level(output2, level);
		}
	}
}

extern "C" void app_main() {
	q = xQueueCreate(1, sizeof(uint32_t));

	xTaskCreate(task, "q-task", 4000, nullptr, configMAX_PRIORITIES - 1, nullptr);

	const gpio_config_t gpioDebugOutput = {
		.pin_bit_mask = 1ull << output1 | 1ull << output2,
		.mode = GPIO_MODE_OUTPUT,
		.pull_up_en = GPIO_PULLUP_DISABLE,
		.pull_down_en = GPIO_PULLDOWN_DISABLE,
		.intr_type = GPIO_INTR_DISABLE
	};
	gpio_config(&gpioDebugOutput);

	const gpio_config_t gpioInput = {
		.pin_bit_mask = 1ull << pin,
		.mode = GPIO_MODE_INPUT_OUTPUT_OD,
		.pull_up_en = GPIO_PULLUP_DISABLE,
		.pull_down_en = GPIO_PULLDOWN_DISABLE,
		.intr_type = GPIO_INTR_ANYEDGE
	};
	ESP_ERROR_CHECK(gpio_config(&gpioInput));
	ESP_ERROR_CHECK(gpio_set_level(pin, 1));
	ESP_ERROR_CHECK(gpio_install_isr_service(0));
	ESP_ERROR_CHECK(gpio_isr_handler_add(pin, isr, nullptr));

	for (;;) { vTaskDelay(pdMS_TO_TICKS(1000)); ESP_LOGI("isr-q-test", "."); }
}
```

I guess that if I would be able to call `taskYIELD()`, it would work much much faster, but calling it crashes ESP. Perhaps there should be a `taskYIELDFromISR()`, but there is none and FreeRTOS's docs call that function port-dependant. Although, ESP's comments in `freertos/queue.h` use just `taskYIELD`.

Still, calling it from the place they're commented crashes with:
```
***ERROR*** A stack overflow in task IDLE has been detected.
abort() was called at PC 0x40086c10 on core 0
0x40086c10: vApplicationStackOverflowHook at (...)/esp-idf/components/esp32/panic.c:553

Backtrace: 0x40086af8:0x3ffc0520 0x40086bf7:0x3ffc0540 0x40086c10:0x3ffc0560 0x40083deb:0x3ffc0580 0x400859f8:0x3ffc05a0 0x40085a89:0x
3ffc05c0 0x40083a56:0x3ffc05f0 0x4008208d:0x3ffc0610 0x400d168f:0x00000000
0x40086af8: invoke_abort at (...)/esp-idf/components/esp32/panic.c:553

0x40086bf7: abort at (...)/esp-idf/components/esp32/panic.c:553

0x40086c10: vApplicationStackOverflowHook at (...)/esp-idf/components/esp32/panic.c:553

0x40083deb: vTaskSwitchContext at (...)/esp-idf/components/freertos/tasks.c:4497

0x400859f8: _frxt_dispatch at (...)/esp-idf/components/freertos/portasm.S:406

0x40085a89: vPortYield at (...)/esp-idf/components/freertos/portasm.S:555

0x40083a56: gpio_intr_service at (...)/esp-idf/components/driver/gpio.c:428

0x4008208d: _xt_lowint1 at (...)/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400d168f: esp_vApplicationIdleHook at (...)/esp-idf/components/esp32/freertos_hooks.c:62
```

chanibal, 2017-10-29:
Um... there is a `portYIELD_FROM_ISR`, not sure how I missed that earlier - using it, delays fell from a few milliseconds to around 20us which is in tolerance.

After adding `rmt_write_bytes` to the handler it stopped crashing (or at least doesn't crash as often). Will try for a longer period of time and report back.

chanibal, 2017-11-01:
It looks like it's stable now. Closing, thank you for the help.

closed 2017-11-01 by chanibal