espressif/esp-idf #1182: sdmmc: command timeouts allowed but not initialized

jkearins, 2017-10-28:
Recently, command timeouts is allowed in sdmmc_type.h. But new field must be initialized in macro SDMMC_HOST_DEFAULT(), defined in sdmmc_host.h
It should be something like this, at the end of macro definition:
`.command_timeout_ms = 0, \
`

igrr, 2017-11-02:
Thanks for pointing this out. 
This doesn't seem to be an immediate issue when SDMMC_HOST_DEFAULT is used in C, because of the way [initializers work](http://en.cppreference.com/w/c/language/struct_initialization) (all structure members for which an initializer is not provided are initialized the same way as if they had static storage duration; for integral types this means setting to zero). 
But it may become an issue if we add more fields in the future, and SDMMC_HOST_DEFAULT is used from C++ code. In this case GCC will complain about initializer skipping some fields. So we will fix this.

igrr, 2017-12-08:
Fixed in https://github.com/espressif/esp-idf/pull/1216 and follow-up commits.

closed 2017-12-08 by igrr