espressif/esp-idf #1183: vfs_fat.c: no TCHAR support

jkearins, 2017-10-28:
In functions defined in ff.h, type TCHAR is used for names and paths. For example:
`FRESULT f_open (FIL* fp, const TCHAR* path, BYTE mode);`
In common case TCHAR is defined as char, but if _LFN_UNICODE is defined as 1 in ffconf.h than TCHAR is defined as WCHAR. There is no support for TCHAR in vfs_fat.c and we have type mismatch, for example:
```
static int vfs_fat_open(void* ctx, const char * path, int flags, int mode)
{
    ...
    FRESULT res = f_open(&fat_ctx->files[fd], path, fat_mode_conv(flags));
    ...
}

```

Perhaps, functins in vfs_fat.c must be declared like:
`static int vfs_fat_open(void* ctx, const TCHAR* path, int flags, int mode)`

igrr, 2017-11-08:
Thanks for reporting this @jkearins.
I think that replacing the path type in `vfs_fat_open` will not immediately help because newlib will still pass `const char*` to `open` syscall. What we probably need to do is to convert multibyte (UTF-8) name (coming from the application) into wide char (UTF-16) name which FATFS expects.
To do that we will probably pull in the `iconv` library.

jkearins, 2017-11-08:
It seems I incorrectly focused the problem. When I change definitions in ffconv.h as:
```
#define _LFN_UNICODE 1 //was 0
#define _FS_EXFAT 1 //was 0

```
I got warnings in vfs_fat.c about incompatible types char* and WCHAR* and sdcard does not mount (errno=2). There are no warnings in commit a3a4a20 (27.09.2017) and sdcard works fine in 4-wire mode.

igrr, 2017-11-08:
Sorry, i can't reproduce the lack of warnings with commit a3a4a20.

With this commit, i get warnings similar to
```
/Users/ivan/e/esp-idf/components/fatfs/src/vfs_fat_spiflash.c: In function 'esp_vfs_fat_spiflash_mount':
/Users/ivan/e/esp-idf/components/fatfs/src/vfs_fat_spiflash.c:73:35: warning: passing argument 2 of 'f_mount' from incompatible pointer type [-Wincompatible-pointer-types]
     FRESULT fresult = f_mount(fs, drv, 1);
```

In addition to setting _LFN_UNICODE to 1 and _FS_EXFAT to 1, i had to set _USE_LFN to 2 and _CODE_PAGE to 437, to pass the compilation without errors.

jkearins, 2017-11-08:
I found my error. I used `#define _LFN_UNICODE 0` in commit a3a4a20, sdcard worked. In last commit I used `#define _LFN_UNICODE 1` and sdcard does not work. It seems we can not use `#define _LFN_UNICODE 1` with fat system because there is no way to pass wchar parameter to esp_vfs_fat_sdmmc_mount. A trick esp_vfs_fat_sdmmc_mount((char*)mount_point) does not work. The function calculates string length using strlen():
```
esp_err_t esp_vfs_register(const char* base_path, const esp_vfs_t* vfs, void* ctx)
{
    return esp_vfs_register_common(base_path, strlen(base_path), vfs, ctx, NULL, NULL);
}

```
Have you got plans to release full unicode support for exFat?

igrr, 2017-12-08:
@jkearins We have recently updated FATFS to release 0.13a, it includes support for UTF-8 at API level. That is, if `CONFIG_FATFS_MAX_LFN` is non-zero, unicode support is enabled, and `FF_LFN_UNICODE` option determines which encoding is used by f_open, f_rename, and other APIs dealing with file names. Currently it is hard-coded to 0 (=ANSI/OEM), but it can be set to 2 (=UTF-8), in which case TCHAR is defined as char, and application can use names in UTF-8 encoding, which will be transparently passed into FATFS.

We are working on an example which demonstrates this, along with a Kconfig option to set `FF_LFN_UNICODE` to 2.

jkearins, 2017-12-08:
@igrr I successfully tested release 0.13a with following parameters:
in menuconfig:
```
CONFIG_FATFS_CODEPAGE=866
CONFIG_FATFS_LFN_NONE=
CONFIG_FATFS_LFN_HEAP=y
CONFIG_FATFS_LFN_STACK=
CONFIG_FATFS_MAX_LFN=255

```
in ffconv.h:
```
#define FF_FS_EXFAT		1	//default is 0
#define FF_LFN_UNICODE	2	//default is 0
#define FF_STRF_ENCODE	3	//default is 3

```
[image]
Thank you for update.

commit 99b8ae34 (2018-03-26): fatfs/add menuconfig options for different encodings

This commit adds character encoding configurations in for the fatfs component.
The FF_LFN_UNICODE definition in ffconf.h can now be changed to accept UTF-8 or
UTF-16 encoded filernames. Test cases using UTF-8 encoded file paths and names in
FatFs have also been added.

Closes #1183
closed 2018-03-26 by igrr