espressif/esp-idf #1184: esp_partition_mmap fails with partitions larger than 3MB

OtherCrashOverride, 2017-10-28:
Using esp_partition_mmap to map a partition larger than 3MB results in failures.  The following reproduces the issue:
https://github.com/OtherCrashOverride/esp32-flashtest

When the partition is:
`storage, 0x40, 0x01, 0x100000, 0x300000` the operation is successful.

When the partion is:
`storage, 0x40, 0x01, 0x100000, 0x400000` an error is returned (ESP_ERR_NO_MEM).

When the partition is:
`storage, 0x40, 0x01, 0x100000, 0x500000` a "double exception" occurs.

The test hardware is an ESP-WROOM32 with the flash chip removed and replaced with W25Q128JVSIQ (16MB).  The flash is correctly detected for flash and boot:
```
I (48) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 16MB
```

Spritetm, 2017-10-28:
The ESP32 memory map implementation cannot map more than 4MiB of RAM into data memory because of a hardware limitation: the MMU doing the mapping only has a 4MiB segment of D-port RAM at its disposal, which also contains the rodata section of the application you're running (resulting in your case in 3MiB of mappable memory). Not sure why the double exception happens, however.

OtherCrashOverride, 2017-10-28:
Does this limitation also apply to PSRAM? The implication is that only 3MB of the 4MB of PSRAM is useable.

negativekelvin, 2017-10-28:
Psram maps to a different address space

OtherCrashOverride, 2017-10-28:
If that is the case, can I leave the 4MB flash on WROOM32 as-is and attach a 16MB flash where PSRAM would be?

negativekelvin, 2017-10-28:
https://esp32.com/viewtopic.php?f=12&t=2884

OtherCrashOverride, 2017-10-28:
Just to clarify, the intent is not to disable the WROOM32 4MB flash.  The intent is to replace PSRAM with flash so that I can mmap it without the 3MB limit.

negativekelvin, 2017-10-28:
No I don't think sram cache is cross compatible with flash cache.  What do you need to mmap all that read only memory for?

OtherCrashOverride, 2017-10-28:
I am not able to disclose further details at this time.

igrr, 2017-10-28:
You can mmap up to 16MB minus application code size if you mmap into
instruction address space (4-byte aligned access only).

OtherCrashOverride, 2017-10-28:
Also, to clarify my understanding of the discussion so far, if I use PSRAM, the **entire** 4MB (exactly 4 * 1024 * 1024 bytes) is available for **unaligned** random access?

For example:
```uint8_t* ptr = (uint8_t*)malloc(4 * 1024 * 1024);```
will succeed?

and:
```uint8_t value = ptr[1];```
will succeed?

And will the above succeed if I also have 3MB of flash mmap'd?

OtherCrashOverride, 2017-10-28:
> You can mmap up to 16MB minus application code size if you mmap into instruction address space (4-byte aligned access only).

@igrr,
I tested this, and it also fails.  With a 4MB mapping I get corrupted data.

With 8MB:
`storage,  0x40, 0x01,    0x100000, 0x800000`
I get  "Double exception".

Testing was done with the following changed in main.c:
`err = esp_partition_mmap(part, 0, part->size, /*SPI_FLASH_MMAP_DATA*/ SPI_FLASH_MMAP_INST, (const void**)&data, &handle);`

negativekelvin, 2017-10-29:
16mb total but irom space is <11mb so 8+app may be pushing it and corruption may be due to fragmentation?

OtherCrashOverride, 2017-10-29:
I did not check spi_flash_mmap_dump.  This is the first I have heard of it.  Is there documentation for it?

> 16mb total but irom space is <12mb so 8+app may be pushing it ?

In the repo I linked that reproduces the issue, the app is substantially less than 1MB.  Since 8MB mapping + 1MB = 9MB, even if the app did take up the entire 1MB there would still be 2MB of unused map space (total 11MB).

I feel I should reiterate that even though a 4MB SPI_FLASH_MMAP_INST does not crash, the resulting memory returned is **not correct** (does not match flash contents).  Changing the partition size causes the data being mapped to also change.  For example, mapping 4MB does not produce the same 4MB region as mapping 6MB (lower 4MB region ignoring upper 2MB).

My immediate concern is determining whether this is a software or hardware issue since this impacts future plans.

OtherCrashOverride, 2017-10-29:
https://github.com/OtherCrashOverride/esp32-flashtest/tree/instmap

The output of spi_flash_mmap_dump follows.  Note that spi_flash_mmap_dump for the 8MB mapping can not be provided since it "double exception" panics:

```
storage,  0x40, 0x01,    0x100000, 0x400000
-------------------------------------------
partition: type=64, subtype=1, address=0x100000, size=0x400000, label='storage', encrypted=0
pre mmap heap=0x46774
post mmap heap=0x4675c
mmap=0x400f0000
handle=2 page=79 count=64
page 0: refcnt=1 paddr=1
page 77: refcnt=1 paddr=2
page 78: refcnt=1 paddr=3
page 79: refcnt=1 paddr=16
page 80: refcnt=1 paddr=17
page 81: refcnt=1 paddr=18
page 82: refcnt=1 paddr=19
page 83: refcnt=1 paddr=20
page 84: refcnt=1 paddr=21
page 85: refcnt=1 paddr=22
page 86: refcnt=1 paddr=23
page 87: refcnt=1 paddr=24
page 88: refcnt=1 paddr=25
page 89: refcnt=1 paddr=26
page 90: refcnt=1 paddr=27
page 91: refcnt=1 paddr=28
page 92: refcnt=1 paddr=29
page 93: refcnt=1 paddr=30
page 94: refcnt=1 paddr=31
page 95: refcnt=1 paddr=32
page 96: refcnt=1 paddr=33
page 97: refcnt=1 paddr=34
page 98: refcnt=1 paddr=35
page 99: refcnt=1 paddr=36
page 100: refcnt=1 paddr=37
page 101: refcnt=1 paddr=38
page 102: refcnt=1 paddr=39
page 103: refcnt=1 paddr=40
page 104: refcnt=1 paddr=41
page 105: refcnt=1 paddr=42
page 106: refcnt=1 paddr=43
page 107: refcnt=1 paddr=44
page 108: refcnt=1 paddr=45
page 109: refcnt=1 paddr=46
page 110: refcnt=1 paddr=47
page 111: refcnt=1 paddr=48
page 112: refcnt=1 paddr=49
page 113: refcnt=1 paddr=50
page 114: refcnt=1 paddr=51
page 115: refcnt=1 paddr=52
page 116: refcnt=1 paddr=53
page 117: refcnt=1 paddr=54
page 118: refcnt=1 paddr=55
page 119: refcnt=1 paddr=56
page 120: refcnt=1 paddr=57
page 121: refcnt=1 paddr=58
page 122: refcnt=1 paddr=59
page 123: refcnt=1 paddr=60
page 124: refcnt=1 paddr=61
page 125: refcnt=1 paddr=62
page 126: refcnt=1 paddr=63
page 127: refcnt=1 paddr=64
page 128: refcnt=1 paddr=65
page 129: refcnt=1 paddr=66
page 130: refcnt=1 paddr=67
page 131: refcnt=1 paddr=68
page 132: refcnt=1 paddr=69
page 133: refcnt=1 paddr=70
[... 312 lines cut ...]
page 167: refcnt=1 paddr=104
page 168: refcnt=1 paddr=105
page 169: refcnt=1 paddr=106
page 170: refcnt=1 paddr=107
page 171: refcnt=1 paddr=108
page 172: refcnt=1 paddr=109
page 173: refcnt=1 paddr=110
page 174: refcnt=1 paddr=111
page 175: refcnt=1 paddr=112
page 176: refcnt=1 paddr=113
page 177: refcnt=1 paddr=114
page 178: refcnt=1 paddr=115
page 179: refcnt=1 paddr=116
page 180: refcnt=1 paddr=117
page 181: refcnt=1 paddr=118
page 182: refcnt=1 paddr=119
page 183: refcnt=1 paddr=120
page 184: refcnt=1 paddr=121
page 185: refcnt=1 paddr=122
page 186: refcnt=1 paddr=123
page 187: refcnt=1 paddr=124
page 188: refcnt=1 paddr=125
page 189: refcnt=1 paddr=126
page 190: refcnt=1 paddr=127
```

igrr, 2017-10-29:
The output looks correct, at the first glance. What method do you use to check mmap'ed data, and how do you write data to flash in the first place?

OtherCrashOverride, 2017-10-29:
The official esptool.py is used to write the flash data.  It is also used again to verify the data.  The data is then used by an application.  Correct behavior is observed when the data is 3MB or less.  When data is 4MB, incorrect behavior is observed.  Simply changing the partition size (which is used in the mapping as demonstrated in the repo) changes the incorrect behavior.  The behavior changes are consistently reproducible based on the size of the mapping (without changing the data).  Since the application does not use or rely on the partition size (the data is self describing), the inference is the change is solely due to the mapping size.

I should note that a change to esptool.py was required.  The timeout value is too small for data larger than 3MB and will timeout depending on the contents (compression).

Once the cause of the "double fault" issue is identified and corrected, I will test again to see if the data corruption issue is also corrected.  If the issue persists, I will create a reproducible test case if required.

negativekelvin, 2017-10-29:
https://github.com/espressif/esp-idf/blob/e6afe28bafe5db5ab79fae213f2e8e1ccd9f937c/components/spi_flash/flash_mmap.c#L164

This may be wrong and should be 4*64 - 77 = 179 available instruction pages?

OtherCrashOverride, 2017-10-31:
To whom is that comment directed?

igrr, 2017-10-31:
@negativekelvin yes, you're right, it should be 4 * 64 not 3 * 64. And the other issue is incorrect bail-out a couple of lines below (shouldn't return from the function with the cache disabled).
Still looking into the data corruption issue here, will update this ticket once we have some progress.

jcmvbkbc, 2023-05-04:
Hello,
I see something that looks like a variation of this issue.
The following code
```
#include <stdio.h>
#include "sdkconfig.h"
#include "esp_system.h"
#include "spi_flash_mmap.h"
#include "esp_partition.h"

static void IRAM_ATTR dump_data(unsigned long base, unsigned long size)
{
        unsigned long i, j;

        for (i = 0; i < size; i += 65536) {
                unsigned long sum = 0;
                for (j = 0; j < 16384; ++j) {
                        sum += j[(unsigned long*)(base + i)];
                }
                printf("%08lx..%08lx: %08lx\n", base + i, base + i + 65535, sum);
        }
}

static const void * IRAM_ATTR map_partition(const char *name, unsigned long off)
{
        const void *ptr;
        spi_flash_mmap_handle_t handle;
        esp_partition_iterator_t it;
        const esp_partition_t *part;

        it = esp_partition_find(ESP_PARTITION_TYPE_ANY, ESP_PARTITION_SUBTYPE_ANY, name);
        part = esp_partition_get(it);
        if (esp_partition_mmap(part, off, part->size - off, SPI_FLASH_MMAP_INST, &ptr, &handle) != ESP_OK)
                abort();
        dump_data((unsigned long)ptr, part->size);
        return ptr;
}

static void IRAM_ATTR map_flash_and_go(void)
{
        const void *ptr0, *ptr;

        ptr0 = map_partition("linux", 0);
        printf("ptr = %p\n", ptr0);

        ptr = map_partition("rootfs", 0);
        printf("ptr = %p\n", ptr);

        spi_flash_mmap_dump();

        asm volatile ("jx %0" :: "r"(ptr0));
}

void app_main(void)
{   
    map_flash_and_go();
    esp_restart();
}
```
with the following partition table
```
## Label          type  ST       Offset      Length
nvs,              data, nvs,     0x0000a000, 0x00005000
phy_init,         data, phy,     0x0000f000, 0x00001000
factory,          app,  factory, 0x00010000, 0x00030000
linux,            0x40, 0x0,     0x00040000, 0x001c0000
rootfs,           0x40, 0x1,     0x00200000, 0x00200000
```
runs to completion without an error, but the dumped data does not match the FLASH contents once the virtual address crosses the `0x40400000` boundary. The complete output:
```
[... 145 lines cut ...]
page 117: refcnt=1 paddr=42
page 118: refcnt=1 paddr=43
page 119: refcnt=1 paddr=44
page 120: refcnt=1 paddr=45
page 121: refcnt=1 paddr=46
page 122: refcnt=1 paddr=47
page 123: refcnt=1 paddr=48
page 124: refcnt=1 paddr=49
page 125: refcnt=1 paddr=50
page 126: refcnt=1 paddr=51
page 127: refcnt=1 paddr=52
page 128: refcnt=1 paddr=53
page 129: refcnt=1 paddr=54
page 130: refcnt=1 paddr=55
page 131: refcnt=1 paddr=56
page 132: refcnt=1 paddr=57
page 133: refcnt=1 paddr=58
page 134: refcnt=1 paddr=59
page 135: refcnt=1 paddr=60
page 136: refcnt=1 paddr=61
page 137: refcnt=1 paddr=62
page 138: refcnt=1 paddr=63
```

 I know that the correct data is in the FLASH, because changing `map_partition("rootfs", 0);` to `map_partition("rootfs", 20 << 16);` prints the missing data.

jcmvbkbc, 2023-05-08:
That is I have two questions here. First there seems to be a bug that the function returns success, but the mapping is not fully functional. I guess I should report it as a separate issue, right? Second it looks very much like a hardware issue, so the question is: is it supposed to work at all on esp32 (simultaneous mapping of 4MB of PSRAM and 4MB of FLASH)?

ESP32DE, 2023-05-19:
@igrr 
[image]

jcmvbkbc-issues, 2023-12-17:
I think I've figured out this one. As long as `DPORT_{PRO,APP}_CACHE_CTRL1_REG` bits `DPORT_{PRO,APP}_CACHE_MASK_IRAM1` are set, reads from the region `0x4040_0000..0x407F_FFFF` (VAddr2 in esp32 TRM terminology) return `0xbad00bad`, and as long as bits `DPORT_{PRO,APP}_CACHE_MASK_IROM0` are set, reads from the region `0x4080_0000..0x40BF_FFFF` (VAddr3) return `0xbad00bad`. Clearing these bits makes the VAddr2 and VAddr3 mappings work.

jcmvbkbc, 2024-01-23:
I think I've figured out this one. As long as `DPORT_{PRO,APP}_CACHE_CTRL1_REG` bits `DPORT_{PRO,APP}_CACHE_MASK_IRAM1` are set, reads from the region `0x4040_0000..0x407F_FFFF` (VAddr2 in esp32 TRM terminology) return `0xbad00bad`, and as long as bits `DPORT_{PRO,APP}_CACHE_MASK_IROM0` are set, reads from the region `0x4080_0000..0x40BF_FFFF` (VAddr3) return `0xbad00bad`. Clearing these bits makes the VAddr2 and VAddr3 mappings work.

GoeBachmann, 2026-11-16:
> I think I've figured out this one. As long as `DPORT_{PRO,APP}_CACHE_CTRL1_REG` bits `DPORT_{PRO,APP}_CACHE_MASK_IRAM1` are set, reads from the region `0x4040_0000..0x407F_FFFF` (VAddr2 in esp32 TRM [...]

We've found this as well. Interestingly, the bootloader seems to use `cache_ll_l1_get_bus({0,1}, irom_load_addr_aligned, irom_size)` and `cache_ll_l1_enable_bus({0,1}, bus_mask)` already (which clear the `DPORT_{PRO,APP}_CACHE_MASK_IROM0` and `DPORT_{PRO,APP}_CACHE_MASK_IRAM1` bits) when the `.text` section in the image is large enough to reach into the Vaddr2 and Vaddr3 regions. Though for some reason the device isn't really stable in that case for us.

- on a device with PSRAM, we get crashes early on during init of the second core. The symptom is a direct reset without a backtrace.
- on a device without PSRAM, we get crashes (reliably!) when enabling the cache error interrupt handler. The cache error interrupt fires immediately after it's enabled. This time with a backtrace.
  - The mappings seem at least partially functional though. Calling functions in the Vaddr2 and Vaddr3 areas works if done before the cache error interrupt is enabled.

@jcmvbkbc is there anything else you needed to set besides the MASK bits in the CACHE_CTRL1 register and increasing the linker script region size for the code segment?