espressif/esp-idf #1186: tcp_output causes random crashes

BlackEdder, 2017-10-29:
I've been using tcp_output after tcp_write in my code, but tcp_output seems to randomly cause the hardware to through an error and reset. If I disable tcp_output I don't get the crashes, but this delays my messages.

The same code works fine on an ESP8266 chip.

ESP-Coco, 2017-10-30:
Hi @BlackEdder , can you post your code for further analysis?

me-no-dev, 2017-10-30:
Check here for how to call those methods safely :) https://github.com/me-no-dev/AsyncTCP/blob/master/src/AsyncTCP.cpp#L220-L247

BlackEdder, 2017-10-30:
The relevant code can be found here:
https://gitlab.com/BlackEdder/painlessMesh/blob/master/src/painlessMeshConnection.cpp#L304

@me-no-dev Thanks for the link, although I am not sure how to interpret that. It seems like very thin wrappers around the tcp_output function, but I see no added checks for when it is safe to call the code.

me-no-dev, 2017-10-30:
those tiny wrappers actually inside that the tcp_* functions are called on the TCP thread and will not cause issues :) use _tcp_write instead of tcp_write (same goes for send and the rest)

BlackEdder, 2017-10-30:
Thanks @me-no-dev. Mind if I ask a couple more stupid questions :) I based my code on http://lwip.wikia.com/wiki/Raw/TCP and they don't mention threads. Does this mean this is esp32 specific? Does the esp8266 need the same wrappers and if so why does the current code work fine on esp8266. Might you have any links to relevant documentation on the TCP thread?

me-no-dev, 2017-10-30:
I don't have documentation, but yes this is less or more ESP32 specific :) you have two cores and threads. 8266 does not need the wrappers :) BTW please close this issue if you are satisfied wth the answer :P

BlackEdder, 2017-10-30:
Just for other people having the same problem. Here is the relevant lwip api calls for threading: http://www.nongnu.org/lwip/2_0_x/group__lwip__opts__lock.html#ga8e46232794349c209e8ed4e9e7e4f011

I guess the easiest way is just to wrap your code with: `LOCK_TCPIP_CORE()` and `UNLOCK_TCPIP_CORE()`

closed 2017-10-30 by BlackEdder

me-no-dev, 2017-10-30:
I advise you towards using the wrappers that I provided :) not sure how those locks will actually work on ESP32.