espressif/esp-idf #1188: WPS : disconnected from station loop

bhaessig83, 2017-10-30:
Hello,
I've asked for some help in the espressif/arduino-esp32 issues, and they proposed me to ask directly in this forum after making some tests.

I've compiled and loaded in my Olimex ESP32 gateway the WPS code from the library (see below) and I see the following feedback in the monitor.
"Starting WPS" and "Station Mode Started" are comming just after the boot and the errors after presssing the WPS button.

> Starting WPS
> Station Mode Started
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> ...
> 

I made some tests with the following routers with the same result:
- Huawei B310s-22
- My android phone (Sony Xperia Z3 compact)
And compiled with the two following boards configuration:
- Olimex ESP32 gateway
- ESP32 Dev Module

Code:
```
/*
Example Code To Get ESP32 To Connect To A Router Using WPS
===========================================================
This example code provides both Push Button method and Pin
based WPS entry to get your ESP connected to your WiFi router.

Hardware Requirements
========================
ESP32 and a Router having atleast one WPS functionality

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#include "WiFi.h"
#include "esp_wps.h"

/*
Change the definition of the WPS mode
from WPS_TYPE_PBC to WPS_TYPE_PIN in
the case that you are using pin type
WPS
*/
#define ESP_WPS_MODE WPS_TYPE_PBC

esp_wps_config_t config = WPS_CONFIG_INIT_DEFAULT(ESP_WPS_MODE);

String wpspin2string(uint8_t a[]){
  char wps_pin[9];
  for(int i=0;i<8;i++){
    wps_pin[i] = a[i];
  }
  wps_pin[8] = '\0';
  return (String)wps_pin;
}

void WiFiEvent(WiFiEvent_t event, system_event_info_t info){
  switch(event){
    case SYSTEM_EVENT_STA_START:
    Serial.println("Station Mode Started");
    break;
    case SYSTEM_EVENT_STA_GOT_IP:
    Serial.println("Connected to :" + String(WiFi.SSID()));
    Serial.print("Got IP: ");
    Serial.println(WiFi.localIP());
    break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
    Serial.println("Disconnected from station, attempting reconnection");
    WiFi.reconnect();
    break;
    case SYSTEM_EVENT_STA_WPS_ER_SUCCESS:
    Serial.println("WPS Successfull, stopping WPS and connecting to: " + String(WiFi.SSID()));
    esp_wifi_wps_disable();
    delay(10);
    WiFi.begin();
    break;
    case SYSTEM_EVENT_STA_WPS_ER_FAILED:
    Serial.println("WPS Failed, retrying");
    esp_wifi_wps_disable();
    esp_wifi_wps_enable(&config);
    esp_wifi_wps_start(0);
    break;
    case SYSTEM_EVENT_STA_WPS_ER_TIMEOUT:
    Serial.println("WPS Timedout, retrying");
    esp_wifi_wps_disable();
    esp_wifi_wps_enable(&config);
    esp_wifi_wps_start(0);
    break;
    case SYSTEM_EVENT_STA_WPS_ER_PIN:
    Serial.println("WPS_PIN = " + wpspin2string(info.sta_er_pin.pin_code));
    break;
    default:
    break;
  }
}

void setup(){
  Serial.begin(115200);
  delay(10);

  Serial.println();

  WiFi.onEvent(WiFiEvent);
  WiFi.mode(WIFI_MODE_STA);

  Serial.println("Starting WPS");

  esp_wifi_wps_enable(&config);
  esp_wifi_wps_start(0);
}

void loop(){
  //nothing to do here
}
```

Thanks is advance for your help
Bertrand

petervflocke, 2017-10-30:
Exactly the same issue in my case:
- the same sketch
- different board: Node32s
- different router: FRITZ!box 7490
Look forward to your support with best regards Peter.

XinDeng11, 2018-01-02:
@petervflocke @bhaessig83 Sorry for later reply, if it possible, could you offer the sniffer log when it fail. I am not sure whether is due to I do some change in scan operation at that time. Yes, there had been some issue about WPS at that time when you ask. However, we have fixed it now, may be you can try on the latest version, if it still happen, maybe you can offer the sniffer log about the STA and AP connect, thanks.

petervflocke, 2018-01-04:
@XinDeng11 & @bhaessig83 
time does not matter (ok it matters a bit) anyway thank you for considering my / our question.
tl;dr version: you have fixed it to some extent!

Long story:
Additionally to arduino I checked also the standard idf development environment.
with xtensa-esp32-elf-linux32-1.22.0-73-ge28a011-5.2.0.tar.gz and the example code from 
https://github.com/espressif/esp-idf/tree/master/examples/wifi/wps
I had the same issue a long loop of "SYSTEM_EVENT_STA_DISCONNECTED"
Monitor extract:

```
I (123) wifi: Init dynamic tx buffer num: 32
I (123) wifi: Init data frame dynamic rx buffer num: 32
I (123) wifi: Init management frame dynamic rx buffer num: 32
I (133) wifi: wifi driver task: 3ffc4f3c, prio:23, stack:4096
I (133) wifi: Init static rx buffer num: 10
I (143) wifi: Init dynamic rx buffer num: 32
I (143) wifi: Init rx ampdu len mblock:7
I (143) wifi: Init lldesc rx ampdu entry mblock:4
I (153) wifi: wifi power manager task: 0x3ffca3a8 prio: 21 stack: 2560
I (183) phy: phy_version: 359.0, e79c19d, Aug 31 2017, 17:06:07, 0, 0
I (183) wifi: mode : sta (30:ae:a4:27:53:18)
I (183) example_wps: start wps...
I (183) example_wps: SYSTEM_EVENT_STA_START
I (193) wpa: wifi_wps_enable
I (49133) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (49253) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (49383) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (49503) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
```
......

```
I (240113) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (240203) example_wps: SYSTEM_EVENT_STA_WPS_ER_TIMEOUT
I (240203) wpa: wifi_wps_disable
I (240203) wpa: wifi_wps_enable
I (360203) example_wps: SYSTEM_EVENT_STA_WPS_ER_TIMEOUT
I (360203) wpa: wifi_wps_disable
I (360203) wpa: wifi_wps_enable
```

then I upgraded to xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz
and cloned the complete esp-idf tree: git clone --recursive https://github.com/espressif/esp-idf.git

**and the WPS example works fine.** 

Then I updated the arduino environment by removing esp32 folder from "/opt/arduino/hardware/espressif/" and re-cloning the complete environment + again downloading & extracting xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz via python get.py and the results from wps example is like folow:

```
Starting WPS
Station Mode Started
Disconnected from station, attempting reconnection
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4000c310  PS      : 0x00060830  A0      : 0x800d3a06  A1      : 0x3ffca440  
A2      : 0x3ffce2b0  A3      : 0x00000004  A4      : 0x00000006  A5      : 0x3ffce2b0  
A6      : 0x3ffca450  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffca420  
A10     : 0x3ffce2c2  A11     : 0x3f403fb4  A12     : 0x00000000  A13     : 0x3ffce2c2  
A14     : 0x00000020  A15     : 0x00060023  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4000c310:0x3ffca440 0x400d3a03:0x3ffca450 0x400d3a6b:0x3ffca480 0x400dcafb:0x3ffca4a0 0x400dcb57:0x3ffca4c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c

Starting WPS
Station Mode Started
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
Connected to :keineverbindung
Got IP: 192.168.0.130
WPS Successfull, stopping WPS and connecting to: keineverbindung
Disconnected from station, attempting reconnection
Connected to :wifixxxx
Got IP: 192.168.0.130
```

Sometimes it takes one reboot some times two to get connected.

```
Decoding 5 results
0x400d3a03: wps_txStart at ?? line ?
0x400d3a6b: wifi_station_wps_eapol_start_handle at ?? line ?
0x400dcafb: timer_process_alarm at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./esp_timer.c line 388
0x400dcb57: timer_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./esp_timer.c line 388
```

It works - let's say this is somehow acceptable assuming that wifi must be setup at the beginning of the apps.

XinDeng11, 2018-01-05:
@petervflocke Could you please offer your example code for me to reproduce the phenmenon?

petervflocke, 2018-01-05:
@XinDeng11 sure I can,
I use the standard examples from your respository:
for arduino code refer to: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WPS/WPS.ino

I tested the code today morning again and - the first start went OK, the second after reboot dumped ones and then it worked - see log below:
```
Starting WPS
Station Mode Started
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
WPS Successfull, stopping WPS and connecting to: wifixxx
Disconnected from station, attempting reconnection
Connected to :wifixxx
Got IP: 192.168.0.130
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c

Starting WPS
Station Mode Started
Disconnected from station, attempting reconnection
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4000c310  PS      : 0x00060a30  A0      : 0x800d3a06  A1      : 0x3ffca440  
A2      : 0x3ffdd800  A3      : 0x00000004  A4      : 0x00000006  A5      : 0x3ffdd800  
A6      : 0x3ffca450  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffca420  
A10     : 0x3ffdd812  A11     : 0x3f403fb4  A12     : 0x00000000  A13     : 0x3ffdd812  
A14     : 0x00000020  A15     : 0x00060023  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4000c310:0x3ffca440 0x400d3a03:0x3ffca450 0x400d3a6b:0x3ffca480 0x400dcafb:0x3ffca4a0 0x400dcb57:0x3ffca4c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c

Starting WPS
Station Mode Started
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
Connected to :wifixxx
Got IP: 192.168.0.130
WPS Successfull, stopping WPS and connecting to: wifixxx
Disconnected from station, attempting reconnection
Connected to :wifixxx
Got IP: 192.168.0.130
```

For the esp-idf version refer to: https://github.com/espressif/esp-idf/tree/master/examples/wifi/wps

Regards and many thanks again for considering my post.
P.

XinDeng11, 2018-03-22:
Hi @petervflocke I tried to reproduce the phenomenon with single STA with one AP, but I can create a crash with two STAs do WPS with the AP. So I want to make sure whether you do WPS operation by two STAs at the same time?

closed 2018-04-04 by FayeY