espressif/esp-idf #1189: Best practice for catching ETH link status event

Gfast2, 2017-10-30:
Hi ESP-IDF,

I do love my ESP32 projects. And they literally let me - a mac user + Arduino Fan-boy start to understanding how microprocessors magically do jobs for my digital lifes everyday! 💘 

Right now I'm working on a *IO-controller* project. It read the GPIO changes and send this change as a event through _Ethernet Interface_ to other devices.
Since the Software Stack for **ETH** need sometime to start, and at last get a ***event*** to indicate that _**it is ready for sending data**_. How should I _catch_ them in my code? I do can change the source code in _esp-idf_'s `components` folder. But I do believe it is only the _**hacky way**_. But what is the ***right way***?

The following lines are the _trace_ about my _Ethernet Interface_. I think the last line indicate the init of this interface is finished.

```
I (4107) emac: emac start !!!

I (4117) emac: emac resetting ....
I (4117) emac: emac reset done
I (4127) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (4147) emac: emac start success !!!
I (8147) eth_example: ETHIP:192.168.178.220
I (8147) emac: eth link_up!!!
I (8157) event: eth ip: 192.168.178.220, mask: 192.168.0.0, gw: 192.168.178.1

```

negativekelvin, 2017-10-30:
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_event.h#L53

Use forum for how-to questions

Gfast2, 2017-10-30:
Hi @negativekelvin , Super cool. Now I know from where I can getting started with. Now I'd close this "***issue***" first. If I still have some questions around this topic. I will discuss them in [esp32.com](es32.com) 👍

closed 2017-10-30 by Gfast2