espressif/esp-idf #1190: SPIFFS config size

forthlightning, 2017-10-30:
Hi, wondering if it would be possible to add a Kconfig menu option to the spiffs component which will allow customization of size?

Looking at line 217 in spiffs config.h 

```
#ifndef SPIFFS_CFG_PHYS_SZ
#define SPIFFS_CFG_PHYS_SZ(ignore)        (1024*1536) // 1.5ish Mb
#endif
```

me-no-dev, 2017-10-30:
customization in size is done in the partitions configuration :)
See: https://github.com/espressif/esp-idf/blob/master/examples/storage/spiffs/partitions_example.csv#L6

forthlightning, 2017-10-30:
Oh, so it will detect the size from the partition table? Pretty snazzy! So if I made a change in the config file there will be no effect?

I was reading into the component and it looks like there are 2 config files... One in default and the other in include. Seems like it may create confusion when including.

me-no-dev, 2017-10-31:
only the one in include is used. And yes :) it reads the partition table and uses the size there.

closed 2017-11-02 by forthlightning