espressif/esp-idf #1191: Bad head at 0x3ffd3e08. Expected 0xabba1234 got 0x00000000 ???

YvedD, 2017-10-30:
Hi,
I took a example and tweaked it to my wishes.
but now the system reboots after one correct loop
what have i done wrong?

my code : `
/* structure that hold data*/
typedef struct{
  uint8_t sender;
  char *msg;
  uint32_t xTime;
}Data;

/* this variable hold queue handle */
xQueueHandle xQueue;
TaskHandle_t xTask1;
TaskHandle_t xTask2;
void setup() {

  Serial.begin(112500);
  /* create the queue which size can contains 5 elements of Data */
  xQueue = xQueueCreate(5, sizeof(Data));

  xTaskCreatePinnedToCore(
      sendTask,           /* Task function. */
      "sendTask",        /* name of task. */
      10000,                    /* Stack size of task */
      NULL,                     /* parameter of the task */
      1,                        /* priority of the task */
      &xTask1,                /* Task handle to keep track of created task */
      0);                    /* pin task to core 0 */
  xTaskCreatePinnedToCore(
      receiveTask,           /* Task function. */
      "receiveTask",        /* name of task. */
      10000,                    /* Stack size of task */
      NULL,                     /* parameter of the task */
      1,                        /* priority of the task */
      &xTask2,            /* Task handle to keep track of created task */
      1);                 /* pin task to core 1 */   
}

void loop() {

}

void sendTask( void * parameter )
{
  /* keep the status of sending data */
  BaseType_t xStatus;
  /* time to block the task until the queue has free space */
  const TickType_t xTicksToWait = pdMS_TO_TICKS(100);
  /* create data to send */
  Data data;
  /* sender 1 has id is 1 */
//  data.sender = 1;
  for(;;){
    Serial.print("sendTask run on core ");
    /* get the core that the task was pinned to */
    Serial.print(xTaskGetAffinity(xTask1));
    Serial.println(" is sending data");
    data.sender = xTaskGetAffinity(xTask1);
    data.xTime = micros();
    data.msg = "Hi Yves!";
    /* send data to front of the queue */
    xStatus = xQueueSendToFront( xQueue, &data, xTicksToWait );
    /* check whether sending is ok or not */
    if( xStatus == pdPASS ) {
      /* increase counter of sender 1 */
      Serial.println("sendTask sent data to Queue");
    }
    /* we delay here so that receiveTask has chance to receive data */
    delay(1000);
  }
  vTaskDelete( NULL );
}

void receiveTask( void * parameter )
{
  /* keep the status of receiving data */
  BaseType_t xStatus;
  /* time to block the task until data is available */
  const TickType_t xTicksToWait = pdMS_TO_TICKS(100);
  Data data;
  for(;;){
    /* receive data from the queue */
    xStatus = xQueueReceive( xQueue, &data, xTicksToWait );
    /* check whether receiving is ok or not */
    if(xStatus == pdPASS){
      Serial.print("receiveTask run on core ");
      /* get the core that the task was pinned to */
      Serial.print(xTaskGetAffinity(xTask2));
      /* print the data to terminal */
      Serial.print(" got data: ");
      Serial.print("sender = ");
      Serial.print(data.sender);
      Serial.print(" msg = ");
      Serial.print(data.msg);
      Serial.print(" time = ");
      Serial.println(data.xTime);
      
      free(&data);
    }
  }
  vTaskDelete( NULL );
}
`
my error 
`sendTask run on core 0 is sending data
sendTask sent data to Queue
receiveTask run on core 1 got data: sender = 0 msg = Hi Yves! time = 46363
CORRUPT HEAP: Bad head at 0x3ffd3e08. Expected 0xabba1234 got 0x00000000
assertion "head != NULL" failed: file "/Users/ficeto/Espressif/ESP32/esp-idf/components/heap/./multi_heap_poisoning.c", line 195, function: multi_heap_free
abort() was called at PC 0x400db39b on core 1
`
my backtrace
`Backtrace: 
0x40086a14:0x3ffd3d20 
0x40086b13:0x3ffd3d40 
0x400db39b:0x3ffd3d60 
0x40086759:0x3ffd3d90 
0x400824ea:0x3ffd3db0 
0x4008293d:0x3ffd3dd0 
0x4000bec7:0x3ffd3df0 
0x400d090d:0x3ffd3e10
`
I am new to this, so have no clue what is wrong
Grtz,

projectgus, 2017-10-30:
Hi @YvedD ,

I apologise, I just edited your comment instead of replying to it! I don't seem to be able to revert the edit, but hopefully it answers your question...

Angus

YvedD, 2017-10-30:
No Problem Angus,

I the meantime i had solved it already by myself (that was a home run for me as a newbie :-) )
I am still not out of the woods with what i am doing, it is all so new for me

now i am looking for something like a "xTaskCreateFromISRPinnedToCore" or something like that
Thanks anyway for the help offered

Yves

projectgus, 2017-10-30:
> now i am looking for something like a "xTaskCreateFromISRPinnedToCore" or something like that

You probably want to somehow give a semaphore or set an event group bit from the ISR and then create the task in another task. Or pre-create the task and have it block one one of these before it starts executing.

For general questions about IDF, the https://esp32.com forums are a great place to post.

closed 2017-10-30 by projectgus