espressif/esp-idf #1193: queue crashes after 01befe4

michprev, 2017-10-30:
Hi,
I am experiencing crashes after commit 01befe4 (add I2C hardware reset if the hw FSM get stuck).
My crash log:

```
/home/michal/esp-idf/components/freertos/./queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40085f07 on core 0
0x40085f07: xQueueGenericReceive at /home/michal/esp-idf/components/freertos/./queue.c:2034

Backtrace: 0x400894f8:0x3ffbdec0 0x400895c3:0x3ffbdee0 0x40085f07:0x3ffbdf00 0x40086039:0x3ffbdf40 0x400832b5:0x3ffbdf60 0x4008339d:0x3ffbdf90 0x4010d77f:0x3ffbdfb0 0x40107b8d:0x3ffbe270 0x40107b15:0x3ffbe2c0 0x400869d3:0x3ffbe2f0 0x40088808:0x3ffbe310 0x400888d1:0x3ffbe330 0x40087c19:0x3ffbe350 0x40087cef:0x3ffbe390
0x400894f8: invoke_abort at /home/michal/esp-idf/components/esp32/./panic.c:553

0x400895c3: abort at /home/michal/esp-idf/components/esp32/./panic.c:553

0x40085f07: xQueueGenericReceive at /home/michal/esp-idf/components/freertos/./queue.c:2034

0x40086039: xQueueTakeMutexRecursive at /home/michal/esp-idf/components/freertos/./queue.c:2034

0x400832b5: lock_acquire_generic at /home/michal/esp-idf/components/newlib/./locks.c:155

0x4008339d: _lock_acquire_recursive at /home/michal/esp-idf/components/newlib/./locks.c:169

0x4010d77f: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)

0x40107b8d: fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50

0x40107b15: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)

0x400869d3: vPortCPUAcquireMutexIntsDisabled at /home/michal/esp-idf/components/freertos/./tasks.c:4950
 (inlined by) vTaskEnterCritical at /home/michal/esp-idf/components/freertos/./tasks.c:4142

0x40088808: xEventGroupSetBits at /home/michal/esp-idf/components/freertos/./event_groups.c:277

0x400888d1: vEventGroupSetBitsCallback at /home/michal/esp-idf/components/freertos/./event_groups.c:614

0x40087c19: prvProcessReceivedCommands at /home/michal/esp-idf/components/freertos/./timers.c:847

0x40087cef: prvTimerTask at /home/michal/esp-idf/components/freertos/./timers.c:847
```
Steps to reproduce:
1) initialize i2c peripheral
2) perform i2c command while bad device / no device connected
3) call i2c_driver_delete
4) call any freertos queue function

Without step 2) everything works fine

michprev, 2017-11-06:
Lowering stack sizes of my tasks solved this problem.

michprev, 2017-11-06:
The problem still occurs sometimes.

I also noticed that I2C is now extremely slow - reading from my sensor used to take 560 us, now it takes 8 ms!!

michprev, 2017-11-07:
It seems like https://github.com/espressif/esp-idf/blob/ed1e32f583980d2fefcd47690b77d5b0e1896667/components/driver/i2c.c#L300 is causing memory corruption when bad I2C device connected.

michprev, 2017-11-07:
Also please note that using event groups forces us to great delays (as I mentioned 560 us operation now takes 8 ms) here:
https://github.com/espressif/esp-idf/blob/01befe4d6a2e94bfc7db6484ff586470edea3086/components/driver/i2c.c#L1146

kostyan5, 2017-11-10:
I'm experiencing i2c timeout after a few seconds of being powered up.  This did not happen prior to 01befe4.

koobest, 2017-11-17:
OK, we are testing on this, will give feedback soon.

InfiniteYuan1, 2017-11-17:
Hi:
We found some reasons, if you can provide your test code, I think we will quickly find out reason. Can you provide your test code? Thanks

michprev, 2017-11-17:
My code is at https://github.com/michprev/flyhero-esp32. This is where I try to connect to i2c device: https://github.com/michprev/flyhero-esp32/blob/master/IMU/main/IMU_Detector.cpp. When I connect SPI device my code recognizes it and then continues here: https://github.com/michprev/flyhero-esp32/blob/55237855ad2c63aa714ef83804197ac73785892b/The_Eye/main/main.cpp#L74

When I connect SPI device program crashes. With I2C device everything works fine.

Please let me know if you need more info. Thank you!

michprev, 2017-11-17:
You can find more minimized code here: https://github.com/michprev/flyhero-esp32/blob/issue_i2c/IMU/main/main.cpp

IMU detector is very the same: https://github.com/michprev/flyhero-esp32/blob/issue_i2c/IMU/main/IMU_Detector.cpp

Sensor which works with no problems (I2C sensor): https://github.com/michprev/flyhero-esp32/blob/issue_i2c/IMU/main/MPU6050.cpp

Sensor which causes crashes (non-I2C sensor): https://github.com/michprev/flyhero-esp32/blob/issue_i2c/IMU/main/MPU9250.cpp

InfiniteYuan1, 2017-11-20:
Can you provide a complete project that contains esp-idf? And , we recommend that you set the xTicksToWait parameter to non-zero when using the xQueueSend () function and the xQueueReceive () function.

michprev, 2017-11-20:
My esp-idf is up-to-date with https://github.com/espressif/esp-idf/tree/7e8c2a9c00a6fb05cd5da306b62c4474a999b1a2 (no modifications). The root of my project is at https://github.com/michprev/flyhero-esp32/tree/issue_i2c/IMU.

Ad xTicksToWait: my application is quite time critical. I need to read sensor data every 1 or 2 ms so that I can not afford running any FreeRTOS function with non-zero ticks argument.

michprev, 2017-11-20:
I believe that I finally found core of the problem.
Consider 3 following scenarios:

(1) MPU6050 (I2C-only sensor)
IMU detector tries to connect to an I2C device with this setup:
IO18 <---> SCL
IO05 <---> SDA

It succeeds and returns MPU6050 sensor (no SPI sensor checking). No problems here...

(2) MPU6000 (SPI-only sensor)
IMU detector tries to connect to an I2C device with this setup:
IO18 <---> SCL
IO05 <---> SDA

It fails at this line -> https://github.com/michprev/flyhero-esp32/blob/ad23bac4a47b9148e422a4c95394ee449cfebe4b/IMU/main/IMU_Detector.cpp#L83 , deletes I2C driver and tries to connect to an SPI device with this setup:
IO18 <---> SCLK
IO05 <---> SDI
IO17 <---> CS
IO16 <---> SDO

It succeeds and returns MPU6000 sensor. No crashes after that, everything OK!

(3) MPU9250 (SPI and even I2C device!)
IMU detector tries to connect to an I2C device with this setup:
IO18 <---> SCL
IO05 <---> SDA

It does not really fail since MPU9250 is capable of talking over I2C too but my code detects it as wrong device here -> https://github.com/michprev/flyhero-esp32/blob/ad23bac4a47b9148e422a4c95394ee449cfebe4b/IMU/main/IMU_Detector.cpp#L88

I2C driver gets deleted and IMU detector tries to connect to an SPI device:
IO18 <---> SCLK
IO05 <---> SDI
IO13 <---> CS
IO16 <---> SDO

It succeeds and returns MPU9250 sensor. After that program crashes.

The conclusion:
Please note that I am using the same pins for I2C and (when I2C fails) SPI too. When the device did not even respond to an I2C command (i2c_master_cmd_begin returned non-ESP_OK value) I uninstalled I2C driver, installed SPI driver on the same pins and it worked. When the device did respond to an I2C command but returned unexpected value I deleted the I2C driver anyway and installed SPI driver. Device did respond over SPI but then the program crashed.

Is it possible that i2c_driver_delete function forgets to uninitialize some pins or forgets to unsubscribe from an irq?

Hope it helps.
Thank you!

InfiniteYuan1, 2017-11-22:
OK, we are doing some tests and we will notify you as soon as possible.

koobest, 2017-11-24:
Hi, michprev, have you ever tried  commicute with MPU9250 only use SPI,  will the program crash?

michprev, 2017-11-24:
@koobest 
Hey, sure. When I comment out I2C code block in IMU detector (lines 143 to 153 of https://github.com/michprev/flyhero-esp32/blob/issue_i2c/IMU/main/IMU_Detector.cpp) there are no crashes.

michprev, 2017-11-24:
I have modified my example code - https://github.com/michprev/flyhero-esp32/commit/836a62adfd9d90181ba52d1bd0946076408b24c3
Now I am getting the same crash log for all 3 sensors (MPU6050, MPU9250, MPU6000).
When I revert 01befe4 there are no crashes.

michprev, 2017-11-24:
Because my code crashes always here https://github.com/espressif/esp-idf/blob/02304ad83e0a5f4815789d581446fa3afdd017b9/components/freertos/portmux_impl.h#L105 I have enabled portMUX debugging. There are my log files:
[log_crash.txt](https://github.com/espressif/esp-idf/files/1501379/log_crash.txt)
[log_working.txt](https://github.com/espressif/esp-idf/files/1501380/log_working.txt)

My ESP-IDF is up-to-date with https://github.com/espressif/esp-idf/commit/02304ad83e0a5f4815789d581446fa3afdd017b9 (in log_working.txt I have reverted 01befe4). Code I have run - https://github.com/michprev/flyhero-esp32/tree/1b6f21ef664158ff2eff6d5c14420db71940c921/IMU.
In both cases I have had MPU6000 connected.

koobest, 2017-11-28:
Hi, michprev. this problem may be related to RTOS, and currently I don't know how to solve it, but you can try to use a single core, maybe it works. configuration steps:
`make menuconfig-->Component config  --->FreeRTOS  --->Run FreeRTOS only on first core`  
hope it helps.

michprev, 2017-11-28:
@koobest 
Hi, good idea. Now I am getting:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4008a3b0  PS      : 0x00060633  A0      : 0x8008a4c1  A1      : 0x3ffb8cf0  
0x4008a3b0: xEventGroupSetBits at /home/michal/git/flyhero-esp32-idf/components/freertos/./event_groups.c:338

A2      : 0x3ffba230  A3      : 0xf5000000  A4      : 0x3ffba23c  A5      : 0x00000000  
A6      : 0x3ffba248  A7      : 0x3ffafc50  A8      : 0x00040000  A9      : 0x00040000  
A10     : 0xf5000000  A11     : 0x00000000  A12     : 0x00000237  A13     : 0x3ffb8d6c  
A14     : 0x3ffba230  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xf5000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008a3b0:0x3ffb8cf0 0x4008a4be:0x3ffb8d20 0x40089412:0x3ffb8d50 0x40089503:0x3ffb8d90
0x4008a3b0: xEventGroupSetBits at /home/michal/git/flyhero-esp32-idf/components/freertos/./event_groups.c:338

0x4008a4be: vEventGroupSetBitsCallback at /home/michal/git/flyhero-esp32-idf/components/freertos/./event_groups.c:696

0x40089412: prvProcessReceivedCommands at /home/michal/git/flyhero-esp32-idf/components/freertos/./timers.c:983

0x40089503: prvTimerTask at /home/michal/git/flyhero-esp32-idf/components/freertos/./timers.c:983
```
It is still being worked with event group after it is deleted maybe?

koobest, 2017-11-29:
Hi, michprev. we have solved this problem, and the IIC driver will be updated soon.

koobest, 2017-11-29:
it caused by eventgroup

Kasperdelasopa, 2018-03-31:
is it updated? I downloaded the latest version but i still get that error.

koobest, 2018-04-02:
Hi,  @Kasperdelasopa,
I'm sorry that the update of i2c did not help you, we have had two updates on the I2C driver.  by the way  are you michprev? The test we did befor wase based on the demo provided by michprev.