espressif/esp-idf #1194: Forced assertion in TCP

thiker5, 2017-10-30:
during tcp allocation on PCB , when the available # of pcbs exceeds the limit (by default 16), the TCP stack will try to kill the oldest PCB. 
one of the options is kill PCBs in fin_wait_2 & fin_wait_1 , which is great, but if this happens it will force assertion, where the function "tcp_kill_state" , do assertion for non closing or last_ack states. Is this intentional?

projectgus, 2017-10-30:
Not intentional, thanks for reporting it.

As it happens we've just merged a fix for this (removing the assertion, essentially). Commit 18dcbfa1 should reach master branch on github soon.

FayeY, 2017-11-08:
Hi @thiker5 , did the above suggestion help you resolve the issue?

closed 2017-11-10 by FayeY

patrykkrz, 2020-02-12:
@projectgus After add lwip as submodule (https://github.com/espressif/esp-idf/commit/b999d10e8c44d912a0286d12f18b183fbd08975e) assert functions are back.
Maybe assert should be configurable via sdkconfig in _Component Config->LWIP_ ?
I currently need to disable assert in _Compiler options-> Assertion level_, but this affects on all modules.

projectgus, 2020-03-02:
Hi @patrykkrz ,

Thanks for letting us know about this and sorry for the late reply. I've submitted a fix in esp-lwip, this issue should automatically update once it's merged to GitHub. May take a little longer before IDF has the fix in its lwip submodule.